micropython/micropython #15144: RP2:

randomwraith, 2024-05-28:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

RP2, Pi Pico W

### MicroPython version

Hi,
This maybe a pico-sdk bug? But posting here since I'm building micropython for Pico W with double precision ...

Building v1.22.0-402-g03cf4d498-dirty, using ARM toolchain 13.2.Rel1-x86_64-arm-none-eabi
System specs are: Linux 6.1.67-gentoo-x86_64,  AMD Ryzen 5 1600X, 32GiB RAM 

Build fails at object linking stage.

Solution provided, but this might break something else downstream?

### Reproduction

make -C mpy-cross
cd ports/rp2/ 
make BOARD=RPI_PICO_W submodules
make BOARD=RPI_PICO_W clean
make BOARD=RPI_PICO_W

### Expected behaviour

Expect object linkage to complete successfully

### Observed behaviour

In file included from /micropython/lib/pico-sdk/src/rp2_common/pico_platform/include/pico/platform.h:72,
                 from /micropython/lib/pico-sdk/src/common/pico_base/include/pico.h:33,
                 from /micropython/lib/pico-sdk/src/common/pico_binary_info/include/pico/binary_info/code.h:13,
                 from /micropython/lib/pico-sdk/src/common/pico_binary_info/include/pico/binary_info.h:28,
                 from /micropython/lib/pico-sdk/src/rp2_common/pico_standard_link/binary_info.c:8:
/micropython/ports/rp2/sys/cdefs.h:234: error: "__packed" redefined [-Werror]
  234 | #define __packed        __attribute__((__packed__))
      | 
In file included from /micropython/lib/pico-sdk/src/common/pico_binary_info/include/pico/binary_info.h:24:
/micropython/lib/pico-sdk/src/common/pico_binary_info/include/pico/binary_info/structure.h:21: note: this is the location of the previous definition
   21 | #define __packed __attribute__((packed))
      | 
cc1: all warnings being treated as errors
make[3]: *** [CMakeFiles/firmware.dir/build.make:6871: CMakeFiles/firmware.dir/micropython/lib/pico-sdk/src/rp2_common/pico_standard_link/binary_info.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:1516: CMakeFiles/firmware.dir/all] Error 2
make[1]: *** [Makefile:91: all] Error 2
See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:60: all] Error 1

### Additional Information

Bug fixed with following patch on 
micropython/lib/pico-sdk/src/common/pico_binary_info/include/pico/binary_info/structure.h

21c21
< #define __packed __attribute__((packed)) 
---
> #define __packed __attribute__((__packed__))

robert-hh, 2024-05-28:
Hmm... compiling with the ARM toolchain 13.2.Rel1-x86_64-arm-none-eabi on a Debian 12 system I do not get these errors. Looking into the include files of gcc-arm-none-eabi, I find in .../arm-none-eabi/include/sys/cdefs.h:
`#define	__packed	__attribute__((__packed__))`
So maybe cdefs.h is not included by default in your set-up.

randomwraith, 2024-05-28:
... agh, I must have messed something up there  ... I deleted the entire tree and recloned. The build runs completely fine end-end without patching.

I'll close this as resolved, although there is the spurious define in /structure.h - thanks for the help