micropython/micropython #15154: Build a esp32 firmware with version 1.22.2 meet a build issuse

ipconfiger, 2024-05-28:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

esp32 port

### MicroPython version

MicroPython 1.22.2

### Reproduction

next is the build script:

#!/bin/bash

export BUILD_DIR=$(pwd)

# echo "--- CLONING ULAB ---"
# git clone --depth 1 https://github.com/v923z/micropython-ulab.git ulab

echo "--- CLONING MICROPYTHON ---"
git clone --depth 1 -b v1.22.2 https://github.com/micropython/micropython.git

echo "--- CLONING ESP-IDF ---"
cd $BUILD_DIR/micropython/
git clone --depth 1 -b v5.0.4 --recursive https://github.com/espressif/esp-idf.git

echo "--- INSTALL ESP-IDF ---"
cd $BUILD_DIR/micropython/esp-idf
./install.sh
. ./export.sh

echo "--- MPY-CROSS ---"
cd $BUILD_DIR/micropython/mpy-cross
make

echo "--- ESP32 SUBMODULES ---"
cd $BUILD_DIR/micropython/ports/esp32
make submodules

echo "--- PATCH MAKEFILE ---"
cp $BUILD_DIR/micropython/ports/esp32/Makefile $BUILD_DIR/micropython/ports/esp32/MakefileOld
echo "BOARD = ESP32_GENERIC" > $BUILD_DIR/micropython/ports/esp32/Makefile
cat $BUILD_DIR/micropython/ports/esp32/MakefileOld >> $BUILD_DIR/micropython/ports/esp32/Makefile

echo "move extmod file"
cp -r $BUILD_DIR/ext_mods/ $BUILD_DIR/micropython/ports/esp32/modules

echo "--- MAKE ---"
make

### Observed behaviour

build error:

[ 98%] Building C object esp-idf/main_esp32/CMakeFiles/__idf_main_esp32.dir/__/__/pins.c.obj
[ 98%] Building C object esp-idf/main_esp32/CMakeFiles/__idf_main_esp32.dir/__/__/frozen_content.c.obj
/Users/alex/Projects/esp/build2/micropython/ports/esp32/build-ESP32_GENERIC/frozen_content.c:36441:27: error: redefinition of 'const_qstr_table_data_urequests'
36441 | static const qstr_short_t const_qstr_table_data_urequests[70] = {
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/alex/Projects/esp/build2/micropython/ports/esp32/build-ESP32_GENERIC/frozen_content.c:10122:27: note: previous definition of 'const_qstr_table_data_urequests' with type 'const qstr_short_t[70]' {aka 'const short unsigned int[70]'}
10122 | static const qstr_short_t const_qstr_table_data_urequests[70] = {
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/alex/Projects/esp/build2/micropython/ports/esp32/build-ESP32_GENERIC/frozen_content.c:36515:27: error: redefinition of 'const_obj_urequests_1'
36515 | static const mp_obj_str_t const_obj_urequests_1 = {{&mp_type_bytes}, 47477, 17, (const byte*)"\x25\x73\x20\x2f\x25\x73\x20\x48\x54\x54\x50\x2f\x31\x2e\x30\x0d\x0a"};
[... 54 lines cut ...]
      |                           ^~~~~~~~~~~~~~~~~~~~~~
/Users/alex/Projects/esp/build2/micropython/ports/esp32/build-ESP32_GENERIC/frozen_content.c:10205:27: note: previous definition of 'const_obj_urequests_11' with type 'mp_obj_str_t' {aka 'const struct _mp_obj_str_t'}
10205 | static const mp_obj_str_t const_obj_urequests_11 = {{&mp_type_bytes}, 1090, 9, (const byte*)"\x4c\x6f\x63\x61\x74\x69\x6f\x6e\x3a"};
      |                           ^~~~~~~~~~~~~~~~~~~~~~
/Users/alex/Projects/esp/build2/micropython/ports/esp32/build-ESP32_GENERIC/frozen_content.c:36527:27: error: redefinition of 'const_obj_table_data_urequests'
36527 | static const mp_rom_obj_t const_obj_table_data_urequests[13] = {
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/alex/Projects/esp/build2/micropython/ports/esp32/build-ESP32_GENERIC/frozen_content.c:10208:27: note: previous definition of 'const_obj_table_data_urequests' with type 'const void * const[13]'
10208 | static const mp_rom_obj_t const_obj_table_data_urequests[13] = {
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/alex/Projects/esp/build2/micropython/ports/esp32/build-ESP32_GENERIC/frozen_content.c:36543:33: error: redefinition of 'frozen_module_urequests'
36543 | static const mp_frozen_module_t frozen_module_urequests = {
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~
/Users/alex/Projects/esp/build2/micropython/ports/esp32/build-ESP32_GENERIC/frozen_content.c:10224:33: note: previous definition of 'frozen_module_urequests' with type 'mp_frozen_module_t' {aka 'const struct _mp_frozen_module_t'}
10224 | static const mp_frozen_module_t frozen_module_urequests = {
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~
/Users/alex/Projects/esp/build2/micropython/ports/esp32/build-ESP32_GENERIC/frozen_content.c:10224:33: error: 'frozen_module_urequests' defined but not used [-Werror=unused-const-variable=]
cc1: some warnings being treated as errors
make[3]: *** [esp-idf/main_esp32/CMakeFiles/__idf_main_esp32.dir/__/__/frozen_content.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [esp-idf/main_esp32/CMakeFiles/__idf_main_esp32.dir/all] Error 2

### Additional Information

No, I've provided everything above.

felixdoerre, 2024-05-28:
What's in `$BUILD_DIR/ext_mods`? Is there a `urequests` contained in there? I believe that `urequests` is now integrated by default in the micropython-variant that you are building, and you are copying another version there. `urequests` is included in `bundle-networking`, pulled in here: `ports/esp32/boards/manifest.py`. If you want to replace that, you'll need to edit that manifest to remove the stock `urequests`.

jonnor, 2025-03-03:
Please provide information on how to reproduce this issue on a clean MicroPython build on a recent release (1.24.1).