micropython/micropython #15157: PPP.active freezes MicroPython

pablogventura, 2024-05-28:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

esp32 port, ESP32_GENERIC_C3

### MicroPython version

MicroPython v1.22.2 on 2024-02-22; ESP32C3 module with ESP32C3

### Reproduction

```python
import network
import time
import machine
uart = machine.UART(1, baudrate=115200, tx=18, rx=19)
ppp = network.PPP(uart)
ppp.active(True)
ppp.connect()
ppp.active(False)
ppp.active(True)
```

### Expected behaviour

I would expect MicroPython to activate PPP or return an exception.

### Observed behaviour

After re-activating the PPP, MicroPython freezes, even to the point of not detecting a KeyboardInterrupt.

### Additional Information

The problem is independent of whether or not there is a PPP module connected to that Uart, and it is also independent of the activation of the PPP module with the AT commands.

I understand that the problem occurs whenever PPP is deactivated and reactivated.

victorallume, 2024-05-31:
What happens if you do a `ppp.disconnect()` before the `ppp.active(False)`

pablogventura, 2024-05-31:
There isn't a method called disconnect in PPP.

```python
>>> ppp.disconnect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'PPP' object has no attribute 'disconnect'
```

Facuu13, 2024-05-31:
I just tried it on version 1.23 and it still crashes. I attach the error:
>>> ppp.active(False)
A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : v1.23.0 on 2024-05-31
IDF version : v5.1.2
Machine     : ESP32C3 module with ESP32C3

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

agatti, 2024-06-10:
I bumped into this as well, although with ESP-IDF 5.0.4.  I haven't looked at the exact reason but in my case it doesn't get out of this loop: https://github.com/espressif/esp-lwip/blob/8dad8d3ee66840deee4acfc1601de4e396c594be/src/core/netif.c#L403-L411

Haven't tried to replace the lwip library with the one that ships with MicroPython, but from a quick glance the code seems more or less the same.

rambo, 2025-04-05:
https://github.com/micropython/micropython/pull/17008 might help you (it seems to work for me, ref https://github.com/micropython/micropython/issues/17079 )

dpgeorge, 2025-05-15:
Please try again now that #17138 is merged.