micropython/micropython #15170: network: wiznet: zeroconf

massimosala, 2024-05-30:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) regarding this feature, and didn't find any.

### Description

Working with boards with a wiznet ethernet interface, sometimes there isn't a DHCP server.
So, in this scenario, the interface is unusable.

It is easy to develop a [zeroconf](https://en.wikipedia.org/wiki/Zero-configuration_networking)
implementation to auto assign a IP address without DHCP server.

Zeroconf checks if an IP address is in use (by other network peers) doing ARP requests.

Problem: the class _WIZNET5K_ has the method _send_ethernet_ but not a _read_ethernet_

I have written, pure python, two function:
_arp_probe_ : about 40 lines
_zeroconf_ : 13 lines

At the moment I can send ARP probes but not check the peers replies... so my _zeroconf_ return a (random) IP address that theoretically can collide with other peers.

Are there other boards with a Ethernet interface that could benefit of _zeroconf_ ?
Class WIZNET5K: if I wtore a PR to add _read_ethernet_, will the maintainers accept it ?

### Code Size

I see in [network_wiznet5k.c(https://github.com/micropython/micropython/blob/30a9ccf4caa72c62cb8656a1572518fef34b08a4/extmod/network_wiznet5k.c#L248)
there are the two functions:
- wiznet5k_send_ethernet
- wiznet5k_recv_ethernet

_wiznet5k_send_ethernet_ is already available to Python, see https://github.com/micropython/micropython/issues/15169

I think adding a wrapper also for _wiznet5k_recv_ethernet_ will take few bytes.

### Implementation

- [X] I intend to implement this feature and would submit a Pull Request if desirable.
- [X] I hope the MicroPython maintainers or community will implement this feature.
- [ ] I would like to [Sponsor](https://github.com/sponsors/micropython#sponsors) development of this feature.

victorallume, 2024-05-31:
Not directly related to zeroconf, but you can use the wiznet interface without DHCP by assigning a static configuration using `ifconfig()` - pass it the network parameters (IP address, netmask, gateway and DNS server) and it should work

massimosala, 2024-05-31:
@victorallume 

> Not directly related to zeroconf, but you can use the wiznet interface without DHCP by assigning a static configuration using `ifconfig()` - pass it the network parameters (IP address, netmask, gateway and DNS server) and it should work

Hi

Yes, I know. 
But  for quick testing, like not at my desk, a straight cable from the wiznet to the laptop is nice.
The wiznet hasn't auto MDI-X but the laptop eth port yes.

jouellnyc, 2024-07-10:
Hello  @massimosala 

Not sure if you have seen these but , these here: https://www.aliexpress.us/item/3256804674673261.html are termed 'USR-ES1 W5500 Ethernet Network Module Hardware SPI to LAN/ Ethernet TCP / IP 51 / STM32 Microcontroller Program Over W5100'  

[image]

They are Wiznet 5500 apparently - I stumbled upon them and  was able to use them w/the wiznet firmware https://micropython.org/download/W5500_EVB_PICO/ and the pi pico. From my perspective, adding `wiznet5k_recv_ethernet`  would be great also anything for `arp`.

jonnor, 2024-07-13:
Hi @massimosala - you say you have written code to implement Zeroconf. Can you provide it here? And if it can be done in just some lines of Python, then maybe documenting it is all that is needed?

greezybacon, 2025-05-19:
This is implemented in #9877. It's a feature of LwIP called AUTOIP. It was implemented here with the description of `autoconf4`. Perhaps it should called something else such as `zeroconf`, but it's simple to implement.