micropython/micropython #15181: ESP32: Unable to use SDCard and any other SPI device in the same bus

kdschlosser, 2024-06-01:
### Port, board and/or hardware

ESP32

### MicroPython version

1.22.2

### Reproduction

If what I am seeing in the code is correct there is no way to use SPI SDCard with any other SPI device. This is due to the code in both the SDCard class wanting to initialize the SPI bus without checking the error output to see if the SPI bus has been initialized. In the SPI class the error is checked specifically for the error for the bus being initialized but still goes on and raises an exception for that error. 

The other issue is the inability to set the frequency based on the device that is attached to the bus. The ESP-IDF has a separation of the "bus" and the "device" for this specific reason. This allows devices running at different speeds to be used on the same bus. It also allows devices to have different settings as well. With how it is currently written is causes a lot of  limitations in what is able to be done and it forces the use of additional pins when it shouldn't be necessary. 

### Expected behaviour

To be able to use SPI SDCard on the same bus as other SPI devices. 

### Observed behaviour

Exception occurs for the bus already being active.

### Additional Information

It would be an ideal arrangement is a single "bus" object could be created and that bus object would then get passed to the different things that use SPI, That  Bus object would need to be declared in a header file where it can be included for use elsewhere. From the looks of it it should be something that can be done across most other ports without to much of an issue. The only one that would be a little more of a pain would be the STM32's and this is due to no easy way to change the bitrate without having to reinitialize the SPI driver each time a device needs to be communicated with but it is still able to be done. 

In my use case the board is something that is bought with the ESP32 already on board and the SDCard, LCD panel and touch panel are all on the same bus. It is also quite common to buy boards that don't have an MCU but have an SDCard and LCD panel that are attached to the same SPI bus. It is an impossibility to use these kinds of boards with MicroPython due to how the SPI is currently done in MicroPython.

### Code of Conduct

Yes, I agree

peterhinch, 2024-06-01:
I don't think this is a bug: rather a deliberate design decision.

It is possible to code around the problem by calling `spi.init()`. As an example [micropython-touch](https://github.com/peterhinch/micropython-touch/tree/master) copes with radically different baudrates for touch controller and screen [here](https://github.com/peterhinch/micropython-touch/blob/8024552a7b7f82479ad9c47288bed884ce0b28f0/gui/core/tgui.py#L270). I have used a similar technique where an SD card shares the bus with a screen.

In my opinion coding round this is preferable to the Adafruit Bus Device complexity of CircuitPython: it keeps things minimal in the more common case where arbitration is not needed. In other words, perform arbitration at the application level.

kdschlosser, 2024-06-01:
The problem is in the handling of the errors in the SPI driver. There is no need for it to raise an exception if the bus is already initialized. the same thing applies for the SDCard as well. If the bus is already initialized then OK don't raise the exception and continue on initializing the device. DO NOT return the same instance if the same pins are used as this is going to be a different device if machine.SPI is called passing the same parameters to it. This would remove the need to write any "work around" code and having to reinitialize the bus every time the bit rate needs to be changed. The bit rate is not used when initializing the bus. 

It doesn't have to be overly complicated like what is seen in circuitpython... It is able to be done much easier. I do not know why it was written the way it was in circuit python.

The lack of being able to pass a cs pin is also a sore point. Instead if letting the SDK handle that pin. Now the user needs to manage when it's state needs to be changed. Couple of downsides are, more memory consumption and it's going to be slower. 

The same format is able to be used where there is a single entry point for SPI. the things that need to be changed are the exception raising if the bus has already been initialized and a new instance needs to be returned not one that has already been created.

closed 2025-03-03 by kdschlosser