micropython/micropython #15182: libhydrogen error when trying to mboot with packing enabled

saraverbeecke, 2024-06-01:
### Port, board and/or hardware

stm32

### MicroPython version

MicroPython v1.22.0

### Reproduction

make BOARD=PYBV11

### Expected behaviour

building mboot with packing enabled i.e. `MBOOT_ENABLE_PACKING = 1`

### Observed behaviour

I get a couple of these lines:
```
../../../lib/libhydrogen/impl/random/Particle.h:8:1: note: previous definition of 'hydro_random_init' with type 'int(void)'
    8 | hydro_random_init(void)
      | ^~~~~~~~~~~~~~~~~
```

and I get a couple of these lines:
```
../../../lib/libhydrogen/impl/random/Particle.h:8:1: error: 'hydro_random_init' defined but not used [-Werror=unused-function]
    8 | hydro_random_init(void)
      | ^~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [build-PYBV11/lib/libhydrogen/hydrogen.o] Error 1
```

The pyhy version that is installed is 0.0.9

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

jonnor, 2024-09-01:
Hi and thank you for the report. This only happens with MBOOT_ENABLE_PACKING=1 ? Does the same issue occur in git master? Do you have any ideas why it happens or what might be done to fix it?

andrewleech, 2024-09-02:
Just to check, try running `make -C ports/stm32 BOARD=PYBV11 MBOOT_ENABLE_PACKING=1 submodules` to ensure the library is updated correctly?