micropython/micropython #15192: RP2 Hanging on file write after soft reboot when a thread was started

eleroy, 2024-06-02:
### Port, board and/or hardware

RP2 Pico

### MicroPython version

MicroPython v1.23.0-5.g3613ad962 on 2024-05-31; Raspberry Pi Pico with RP2040

### Reproduction

- Copy this code to main.py
- Run the code
- Do soft reboot

If the thread is started before writing to the file, it works. If the thread has not been started, it hangs.

```
import _thread
import time
thread_running = True
def second_thread():
    while thread_running:
        print("Second thread running")
        time.sleep(1)
#thread = _thread.start_new_thread(second_thread,()) # Works if thread is started here

print("Writing file 1") # Hangs here after soft reboot
with open("test1.bin","w") as fh:
    fh.write(b'test')
    
print("Writing file 2") # Hangs here after soft reboot
with open("test.bin","ab") as fh:
    fh.write(b'test')
print("End writing file")

thread = _thread.start_new_thread(second_thread,())

try:
    while True:
        print("First thread running")
        time.sleep(1)
except KeyboardInterrupt:
    thread_running = False
    time.sleep(2)

```

### Expected behaviour

The file should be written without blocking after a soft reboot whether the thread is started before or after writing the file.

### Observed behaviour

After soft reboot, the micro hangs and it needs a hard reset.

### Additional Information

I think everything is here. I'm running the code on a custom hardware (compatible with pico), I'll try with a pico whenever I can access one. 
Please ask if you want me to try something or you need more info.
Also, I did not experience bugs on 1.21.0 (although I did not try this specific code yet). It seems it happens in 1.22/1.23.

### Code of Conduct

Yes, I agree

eleroy, 2024-06-03:
I tested on a pico, I can confirm I observe the bug.
The bug occurs on 1.23,1.22.1,1.22.2
But it works fine on 1.21

Also a simple workaround is to add `thread = _thread.start_new_thread(lambda: None,())` at the beginning of the script.

eleroy, 2024-06-06:
Hello, 

For people using Thonny, the IDE seems to soft reboot and interrupt when pressing stop. 

If you had a running thread in your micro python code it seems you will not be able to upload files after.

It's best to put the workaround in the boot.py file so that uploading files with Thonny works well.

massimosala, 2024-06-13:
Hi @eleroy

I encounter a similar problem but in a different scenario: rp2 with wiznet5k.

When my program - without threads - ends, it calls

> machine.reset()

The rp2 reboots but at the REPL I have to hit ENTER otherwise the micro doesn't run.

I don't known if my issue is related to yours (the wiznet internal code can use threads) or should I have to open another issue?

I hope some guru read these comments and let me know.

massimosala, 2024-06-13:
I read other posts about problems with _rp2_ at reboot.

I will try this: deactivate _i2c_ and _spi_ buses before calling _reset_
If I found a solution, I update my comment.

projectgus, 2025-02-04:
Sorry no one got back to you earlier. This sounds like the issue reported recently as #16619, with a fix submitted in #16620.

closed 2025-02-07 by dpgeorge