micropython/micropython #15196: UART device only responds when running in REPL, but not when running standalone on Raspberry Pi Pico W

1-rafael-1, 2024-06-03:
### Port, board and/or hardware

rp2 port, Raspberry Pi Pico W

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040

### Reproduction

I am fairly new to this, so assuming i am doing something wrong.
Full code is here: https://github.com/1-rafael-1/pi-pico-alarmclock

I have tried to strip down a smaller repro, but to my surprise the stripped-down code works as expected, while the full code does not. I am at a loss as to why and hopefully somebody can give me a hint. 

Here is a stripped-down version:

[image]

file `main.py` in base folder:

```python
import micropython
from classes.app_manager import ApplicationManager

@micropython.native
def main():
    micropython.alloc_emergency_exception_buf(100)
   
    try:
        app = ApplicationManager()
        app.initialize()
        app.run()
    except Exception as e:
        print("An unexpected error occurred in main.py: ", e)
    except KeyboardInterrupt:
        pass
    finally:
        pass

if __name__ == "__main__":
    main()
```
file `dfplayer_mini.py` in folder `drivers`:

```python
# taken from https://github.com/tomdionysus/dfplayermini-micropython
# modified to choose uart instance
# fixed missing import of utime
# fixed accessing class-level constants
# fixed missing method validate_checksum
# fixed missing constants
# added option to control power to the module
# added option to check if the module is busy

import machine
import utime

class DFPlayerMini:

    # Start and End Bytes
    START_BYTE = 0x7E
    END_BYTE = 0xEF
    VERSION_BYTE = 0xFF

    # Command Codes
    CMD_PLAY = 0x0D
    CMD_PAUSE = 0x0E
    CMD_NEXT = 0x01
    CMD_PREV = 0x02
[... 574 lines cut ...]

(Code is here: https://github.com/1-rafael-1/pi-pico-alarmclock -> alarm_mgr holds the code calling the device and sound_mgr manages the device/loads the driver)

### Observed behaviour

Run the system via VS Code/REPL: everything works.
Run the system after loading everything on the Pico by powering the pico: Everything works, except the sound. 

I have tried to capture logs, basically in this mode all UART commands to read state from the mp3-player time out. 

Change nothing, just attach the Pico back to VS Code, run main.py again: Everything works including the mp3-player.

### Additional Information

I have tried to isolate things as much as I had ideas how to:

- move the sequence finally calling the device out of the thread it tuns in -> no change
- remove initiating threads altogether -> no change
- call the device independently of the rest of the system, just placing it in the main loop in app_mgr -> no change
- checked the wiring 100 times, measured voltage supplied to the device in both modes -> no clue
- and finally: for this issue strip down a version, removing basically all other components.... -> works, but I do not understand why

### Code of Conduct

Yes, I agree

TimHanewich, 2024-08-07:
Experiencing this with an RYLR988 LoRa module as well (interfaced over UART). Works well within REPL via Thonny, does not work when plugged in standalone.

robert-hh, 2024-08-08:
@TimHanewich That looks like a Thonny problem. Please be sure that the code you want to run was actually copied permanently to the device. Thonny may hide the fact that it was not. Use another terminal emulator like mpremote, PuTTY, screen, picocom, ... Any of these are less intrusive than Thonny. The recommended tools is mpremote.