micropython/micropython #15200: Unable to flash firmware via fwupdate.py on micropython 1.23.0

victorallume, 2024-06-04:
### Port, board and/or hardware

stm32/PYBD

### MicroPython version

MicroPython v1.23.0 on 2024-05-31; PYBD-SF6W with STM32F767IIK
Have compiled and flashed mboot from the v1.23.0 tree
lfs-fuse version:
```❯ ./lfs --version
littlefs-fuse version: v2.7
littlefs version: v2.8
littlefs disk version: lfs2.1
```

### Reproduction

This works (using FAT filesystem):
```
import pyb, os
b = pyb.Flash(start=0)
os.VfsFat.mkfs(b)
# copy firmware fw.dfu.gz and fwupdate.py onto filesystem via PC
# reboot
# back on the pyboard:
import stm
import fwupdate
status_addr=stm.RTC + stm.RTC_BKP1R
stm.mem32[status_addr]
fwupdate.update_mpy('fw.dfu.gz', 0x80000000, 2*1024*1024, fs_type=fwupdate.VFS_FAT, status_addr=stm.RTC + stm.RTC_BKP1R)
```

Whereas this (using Lfs2) does not work:
```
import pyb, os
b = pyb.Flash(start=0)
os.VfsLfs2.mkfs(b)
# copy firmware files and fwupdate.py onto filesystem via PC (lfs-fuse)
# reboot
# back on the pyboard:
import stm
import fwupdate
status_addr=stm.RTC + stm.RTC_BKP1R
stm.mem32[status_addr]
fwupdate.update_mpy('fw.dfu.gz', 0x80000000, 2*1024*1024, fs_blocksize=4096, fs_type=fwupdate.VFS_LFS2, status_addr=stm.RTC + stm.RTC_BKP1R)
```
It doesn't flash, and status code -222 is returned in `stm.mem32[status_addr]`

### Expected behaviour

Both FAT and LFS2 can flash firmware via fwupdate/mboot

### Observed behaviour

FAT works; LFS2 doesn't work

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2024-06-04:
> Whereas this (using Lfs2) does not work:

The default configuration for PYBD mboot does not include littlefs 2 support.  It only includes FAT support (that's because the default filesystem on PYBD is FAT, which is due to mass storage being enabled).

If you configure your own board, you can enable this option in `mpconfigboard.h` and rebuild mboot:
```c
#define MBOOT_VFS_LFS2 (1)
```
But note that, because littlefs on-disk version can change in the future, it may not be a good idea to use a littlefs partition to update the main firmware through mboot.  Because if the application updates littlefs and that changes the on-disk version, then mboot won't be able to read it and won't be able to do any updates.  In that case you'll need to update mboot first to use the new littlefs version, then the application, then finally upgrade the filesystem to the new on-disk version.

Because of this issue with the filesystem format possibly changing (although that won't happen for FAT), there's also a new filesystem type `MBOOT_VFS_RAW` which is just a very simple contiguous region containing the application data.  For PYBD_SF6 one could use the secondary SPI flash with that raw filesystem.

victorallume, 2024-06-05:
Thanks, that fixed it. 

What you say secondary SPI flash - is that the 'bonus' 2MB (i.e. not the internal flash, nor the 2MB that is used by the filesystem)? How would we access that?

dpgeorge, 2024-06-05:
> What you say secondary SPI flash - is that the 'bonus' 2MB

Yes, it's the second external 2MiB SPI flash.  This can be used for overflow firmware, but on the PYBD_SF6 it's rarely used because that chip has a lot of internal flash on the MCU.

> How would we access that?

I'll need to make some adjustments to the main stm32 firmware to make this accessible for use as a filesystem.

victorallume, 2024-06-05:
OK thanks. I remember discussing it with you at some point. Is it possible to currently access this flash region not as a filesystem? i.e. if we wanted to use it for mboot in raw mode? How would we write to it from micropython?

dpgeorge, 2024-06-13:
> Is it possible to currently access this flash region not as a filesystem?

This second flash is currently only accessible as read-only memory-mapped data.  It can't currently be written to by the main firmware.

It would be good to be able to write to it.  I will look into this.