micropython/micropython #15205: ImportError: No module named 'bluetooth'

RobeAlVa, 2024-06-04:
### Port, board and/or hardware

XIAO Seeed Sense BLE nrF52840

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; XIAO nRF52840 Sense with NRF52840

### Reproduction

import bluetooth

### Expected behaviour

Run the code

### Observed behaviour

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'bluetooth'

### Additional Information

I'm using a XIAO Seeed Sense BLE nrF52840 in Micropython where I want to initialize BLE communication but when I try to run the code I get an error that says "ImportError: no module named 'bluetooth'"

### Code of Conduct

Yes, I agree

robert-hh, 2024-06-04:
The nrf port is different and is especially missing documentation. A few examples showing how to use Bluetooth are here: https://github.com/micropython/micropython/tree/master/ports/nrf/examples

RobeAlVa, 2024-06-04:
Oh, I didn't know about that, thanks a lot!

andrewleech, 2024-06-05:
Yeah the nrf port still uses a legacy bluetooth api, different to the normal one used by other ports (so things like aioble doens't work here yet)
I've been working on getting nrf up to speed with the standard api here https://github.com/micropython/micropython/pull/8318 but it needs some more testing and refinement before it's ready to merge.

aphor, 2024-09-01:
How are things going @andrewleech?

I just started out, but I might be able to test your branch. I am able to build micropython bin files from outside the micropython tree, and I have been testing them on an arduino nano33BLE.

andrewleech, 2024-09-02:
Cheers, I've recently rebased that PR to master but haven't had a chance to re-test it still or run the test suite again.

aphor, 2024-09-03:
OK. I've got your branch from #8318 in my local, and I'll try it on my nano33BLE.

aphor, 2024-09-05:
I managed to hack together a build on `ARDUINO_NANO_33_BLE_SENSE` from #8318 but I had to keep the soft device in the build to get USB REPL for debugging. It acts like there are conflicting drivers :)

I was able to connect to the nano33ble over bluetooth, and pair, but my test code also uses the IMU and gets an ambiguous memory error when I call the python method to try to read from that device.

The alternative debugger options are all each a hot mess in their own ways. I will look into getting REPL running on a BLE UART or something.

andrewleech, 2024-09-08:
Ah interesting / sorry about the repl issues, I rebased that PR recently after other big repl / stdio changes were merged but haven't been able to test it properly so might have broken something there. It certainly is intended to work without SD

Yeah as far as debugging and flashing is concerned it is historically quite different on every chip / port so far, presumably driven by every vendor having different default / recommended practices. I'm pretty keen though to try to consolidate more of this, there are newer open source tools (hw and sw) that make this far easier these days. Take a look at probe.rs it's a rather impressive project! That combined with a cmsis/daplink probe should with with most chips on the market it seems.
As much as I love the speed on jlink its price is too prohibitive for most people if they're not working on this stuff professionally.

robert-hh, 2024-09-08:
> As much as I love the speed on jlink its price is too prohibitive for most people if they're not working on this stuff professionally.

jlink edu mini's price tag comes is USD $15. It does support only the basic features like firmware loading and debugging, but that's already good.

andrewleech, 2024-09-08:
Where do you get it at that price @robert-hh ? I generally see it starting at USD 60 eg.
https://www.sparkfun.com/products/24078

Oh I see [thepihut](https://thepihut.com/products/segger-j-link-edu-mini-jtag-swd-debugger?srsltid=AfmBOoqnJrG97TrlIsgGOb1xzwwlTBHG4K3xNd51u0qTjFbhn3A4_pEG) have it for roughly half that. Still a lot more expensive than my latest high speed daplink that cost me about $3

To be honest my other issue with jlink is my inability to find any way to conveniently flash a chip with it from command line in a simple one-liner. The official way is jlink commander tool with a custom script file which is far more tedious. Most other tools like probe.rs or pyocd that can talk to jlink and flash with simple commands are slow because they're not using the jlink binary libraries. The jlink is fast for debugging in vscode though!

robert-hh, 2024-09-08:
You're right. I had purchased one at Antratek for that price. But now they have a price tag of 57€. Then it might be better for a little extra to use the Jlink EDU device.
And yes, the flashing tool to be used with the jlink edu mini is not convenient. When possible I use the Ozone debugger for flashing.