micropython/micropython #15208: ESP32+SPIRAM Neopixel issue since v1.22.0

melazarus, 2024-06-05:
### Port, board and/or hardware

ESP32 WROOVER module on various boards

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Generic ESP32 module with SPIRAM with ESP32
MicroPython v1.22.2 on 2024-02-22; Generic ESP32 module with SPIRAM with ESP32
MicroPython v1.22.1 on 2024-01-05; Generic ESP32 module with SPIRAM with ESP32
MicroPython v1.22.0 on 2023-12-27; Generic ESP32 module with SPIRAM with ESP32

### Reproduction

1) make a connection to the REPL using a simple tool like mpremote, screen, putty etc..
    Don't use a IDE like Thonny, pycharm etc.. these modify someting that makes it hard to reproduce.
2) run the following code (don't close the connection after this yet)
`from machine import Pin`    
`from neopixel import NeoPixel`    
`np = NeoPixel(Pin(13), 10)`    
`np.fill((8,0,0))`    
`np.write()`
This should produce 10 red neopixels, but instead you see a mix of red, green/yellow and "black"
3) run the following code
`
np.write()
`
now all leds are red as they should be
4) run the following code to create an exception
`
1/0
`
followed by 
`
np.write()
`
now the LEDS are showing the wrong colors again.

### Expected behaviour

[image]

All LEDS should light up RED

### Observed behaviour

[image]

Instead the colors are wrong

### Additional Information

We tried using multiple EPS32's, different power supplies, and different LED strings.
They all show the same behaviour. If you run the code at startup in main.py it occurs much less frequent.

### Code of Conduct

Yes, I agree

melazarus, 2024-06-05:
Some extra things I found out.
commit, 4365edb81 says: esp32: Change minimum supported IDF version to v5.0.4
So I compiled the commit before that (d0758d8a3) one time with 5.0.2 and one time with 5.0.4
The issue only occurred when compiled with IDF 5.0.4.

So it looks that the version of IDF breaks something in the SPIRAM version of micropython for ESP32.

I'll play around with compiling 1.23 with different versions of the IDF to see if that changes anything.

melazarus, 2024-06-05:
Ok, The issue is fixed when compiling micropython with IDF 5.1.2

Can we raise the minimal IDF version from 5.0.4 to 5.1.2 for this issue?

I can make the changes and create a PR.

dpgeorge, 2024-08-16:
Should be fixed by fbb02d3aeeaa9da082f3ccd411d8e164a526046e

closed 2024-08-16 by dpgeorge