micropython/micropython #15216: Error freezing a python source file

kdschlosser, 2024-06-05:
### Port, board and/or hardware

any

### MicroPython version

1.23.0

### Reproduction

This error happens if you freeze a python source file with code like what is seen below.

```python

TEST1 = 'Test1'
TEST2 = 'Test2'
TEST3 = 'Test3'

print(
    f'{TEST1}, '
    f'{TEST2}, '
    f'{TEST3}'
)
```

The error is "invalid syntax"

If I change the code to the following no errors occur.

```python
TEST1 = 'Test1'
TEST2 = 'Test2'
TEST3 = 'Test3'

print(f'{TEST1}, {TEST2}, {TEST3}')
```

the error occurs because of the formatted text being on multiple lines. This is correct python syntax and I am not sure why it fails. If I do this there is no error.

```python
print(
    'Test1, '
    'Test2, '
    'Test3'
)
```

the problem only occurs when using the format specifier. 

### Expected behaviour

For the code to properly run.

### Observed behaviour

Syntax Error occurs

### Additional Information

I am not sure if there is an error if I am not freezing the file. For some reason I believe that it does still occur if I run the source file instead of freezing it.

### Code of Conduct

Yes, I agree

kdschlosser, 2024-06-05:
It also appears doing this also triggers a syntax error.

```python
print(f'{self.__class__.__name__}(raw_x={self._last_x}, raw_y={self._last_y}, x={data.point.x}, y={data.point.y}, state={"PRESSED" if data.state else "RELEASED"})')
```

I am testing to see if it is specifically this part

```python
{"PRESSED" if data.state else "RELEASED"}
```

I am thinking the use of the double quotes is not taken into consideration and that is what is throwing the syntax error. It is correct Python syntax and it does run properly in C python.

dpgeorge, 2024-06-05:
MicroPython doesn't support concatenation of f-strings by having them adjacent to each other, eg `f"" f""`.  (But it does support normal strings and bytes like `"a" "b"`.)

dpgeorge, 2024-06-06:
Actually, adding support for concatenated f-strings turned out to be very simple.  See #15217.

kdschlosser, 2024-06-06:
Hey, Thanks a bunch for fixing this. It has been something that has been a thorn that would come up every now and again when I forgot that the f-strings couldn't do the concatenation.

 so we should be able to close this issue then since it has been fixed, yes?

dpgeorge, 2024-06-06:
Yes, we can close this now.

f-string concatenation implemented in a066f2308f7b0d872352073cec0a945dca3a7a9c

closed 2024-06-06 by dpgeorge