micropython/micropython #15220: rp2: Pico W, hang setting up board LED

Gadgetoid, 2024-06-06:
### Port, board and/or hardware

RP2 port on a Raspberry Pi Pico W

### MicroPython version

MicroPython v1.24.0 preview. Built locally from source at "master."

### Reproduction

1. Checkout "master"
2. Build the RPI_PICO_W port
3. Flash it
4. Run
```
from machine import Pin
led = Pin("LED", Pin.OUT)
print("Hello World")
```

### Expected behaviour

5. Observe "Hello World"

### Observed behaviour

Indefinite hang.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

Gadgetoid, 2024-06-06:
This probably means wireless is hosed in general. My money is on https://github.com/micropython/micropython/commit/74fb42aa82d537b6fefb463907274bd660c6ce8d. Just cherry-picking that on to v1.23.0 to confirm.

robert-hh, 2024-06-06:
Confirmed that the actual master branch shows this behavior, but version 1.23.0 is fine. So something broke in the communication between the RP2040 and the CYW43. At Pico W, the LED is conncted to the CYW43 module.

Gadgetoid, 2024-06-06:
I’ve had reports that stuff as simple as “time.sleep” is broken but I’ve not been able to replicate.

I may have included the offending commits in our v1.23.0 custom release in an attempt to get my pins.csv fix 😭 but I’ve sorted that now.

dpgeorge, 2024-06-07:
Thanks for the report.  I can confirm the issue.  See #15222 for a fix/workaround.

dpgeorge, 2024-06-07:
Fixed by 80a4f632ee812de9f2b22e88abb36ca36d230ec0

closed 2024-06-07 by dpgeorge