micropython/micropython #15230: [RP2] Not getting serial device with 1.23 (good with prior versions)

TRadigk, 2024-06-08:
### Port, board and/or hardware

Raspberry Pico

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040

### Reproduction

I took my program code and stripped it down to a working repro. There are more lines that could be deleted but help with visualization. The content is saved to boot.py. There is no other file saved on the microcontroller.
With any version **prior** 1.23 this will yield a serial device in device manager, whenever the device is connected to the computer. With version 1.23 it will not and can only recovered by flashing a firmware version prior 1.23:

```python
import _thread
import asyncio  # type: ignore
from time import sleep
import gc
import machine

synchronousLoopAlive: bool = True
anyLoopAlive: bool = True
led = machine.Pin(25, machine.Pin.OUT)

async def yieldingBackgroundTaskOne():
    while anyLoopAlive:
        print("One")
        led.value(not led.value())

        await asyncio.sleep_ms(1000)

def synchronousTask():
    while synchronousLoopAlive:
        sleep(2)

if __name__ == "__main__":
    try:
        sleep(3)
        loop = asyncio.get_event_loop()
        gc.collect()

        programThread = _thread.start_new_thread(synchronousTask, ())
        print("program thread started")
        taskOne = loop.create_task(yieldingBackgroundTaskOne())
        print("task One started")

        loop.run_forever()
    except KeyboardInterrupt:
        synchronousLoopAlive = False
        anyLoopAlive = False
        loop.stop()
        print("INTERRUPTED - STOPPING EVERYTHING")
    except Exception as e:
        print(f"error occurred: {e}")

```

### Expected behaviour

Expected a serial device in device manager as in all versions prior 1.23.

### Observed behaviour

No serial device in device manager with micropython 1.23, but with any version prior 1.23.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

robert-hh, 2024-06-08:
I cannot replicate that with a v1.24.0-preview 34. The serial device always registers, with and without your code. Tested with Linux. If you use a USB hub, try to power cycle the hub or use a direct port at your PC.

TRadigk, 2024-06-08:
Hi Robert,
thank you very much for your answer and testing the issue on your end.

Though you did not attempt to reproduce with the exact version, I used your comment as motivation to download and install v1.24.0-preview 34. I then tested if the device is enumerated on my desktop PC, which is totally different to my usual machine (intel vs. amd platform) - the device was also not found. I've tried every possible USB Port and also another cable. Since my desktop PCs graphics card is a nVidia 2080 Ti I even have another USB connection there which already prove to behave differently in regards to other ports on my PC - the device was not found.

I then went ahead and plugged it into my raspberry pi zero with bookworm image, which currently is sitting next to me due to another project and executed lsusb - the device is not found.

Last but not least I flashed the same micropython release onto an ESP32 S3 and copied over the given code. The device is always found.

eleroy, 2024-06-11:
Hi,

I can confirm the bug.

I think this is due to the synchronousTask being shutoff by the soft reboot while still running. 
If you add a sleep after the loop.stop() which is larger than 2s (same sleep as in synchronousTask, it works).
```
import _thread
import asyncio  # type: ignore
from time import sleep
import gc
import machine

synchronousLoopAlive: bool = True
anyLoopAlive: bool = True
led = machine.Pin(25, machine.Pin.OUT)

async def yieldingBackgroundTaskOne():
    while anyLoopAlive:
        print("One")
        led.value(not led.value())

        await asyncio.sleep_ms(1000)

def synchronousTask():
    while synchronousLoopAlive:
        sleep(2)

if __name__ == "__main__":
    try:
        sleep(3)
        loop = asyncio.get_event_loop()
        gc.collect()

        programThread = _thread.start_new_thread(synchronousTask, ())
        print("program thread started")
        taskOne = loop.create_task(yieldingBackgroundTaskOne())
        print("task One started")

        loop.run_forever()
    except KeyboardInterrupt:
        synchronousLoopAlive = False
        anyLoopAlive = False
        loop.stop()
        sleep(2)
        print("INTERRUPTED - STOPPING EVERYTHING")
    except Exception as e:
        print(f"error occurred: {e}")
```

TRadigk, 2024-08-09:
Tested today: RPI_PICO-20240808-v1.24.0-preview.181.g91f4a6b9e - same result.

projectgus, 2024-09-04:
Sorry I missed this one when it first came up.

Something that changed when we added Runtime USB support is that after a hard reset USB isn't initialised until after `boot.py` finishes running. So if you put a program that doesn't exit in `boot.py` then USB won't initialise. This is how it's always worked on some ports such as stm32, but it's a recent change on rp2. Sorry it's caused so much of a hassle.

The recommended approach is to put the program in `main.py` instead of `boot.py`. `main.py` runs after hardware is fully initialised, on all ports.

This of course doesn't solve the problem of soft-bricking the rp2 if boot.py doesn't exit, though. Some ports have a "safe mode boot" that skips boot.py and main.py when starting up, but rp2 port currently doesn't have this - and maybe we should consider one.

 There's a file called `flash_nuke.uf2` that formats the entire flash if copied to the bootloader in BOOTSEL mode. This will erase the entire flash and allow you to flash a recent MicroPython and copy files to the filesystem again. There's no official source for this, but I'd suggest downloading this one from @robert-hh: https://forum.micropython.org/viewtopic.php?t=10838#p59604

TRadigk, 2024-09-04:
Hi @projectgus 
I thank you so much for this answer!

In my case boot.py was existing, but main.py was not. Like in desktop Python I would run my "infinite loop" in boot.py. This had worked in the past and thus I never needed a main.py.

Today I reworked my code to basically no content at all in boot.py, since I am not using any feature in there and put my startup and "infinite loop" into main.py. After verifying the code working on uPy 1.21 I updated to RPI_PICO-20240903-v1.24.0-preview.278.g5e692d046.uf2 again.

And now the device enumerates as USB serial device correctly and I have not lost any functionalities. 🥳🎉

I will keep this in mind for further projects.
I also found hints about this requirement of having **both** boot.py and main.py in here: https://docs.micropython.org/en/latest/pyboard/tutorial/script.html, though it is not stated that boot.py is required to "run to end". Also I think this "getting started" guide should not reside at pyboard section but in a layer above, since this is not specific to pyboard but to micropython. One would not find the hints when "getting started with ESP32" or "getting started with RP2".

eleroy, 2024-09-04:
Hi,

I didn't notice you were using boot.py to run your script.

Still if I interrupt the program (in main.py) without waiting on the other thread to stop I get 

```shell
program thread started
task One started
One
One
INTERRUPTED - STOPPING EVERYTHING
MicroPython v1.24.0-preview.278.g5e692d046 on 2024-09-03; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> FATAL: uncaught exception 200076d0
OSError: 
```

And I loose usb serial. 
With sleep(2) to wait on the thread to stop, it works fine.

```
program thread started
task One started
One
One
One
One
INTERRUPTED - STOPPING EVERYTHING
MicroPython v1.24.0-preview.278.g5e692d046 on 2024-09-03; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> 
```

tested on v1.24.0-preview.278.g5e692d046

I think there is something going on with thread deinit on soft reboot/repl init (see my other issue #15194)

TRadigk, 2024-09-04:
Yes, you're right. The program should wait for the second thread to exit. 
Unfortunately the `_thread` module does not provide any `.isAlive()` method or something similar to check the state of the thread. Using `_thread.interrupt_main()` seems to stop the main thread (core 0), but not the object created by `_thread.start_new_thread()` (core 1). 

This is not an issue in my realized project since I use the second thread to read certain digital IO as fast as possible and can exit early when the flag to run the loop is set `False`.

projectgus, 2024-09-10:
Glad you got it working again, and thanks for the link to the canonical flash_nuke source!

I agree that this should be in the docs in a more general place, so reopening the issue to track that.

PR #15956 (2024-11-19): docs: Add reference for reset and boot sequence, document factory reset for esp32 & rp2.
### Summary

This started out as a task to document `boot.py` and `main.py` on the rp2 port (Closes #15230). However, I realised the boot sequence and reset types were documented per-port and sometimes inconsistent so I decided to consolidate these into a common reference page that each port can link to.

In follow-up commits I added port-specific factory reset docs for esp32 and rp2 ports. These ports don't have any MicroPython-specific recovery features like Safe Mode boots, the factory reset process is  [... cut]
closed 2024-11-19 by dpgeorge