micropython/micropython #15236: round() returns incorrect results

tdunning, 2024-06-09:
### Port, board and/or hardware

Pico W

### MicroPython version

MicroPython v1.23.0-preview.322.g5114f2c1e on 2024-03-30; Raspberry Pi Pico W with RP2040

### Reproduction

1. This session shows the problem
```
>>> f = 7123.454
>>> f + 0.001
7123.455
>>> (f+0.001)/0.001
7123455.0
>>> round((f+0.001)/0.001)
7123454
>>> round((f+0.001)/0.001, 0)
7123454.0
>>> round((f+0.001)/0.001, 1)
7123455.0
>>> 
```
2. The result of the first two calls to `round()` should have been `7123455` and `7123455.0` respectively.
3. Python3 running on a Raspberry Pi4 gets the correct results.

### Expected behaviour

Should have given the right results as described.

### Observed behaviour

The value of `round((7123.454 + 0.001)/0.001)` should be `7123455`

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

tdunning, 2024-06-09:
Wow.

tdunning, 2024-06-09:
That makes a lot of code pretty hard to write.

tdunning, 2024-06-09:
Is there a work-around?

andrewleech, 2024-06-09:
These are standard limitations of float on any computer system, though embedded ones are where you're more likely to see single precision float these days. 

I'm not sure but you may be able to compile rp2 firmware with double float enabled however it's going to be _slow_ because that chip doesn't have hardware float support.

The other standard workaround did embedded programming is to stick to integers or use decimal fixed point arithmetic. Here's a library for that, but I haven't tried it myself so YMMV https://github.com/mpy-dev/micropython-decimal-number

tdunning, 2024-06-10:
Thanks for following up and thanks for the link!

scruss, 2024-06-10:
> I'm not sure but you may be able to compile rp2 firmware with double float enabled …

You can. It's very slow and uses lots of memory. I don't recommend it. 

With floats of any type, avoid addition and subtraction of numbers with vastly different magnitudes. 7123.454 (~= 7.12E+03) and 0.001 (~= 7.12E+03) are six orders of magnitude different, and single precision is very iffy at that range.

Compounded with that, there's a really good chance that there are no exact binary representations of either 7123.454 or 0.001, so you're likely not adding what you think you're adding:
```
>>> "%.12f" % (7123.454)
'7123.454101562500'
>>> "%.12f" % (0.001)
'0.001000000119'
```
This is not a bug, this is how binary floating point works

massimosala, 2024-06-10:
Hi @tdunning  Ted

It can be far worse!

I rewrite the _ntptime_ module because the default ones doesn't check the sanity of the server NTP responses.
And I hit this...

```
_POW32					= const(1 << 32)
ntp = sec + frac / _POW32
if abs(ntp - sec) > 1 :								# floating point error?
	raise ArithmeticError
```

sec and frac are both integers (uint32)

On some boards, their addition can lead to incorrect results... up to 5...10 minutes !!!