micropython/micropython #15244: callin a viper function inside another viper function

massimosala, 2024-06-10:
### Documentation URL

https://docs.micropython.org/en/v1.9.3/pyboard/reference/speed_python.html

### Description

Example code:

```
@micropython.viper
def func1(x: int) -> int:
    return x + 1

@micropython.viper
def func2(mp_func, viper_func, x: int) -> int:
    y = int(mp_func(x))
    y += viper_func(y)
    return y

from time import ticks_ms
print(func2, ticks_ms, func1, 1234)

```

It is clear I have to cast to int the result of the _mp_func_, that is an object.

Compiling to MPY this code, *mpy-cross* reports:

> ViperTypeError: local 'c' has type 'int' but source is 'object'

referred to this line:

`    y += viper_func(y)
`

Why?
*func1* is a *viper* type and returns and _int_ ...

If I fix it with:
`    y += int(viper_func(y))
`

Does *mpy-cross* recognize the return values is a viper _int_ or does it perform these operations:

- call viper_func
- transform the return value from _int_ to _object_
- due to the explicit cast, transform the _object_ to _int_

IMHO the compiler should recognize these scenario without an explicit cast.

### Code of Conduct

Yes, I agree

bixb922, 2024-06-11:
Each Viper function is treated as a separate entity. There are no cross-function optimizations. Viper variables are not objects and exist only within a viper function. So returning a viper int will always convert it into a MicroPython int object (a bulitins.int)

A detailed description of Viper mode was recently added here: https://github.com/micropython/micropython/wiki/Improving-performance-with-Viper-code

rkompass, 2024-06-11:
A feature request might perhaps be justified here.
Because as you (@massimosala and @bixb922) described, the intermediate transformation into objects and back is a relatively big performance loss of viper-viper function calls.

massimosala, 2024-06-13:
Yes, inside the *viper* _domain_ this boxing from int to object is unnecessary and hurts the benefits of using *viper*

I think MP could avoid this when

- calling *viper* functions from *viper* code
- calling *asm* functions from *viper* code 

How can I migrate this issue to a feature request?

massimosala, 2024-06-13:
Hi @bixb922

I read the doc. It is interesting but IMHO not clear.

For example, I read two sentences that are contradictory to me:

> Viper int variables allow values from -2**31 to 2**31-1

> x = 2**30        #  2**30 yields a builtins.int

bixb922, 2024-06-13:
I have tested that again and get the result stated in the document.

The underlying issue is that the ** (__pow__) operator is not supported for viper int. If a and b are of the viper int data type, the expression a**b will be rejected at compile time is ```ViperTypeError: binary op __pow__ not implemented````

The docs

bixb922, 2024-06-13:
I tested that again, and found it to be true. 

Viper int constant evaluations are done at compile time (that's nice!) and have some restrictions. You get a static compile time error if you mix builtins.int and viper int, so those restrictions get flagged at compile time and can be easily resolved by casting with the viper ```int```operator, just like you describe in the OP.

With respect to the OP: I don't think there is much benefit in doing a global viper optimization, and it is difficult to do, for example:
```
if some_expression:
    call_this = a_viper_function
else:
    call_this = a_non_viper_function
@micropython.viper
def my_viper_function():
    x:int = call_this()
````
Viper is fast because the compiler decides about viper data types at compile time. The ```if``` in the example above even could be in another module, or in C code, or a MPY code without source so it can be quite undecidable at compile time. I think that it is a big advantage that viper functions always behave like any other function, so they can be used in any context.   

On the other hand, the way I see it,  the return statement in the viper function is already optimized and compiled to machine code. Some measurements could decide if this has a merit. I could not find a relevant impact, except perhaps for extremely small functions (for example one statement functions) which probably should be incorporated inline in the calling viper function anyhow.

rkompass, 2024-06-14:
> How can I migrate this issue to a feature request?

@massimosala : You might edit the heading of this issue and perhaps the introductory post to indicate that this is now a feature request, or better mark this issue as closed and open a new issue as feature request.
I thing the labels are set by the maintainers, so you just indicate that in the heading. Refer to this closed issue also, perhaps.
As development currently proceeds, this will probably be ignored anyway (at least the impression will be there), but at some time Viper features will be added again, without much advertisement, as Damien follows his plan to perfect MP at all fronts :-), so just closing this probably also is justified (less noise/traffic).

massimosala, 2024-06-18:
IMHO at least for the simplest case

`var = result_of_viper_func()
`

there should be no boxing from _viper int_ to _object int_

And if we are assigning the result to a variable, the value assigned to the variable should be _viper int_

---

Eventually - but not mandatory - there will be a boxing if the same variable is used where it is expected a _object int_
NOT mandatory: it is fine if the compiler will emit  an error about mismatch of types.

andrewleech, 2024-06-20:
> IMHO at least for the simplest case ... there should be no boxing from viper int to object int

By the description written in https://github.com/micropython/micropython/wiki/Improving-performance-with-Viper-code#the-viper-data-types-int-uint-bool-ptr32-ptr16-and-ptr8

> Viper variables are "raw" variables and are not stored as MicroPython objects. In contrast the string, tuple, list and integer variables we all know are always stored as MicroPython objects.

Inside a viper function these viper variables are more like a C variable, ie "unboxed" by the terminology used above. 

By definition, _all_ python functions of any kind can only return a single python object (multiple return values is just a single tuple object really). 

If you want to be able to declare a "raw viper function" that does _not_ return an object, then it's no longer a python function at all, it would need to be something new entirely, with a new call / return structure in place to handle being able to return a specific type. 

This "raw viper function" would need to be able to declare what it's return type is - normal functions don't know/care about this, everything currently is "just an object".

A regular python function could not call this "raw viper function" at all, it could only be used from within a viper function and would probably need to have a new decorator created to declare it as such.

I'm not that familiar with the internals of how viper is written / works however I'm pretty certain this would be a significantly _large_ amount of work which on some platforms would gain little to no speed improvement at all in many cases...

python objects of some base types like int are _very_ cheap and fast on many ports where there are already shortcuts in place in the existing vm where things like a "small int object" is nothing more than a "C interger bit shifted slightly". https://github.com/micropython/micropython/blob/88513d12268f5a5c3f3488f1a2d70cb605129537/py/obj.h#L89

Keep in mind the cpu overhead of simply calling a function is _far_ larger than converting any viper variable to/from a python object. AKA if there's a measurable speed improvement from removing this type conversion, there would be a much larger improvement by inlining the function to remove the function call altogether - even if this meant duplicating the function contents everywhere you want to call it.

massimosala, 2024-06-20:
Hi @andrewleech Andrew

Many thanks for your attention and detailed post, I think it is interesting also for other users.

Andrew:
_This "raw viper function" would need to be able to declare what it's return type is - normal functions don't know/care about this, everything currently is "just an object"._

My hopes stem from this statement!
Declaring

> def f1() -> int :

there is a clear type hint for the result.
I thought it is used by *mpy-cross* to avoid boxing the result from _int_ to obj _

---

Andrew:
_by inlining the function to remove the function call altogether - even if this meant duplicating the function contents everywhere you want to call it._

On *esp8266* it isn't possible, the available RAM is about 500 bytes, very precious.

---

Limiting the scope to integers and taking into account your consideration - for _int_ the conversion is just a bit shift - I agree with you: for _int_ is a low priority optimization.

Thanks

closed 2024-06-20 by massimosala

andrewleech, 2024-06-20:
> _This "raw viper function" would need to be able to declare what it's return type is - normal functions don't know/care about this, everything currently is "just an object"._
>
> My hopes stem from [...]

That's a syntax that could be used when writing the function sure, but I was referring to the internal / C representation of a function inside micropython itself. It's not just the compiler that would need to know the return type but the runtime also. The function structure has no way to declare this, internally micropython functions of any kind can only return an object.

> ---
> 
> Andrew:
> _by inlining the function to remove the function call altogether - even if this meant duplicating the function contents everywhere you want to call it._
> 
> On *esp8266* it isn't possible, the available RAM is about 500 bytes, very precious.
> 

Well yes, speed vs ram vs flash size are often competing interests. There's a reason not many people actually use the 8266 any more, it's just too limited. 

This requested feature is essentially a speed request. Implementing it would make flash use grow as it's going to require extra code.

> ---
> 
> Limiting the scope to integers and taking into account your consideration - for _int_ the conversion is just a bit shift - I agree with you: for _int_ is a low priority optimization.
> 

If int was excluded... there's really not many more base or raw types that could be returned directly? Because float already does not have a raw form. A raw char really is just an int until you convert it to a python string object or buffer. What else would you want passed as raw?