micropython/micropython #15245: docs: viper and garbage collection

massimosala, 2024-06-10:
### Documentation URL

https://docs.micropython.org/en/v1.7/pyboard/reference/speed_python.html

### Description

Let say the garbage collector is enabled.

During the execution of functions decorated with *viper*, can the gc collector kick in or not?

Sorry I don't find this specification in the docs.

### Code of Conduct

Yes, I agree

andrewleech, 2024-06-10:
In my understanding gc.collect kicks in automatically when you try to allocated ram for an object and that allocation fails due to lack of space - then gc.collect will be run and the allocation retried.

viper functions can certainly allocate object / types that take ram so when that's done I'm sure gc.collect can be run.

peterhinch, 2024-06-11:
By the same token if a Viper function doesn't allocate then GC won't kick in.