micropython/micropython #15246: const declaration: constancy is only enforced within a module

massimosala, 2024-06-10:
@dpgeorge There are limits to this, though. Unlike a tuple, constancy is only enforced within a module. If you create a file `rats16.py` containing
```py
a = const(42)
```
and then import it, the value of `rats16.a` can be changed:
```py
>>> import rats16
>>> rats16.a
42
>>> rats16.a = 99
>>> rats16.a
99
>>> 
```
I'm not suggesting this is a bug: merely something of which people need to be aware.

_Originally posted by @peterhinch in https://github.com/micropython/micropython/issues/15115#issuecomment-2134886618_

massimosala, 2024-06-10:
For me this is a nasty and unexpected side effect.

1. Is there a specification in the docs?
2. Is it possible to improve MP and have real immutable values, across modules?

sosi-deadeye, 2024-06-10:
A hint could help, but this is the normal CPython behavior.

If you assign a new object to a name, then the old object is no longer referenced and will be garbage collected, if no other name is referencing the old object. Python does not have write protected constants.

You could make a helper class, to prevent overwriting accidentally the attribute.

```python

from micropython import const

class Constants:
    def __init__(self, **kwargs):
        for key, value in kwargs.items():
            setattr(self, f"_{key}", const(value))

    def __getattr__(self, key):
        return getattr(self, f"_{key}")

    def __setattr__(self, key, value):
        if not key.startswith("_"):
            raise RuntimeError("Not allowed to assign a new object to attribute.")

        super().__setattr__(f"_{key}", value)

constants = Constants(a=42, b=1337)

# RuntimeError
constants.a = 10
```

massimosala, 2024-06-10:
Hi @sosi-deadeye Andre

A few years ago I wrote a similar class... to extend dictionaries and allow to use the syntax _Dict.attribute_

Regarding immutable data types, this class is interesting but don't respect the _const_ promises about efficiency and ram usage.

IMHO MP has to be extended to have a complete _const_ implementation.
If Python doesn't have it, *it is a bug* and how is the motto "if everyone jumped off a cliff..."

andrewleech, 2024-06-10:
`const` is really only provided as a compiler hint for optimisation, it doesn't make promises of global immutability. 
https://docs.micropython.org/en/latest/library/micropython.html#micropython.const

If you'd like to write some more explicit notes in the docs above calling out functionality and limitations of const I'm sure that a PR would be welcome! I for one didn't even realise the compiler threw a syntaxerror at you if you tried to modify a const inside a file, this is more protection than I expected!

When comparing with const in other (statically typed) languages it's easy to see how asusmptions and confusions may arise, however (micro)python is a dynamic language first and foremost, so no one programming with it should assume the language / compiler / vm will protect you from yourself.

I'm not that familiar with the vm internals myself, but currently const is _only_ really implemented in the compiler. To protect against changes to variables globally / from other modules some extensive changes would likely need to be made to the structure of _modules_ and the _vm_ to provide support for immutable objects. This would cost likely flash / ram / runtime cpu to implement so is _very_ different to the current implementation that's just used in the compiler.

In my opinion if we wanted extra checks and balances against overwriting a const from outside a module, this should be the job of a linter / mypy which could be done without extensions to the vm. That being said I don't know if it's possible to write extensions to mypy to add support for micropython-only functions like const.

sosi-deadeye, 2024-06-11:
The function `micropython.const` is ignored if the program is not compiled with `mpy-cross`?

I like the linter way. Then Micropython could introduce `Const`: https://github.com/python/typing/issues/242
During runtime, `TypeHints` do nothing. But tools like `mypy` do a static analysis of the code and points to errors.

andrewleech, 2024-06-11:
> The function `micropython.const` is ignored if the program is not compiled with `mpy-cross`?

No it's also handled by the compiler on-device / in micropython itself when running a `.py` file. I just mean it's only processed during the per-module compile stage, not during the "whole program" execute. The diagrams here might give some idea what I mean: https://github.com/orgs/micropython/discussions/11417

andrewleech, 2024-06-11:
@peterhinch my reading is this means it wouldn't just be an immutable int like immutable string - it's also the variable name that's requested to be immutable. 
Aka disable setattr for variables that are declared const / moving the const variable into a fixed globals table.

Josverl, 2025-04-09:
> I like the linter way. Then MicroPython could introduce `Const`: [python/typing#242](https://github.com/python/typing/issues/242)
> During runtime, `TypeHints` do nothing. But tools like `mypy` do a static analysis of the code and points to errors.

I [checked](https://github.com/python/typing/discussions/1848) with the static typechecker community if it would be possible to typehint `the result of const()` as `Final()`.
That is not possible, and considered to a be  an edge case, so will not be implemented.

What you can do is 
```py
#fo.py
from typing import Final

FOO :Final = const(42)
```
```py
# bar.py
```
from foo import FOO
FOO = 100  # Error: "FOO" is declared as Final and cannot be reassigned

import foo
foo.FOO = 100 # depends on type-checker
```
**Checking with pyright and mypy (with micropython-stubs) :** 
> pyright .\src\bar.py
d:\mypython\TTGO_1.8\src\bar.py
  d:\mypython\TTGO_1.8\src\bar.py:2:1 - error: "FOO" is declared as Final and cannot be reassigned (reportGeneralTypeIssues)
1 error, 0 warnings, 0 informations 

> mypy .\src\bar.py
src\bar.py:2: error: Cannot assign to final name "FOO"  [misc]
src\bar.py:6: error: Cannot assign to final name "FOO"  [misc]
Found 3 errors in 2 files (checked 1 source file)
```