micropython/micropython #15258: docs:

tdunning, 2024-06-11:
### Description

Python normally uses double precision for all (floating point operations)[https://docs.python.org/3/library/numbers.html#numbers.Real]. MicroPython, depending on the platform, usually does not. This leads to surprises and errors, occasionally quite serious. 

Unfortunately, this doesn't seem to be documented anywhere except in discussion groups.

I will create a pull request which attempts to deal with this lack.

### Code of Conduct

Yes, I agree

massimosala, 2024-06-13:
I agree with your rational thinking.

It is a pain to port code using floats, for example, from esp32 to esp8266 and rp2...

tdunning, 2024-06-13:
Well, there is a limit to what can be accomplished in trying to actually fix the problem since Python only supports a single floating point precision. Obviously, the real solution would be to have a flexible type system like Julia where you can have multiple precisions of float (16, 32 and 64 out of the box) and can define more that work at hardware performance. But that wouldn't be Python as we currently know it. And it probably wouldn't be very micro.

So, in advance of Python 57.0's release, I think that we just need some documentation.