micropython/micropython #15260: esp32.wake_on_ext0 not workingon esp32s2

kjm1102, 2024-06-12:
### Port, board and/or hardware

esp32 port, waveshare ESP32-S2-Pico

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Generic ESP32S2 module with ESP32S2

### Reproduction

import os,time,esp32
from machine import Pin,deepsleep

time.sleep(5)
pb=Pin(0, Pin.IN); esp32.wake_on_ext0(pb,0)
deepsleep(10*1000)

### Expected behaviour

Loaded the file as main.py then reset the board. Expected s2 to either deepsleep for 10s or exit deepsleep when I press the button connected to gpio0.

### Observed behaviour

The s2 exits deepsleep immediately after entering it. This can be verified with a usb ammeter in series with the usb powering the board. I did not include print commands because having a usb repl active often plays badly with deepsleep.

### Additional Information

 esp32.wake_on_ext0 works OK with deepsleep on an esp32 v1.23.0 so I figure the instant wakeup on the s2 is down to it being an s2.

### Code of Conduct

Yes, I agree

kjm1102, 2024-06-21:
Putting hold=1 in the pin definition for an s2 gpio0 stops the pullup switching to pulldown on deepsleep, which is what causes the immediate wakeup. Not required on regular esp32 gpio0 but is on the s2, see https://github.com/orgs/micropython/discussions/15234#discussioncomment-9751258

projectgus, 2024-07-02:
Thanks for posting the update, @kjm1102 . 

I think the fix for us to do here is probably to document the `hold` parameter in the esp32 module wakeup functions. You're right that it's not needed on esp32, some more hardware features were added for the later chips. However as far as I can see the Python features aren't documented outside of the code.

Does that sound right?

kjm1102, 2024-07-02:
Anything that saves poor plods like me from hours of discovery learning. Something in the docs that makes it obvious that if you want to use gpio0 for deepsleep wakeup on an s2 you need to put hold=1 in your gpio0 input pin definition.

jonnor, 2024-09-01:
Merge request for docs welcomed! They are in the git repository, in the docs/ folder.