micropython/micropython #15269: ISR latency affected by BLE, unable to interface with 2kHz sample rate ADC.

vladkvit, 2024-06-12:
### Port, board and/or hardware

ESP32-S3

### MicroPython version

`MicroPython v1.23.0 on 2024-06-02; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3`

### Reproduction

I wrote a small test program to reproduce. In this program, I set one pin high while I send a BT packet. I also set a 20kHz timer with an interrupt where I pull a pin high then low. timer When I probe the two test pins with an oscilloscope, I notice that the 20kHz timer ISR fails to trigger right after the BT send. In the image below, the top row is the BT send pin, and the bottom is the 20kHz ISR pin. Note how there's that weird gap near the red lines.

[image]

```
import micropython
import asyncio

import aiorepl
import aioble

from time import ticks_us

from machine import Pin
from machine import Timer

import bluetooth

# For exceptions in interrupt subroutines
micropython.alloc_emergency_exception_buf(100)

test_value = 0

# BT setup
GATT_UUID = "079b8c5b-2961-4287-ac4a-9c8bc70f7a6c"
_ENV_SENSE_UUID = bluetooth.UUID(GATT_UUID)
_ENV_SENSE_TEMP_UUID = bluetooth.UUID(GATT_UUID)
_ADV_APPEARANCE_GENERIC_THERMOMETER = const(768)
_ADV_INTERVAL_MS = 250_000

# Register GATT server.
temp_service = aioble.Service(_ENV_SENSE_UUID)
temp_characteristic = aioble.Characteristic(
    temp_service, _ENV_SENSE_TEMP_UUID, read=True, notify=True
)
aioble.register_services(temp_service)

debug_pin1 = Pin(1, Pin.OUT)
debug_pin2 = Pin(2, Pin.OUT)

def timer_isr(timer):
    debug_pin2.on()
    test_value = ticks_us() # give MP something to do
    debug_pin2.off()

tim = Timer(0)
tim.init(freq=20000, callback = timer_isr)

async def timer_task_f(event):
    while True:
        await asyncio.sleep_ms(4)
        event.set()

async def sensor_task(event):
    my_buf = bytearray(100+4)
    
    while True:
        debug_pin1.on()
        temp_characteristic.write(my_buf, send_update=True)
        debug_pin1.off()

        await event.wait()
        event.clear()

async def peripheral_task():
    while True:
        async with await aioble.advertise(
            _ADV_INTERVAL_MS,
            name="mpy-temp",
            services=[_ENV_SENSE_UUID],
            appearance=_ADV_APPEARANCE_GENERIC_THERMOMETER,
        ) as connection:
            print("Connection from", connection.device)
            await connection.disconnected(timeout_ms=None)

async def main():
    repl = asyncio.create_task(aiorepl.task())

    timer_event = asyncio.Event()
    timer_task = asyncio.create_task(timer_task_f(timer_event))
    t1 = asyncio.create_task(sensor_task(timer_event))
    t2 = asyncio.create_task(peripheral_task())
    await asyncio.gather(repl, t1, t2, timer_task)
        
asyncio.run(main())
```

### Expected behaviour

I'm writing firmware that reads from an ADC at 2kHz and sends the data over Bluetooth at 200Hz. I noticed that the ADC read interrupts are failing to trigger right after a BT send. This makes me unable to maintain 2k samples / second from the ADC.

### Observed behaviour

See screenshot above. The ISR is not triggered on time.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

massimosala, 2024-06-13:
Hi @vladkvit  Vlad

Are you correct?
> ADC @ 2 KHz
> Bluetooth @ 200 Hz

This will never work.

---

You are using timers: hardware or software ones?

---

Have you thought about a ring buffer where you insert the samples and another routine (not the timer callable) where you send then on BT ?

vladkvit, 2024-06-13:
Hi @massimosala massimosala

Sorry, I should have been more clear. Having _any_ BT communication (even just sending dummy packets at whatever rate) prevents me from reading an external ADC at 2kHz. My ADC pulls down a pin every ~0.5ms; sending a BT packet will delay ISRs by more than 0.5ms.

> Have you thought about a ring buffer where you insert the samples and another routine (not the timer callable) where you send them on BT ?

That was my original implementation, but that doesn't work for the reason I'm describing - I'm missing ADC samples.

> You are using timers: hardware or software ones?

For BT, I use async.sleep just so that I get some packets going.

For the ADC, I was at first using a pin interrupt, then I wanted to debug so I use a callback from `from machine import Timer` and don't even do anything with an ADC. Both methods (pin interrupt, timer callback) are delayed after sending any BT packets.

Here's a more annotated example of what I'm talking about. I toggle a pin when sending a BT packet, and I toggle another pin when inside a timer callback. There is no ADC involved below, it's there to demonstrate the >0.5ms gap with no interrupts.

[image]