micropython/micropython #15270: Include a WASM interpreter as part of the Micropython standard library / standard distribution?

huyng, 2024-06-13:
### Description

What are your thoughts on adding a WASM interpreter to Micropython's standard library / standard distribution?

The main benefit would be that I or anyone could for example write a library in C, C++, Zig, Rust, or Nim; compile it to WASM, and then run that same code in MicroPython through the WASM interpreter. Given a sufficiently optimized interpreter, one could get near the same performance as native C code. 

Since many languages already target WASM as a compilation target, this would allow library developers to build for a single target while gaining the ability to distribute and run their code on many different microcontrollers. This is in contrast to the current process where one needs to compile for each microcontroller target if you want your library to run across many devices. 

[pywasm](https://github.com/mohanson/pywasm/tree/master) for example, is one such WASM  interpreter written in pure Python. You can see an example of how WASM code compiled from a simple C function is loaded and ran [here](https://github.com/mohanson/pywasm/blob/master/examples/sum.py).

Here is a WASM interpreter written in  [C](https://github.com/andrewrk/zig-wasi/blob/main/src/main.c)  and one in [Zig](https://github.com/andrewrk/zig-wasi/blob/main/src/main.zig).

I assume the C implementation would give the most optimal performance.  But you could always include the pure-python wasm interpreter first into micropython, and then later on replace it with the C implementation.

### Code Size

I am unsure at this point how much a WASM interpreter would add to the Micropython runtime and would need to do further investigation.

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

massimosala, 2024-06-13:
Hi Huy

I too consider this a nice feature... but not a core one.

Some random thoughts about new modules... specifically modules built in in the firmwares:

- they increase the base firmware size
- conditional enabled? the common man doesn't recompile the MP firmware -> MP loses users
- if there are bugs in the module, all the firmware will be tainted as bugged

Sorry for all this fuss, but reading other open issues... I'm more and more convinced of the mistake of having non-core modules within the firmware. Even optionals.

huyng, 2024-06-13:
I looked into memory and storage requirements of a few available WASM interpreters that are written in C and could be embedded inside of micropython:

- [wasm3](https://github.com/wasm3/wasm3): ~64Kb for library size, ~10Kb RAM to run the runtime.
- [wasm micro runtime (wamr)]: ~59KB for library size, I do not know ram requirements.

huyng, 2024-06-13:
Another approach is to to build an easily installable `mpremote mip` module that one could distribute. However if the wasm interpreter we want to use must be compiled to native code, the final end user would need to somehow get the correct version of the compiled binary for the correct microcontroller architecture. Is this something the current `mpremote mip` command would be able to figure out for the user depending on which microcontroller interface they are installing into?  This is one of the drawbacks with this approach.

Ideally if the wasm runtime was included at the micropython firmware level, the users would not need to think about it and can focus solely on developing their performance critical code and compiling to a single WASM target.

jonnor, 2024-07-27:
`mip` can currently not install native modules. So users have to copy in the correct .mpy for their architecture and MicroPython version. It is not super smooth, but it works. Extending `mip` to cover this usecase would be great, and be beneficial across the ecosystem. I would love to use that for https://github.com/emlearn/emlearn-micropython, for example.

Doing a third party module (C user module or native dynamic) is probably the best way to start on a WebAssembly interpreter. To demonstrate the functionality, performance, usefulness etc.

vshymanskyy, 2024-08-10:
> Given a sufficiently optimized interpreter, one could get near the same performance as native C code.

No.
- Even the best AOT/JIT engines do not reach the native performance yet. WASM is always at least ~10-20% slower.
- Wasm will inevitably require more RAM to run (also consider that in embedded systems most of the code runs from flash)

> write a library in C, C++, Zig, Rust, or Nim; compile it to WASM, and then run that same code in MicroPython

This is nice (and achievable), but I think there's a way to implement it [differently](https://github.com/wasm3/embedded-wasm-apps). It's a cheap way to provide AOT modules in a **very** portable way. Yes, this sacrifices the dynamic nature of wasm modules, but hey, we can do OTA to update the system, right?

---
Overall, I don't think integrating any dynamic WASM runtime is in the scope of MicroPython, but should rather be a third party framework that incorporates many components.

---
Disclaimer: I'm the author of `wasm3`

vshymanskyy, 2024-08-22:
> If C99 could be compiled to `mpy` (native code only)...

Wow. Looks like it's available in MicroPython out of the box:
https://docs.micropython.org/en/latest/develop/natmod.html

vshymanskyy, 2024-08-24:
Here is a Proof of Concept: https://github.com/orgs/micropython/discussions/15702

vshymanskyy, 2024-08-28:
@dpgeorge I think we can close this ticket

closed 2024-09-10 by projectgus