micropython/micropython #15272: Renesas-RA: Pin-IRQ does not behave as expected.

robert-hh, 2024-06-13:
### Port, board and/or hardware

Rensas-RA, Board RA6M2

### MicroPython version

MicroPython v1.24.0-preview.36.gdc9e733ad on 2024-06-12; EK-RA6M2 with RA6M2

### Reproduction

```
from machine import Pin, idle

p5=Pin("P005", Pin.IN)
p6=Pin("P006", Pin.OUT, value=0)

def cb(pin):
    global p6
    p6(1)
    p6(0)
    
p5.irq(handler=cb, trigger=Pin.IRQ_FALLING)

while True:
    idle()
```

### Expected behaviour

Response pulse at Pin P006.

### Observed behaviour

No pulse. If I call the callback directly, I get a pulse.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

TakeoTakahashi2020, 2024-06-13:
@robert-hh san, Thank you for your report. I will check.

TakeoTakahashi2020, 2024-06-14:
@robert-hh san, I have made a PR #15283 to fix this issue.
I appreciate it if you could check it.  Thanks.

robert-hh, 2024-06-14:
Thank you. I will try it later today.

robert-hh, 2024-06-14:
@TakeoTakahashi2020 Pin.irq() works now in my test as expected. It triggers at and does so the right time.
Thank you for the fix.

TakeoTakahashi2020, 2024-06-17:
@robert-hh san, Thank you very much for your check.

dpgeorge, 2024-09-26:
Fixed by 6381ad5d0c4c594dbb8feea3bebf6222fd78f3a1

closed 2024-09-26 by dpgeorge