micropython/micropython #15274: MQTTS fail on  ESP32-S3 but works in ESP32

littlegreenbean33, 2024-06-13:
### Port, board and/or hardware

esp32 generaic and SeeedStudio edgebox-100

### MicroPython version

 ESP32_GENERIC v1.22.2.bin

ESP32_GENERIC_S3 v1.22.2.bin

### Reproduction

```python
    ssl_params5={'cert_reqs':ussl.CERT_NONE,'key':data,'cert':cert, 'server_hostname':mqtt_server,
             'cadata':ca_certs}
    client = MQTTClient(client_id,mqtt_server,port=mqtt_port,ssl=True,
             keepalive=server_keepalive,ssl_params=ssl_params5)

    printl("Setting callback function")
    client.set_callback(sub_cb)
    printl("Connecting to MQTTS")
    client.connect()
    printl("Connected to MQTTS")

```

[image]

### Expected behaviour

Returned values as with ESP32

### Observed behaviour

As soon as connect is issued, we can see the error in the picture.
We can see in the firewall that conectiond are dropped due to aged_out and a FIN packet is returned to the esp32. At which point the error can be seenin the serial console.

It is like the lwip stack was different in the ESP32-S3 implementation and the regular ESP32. I can't figure out other issues.

### Additional Information

Firewall is PaoloAlto.

### Code of Conduct

Yes, I agree

jonnor, 2024-07-27:
Hi and thank you for this report. Have you tried this on the latest MicroPython - either release 1.23.0 or one of the nighly builds (can be found on the download page)?

littlegreenbean33, 2024-07-30:
@jonnor  Apologies for late return. Holidays period.

We stick to 1.22.2 due to changes in ssl/tls. We eventually got to access the logs of greengrass and the issue was that the connection was not being fired with HTPPS but HTTP.....not sure if it is a proxy thing or what...

I'll close this one. I presume I am missing something in our environment.

Thanks for reaching out.

JC

closed 2024-07-30 by littlegreenbean33