micropython/micropython #15275: Cannot instantiate FrameBuffer from a bytes object

peterhinch, 2024-06-13:
### Port, board and/or hardware

RP2, Pyboard 1.1

### MicroPython version

MicroPython v1.22.1 on 2024-01-05; Raspberry Pi Pico with RP2040

### Reproduction

Enter the following at the REPL:
```py
from framebuf import FrameBuffer, GS8
b = b'\x00' * 100
f = FrameBuffer(b, 10, 10, GS8)
```

### Expected behaviour

I would expect the instantiation of the `FrameBuffer` to succeed (as it does if a `bytearray` is passed). 

### Observed behaviour

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: object with buffer protocol required
```

### Additional Information

Use case:
Instantiating from a `bytes` object (or a `memoryview` of `bytes`) would enable graphical objects frozen as bytecode to be blitted onto an existing frame buffer.

A further observation: the error message is most misleading.

### Code of Conduct

Yes, I agree

massimosala, 2024-06-13:
And from the
[Python docs](https://docs.python.org/3/c-api/buffer.html)
 it seems _bytes_ should implement the buffer protocol:

massimosala, 2024-06-13:
This doesn't give errors... if you wanna try it and report back:

> f = FrameBuffer(bytearray(b), 10, 10, GS8)

peterhinch, 2024-06-14:
@massimosala Perhaps my report wasn't clear enough.

See "Expected behaviour" above: a `bytearray` does work.

A `bytes` object does support the buffer protocol. That is why I described the error message as misleading.

Gadgetoid, 2024-06-14:
Looks like `bytes` fails the inside check in `mp_get_buffer`:

```c
MP_OBJ_TYPE_GET_SLOT(type, buffer)(obj, bufinfo, flags & MP_BUFFER_RW) == 0
```

Because it's effectively a string, and you can't write to a string:

```c
mp_int_t mp_obj_str_get_buffer(mp_obj_t self_in, mp_buffer_info_t *bufinfo, mp_uint_t flags) {
    if (flags == MP_BUFFER_READ) {
        GET_STR_DATA_LEN(self_in, str_data, str_len);
        bufinfo->buf = (void *)str_data;
        bufinfo->len = str_len;
        bufinfo->typecode = 'B'; // bytes should be unsigned, so should unicode byte-access
        return 0;
    } else {
        // can't write to a string  <--- YE CANNAE WRITE TO A STRING!
        return 1;
    }
}
```

Is the fix a discrete `mp_obj_*bytes*_get_buffer`?

peterhinch, 2024-06-14:
The solution appears to be to change https://github.com/micropython/micropython/blob/d7d77d91becd0716ac1672c92652d9dd72ec613f/extmod/modframebuf.c#L315

to read
```C
    mp_get_buffer_raise(args_in[0], &bufinfo, MP_BUFFER_READ);
```
I'm not too confident about the C code, but I have tested this and it works with a variety of applications including where the buffer is a `bytes` object stored as frozen bytecode.

Gadgetoid, 2024-06-14:
Are there instances where a FrameBuffer would *need* to write its underlying buffer? Or is this buffer a one time, read-only, copied-into-the-FrameBuffer’s-own-memory thing? Despite not-inventing-here my own alternate, I haven’t really delved into what it is and how it works 😬

I’ll have a further poke if I get a moment, I’m keen to pick up a more general knowledge of MicroPython internals!

Gadgetoid, 2024-06-14:
Okay since the FrameBuffer instance is instantiated with the buffer-type object as its memory:

https://github.com/micropython/micropython/blob/d7d77d91becd0716ac1672c92652d9dd72ec613f/extmod/modframebuf.c#L323

Then it necessarily *must* be writable, or all attempts to mutate the FrameBuffer - ie: primitive drawing operations - would fail.

There is no such thing as a read-only FrameBuffer.

There is a bug here, though, when given a read-only buffer, the FrameBuffer constructor should raise an appropriate "buffer must be writable" or similar error. Indeed the error message is misleading and should be changed.

But for your use case, MicroPython would need to:

* Allow for read-only FrameBuffer somehow (either raise exceptions on write, or return a class instance with an alternate locals dict that stubs out write functions with a catch-all not-implemented error)
* Or somehow make `blit` understand raw bytes as a blittable surface.
* Or provide some other method (`raw_blit`?) to handle blitting raw bytes

For solving your specific use-case I tend to prefer the latter. It's fast. It's simple and it's relatively unobtrusive. But the former might be less alien and more consistent to existing FrameBuffer users.

peterhinch, 2024-06-15:
> Then it necessarily must be writable, or all attempts to mutate the FrameBuffer - ie: primitive drawing operations - would fail. There is no such thing as a read-only FrameBuffer.

I think you are misunderstanding my use case. I have a means of converting an image (jpeg or other) to a Python source file. This creates a `bytes` object containing a bitmap of the image in one of the `FrameBuffer` color maps. The user has the option to freeze the Python file. To display the image the application instantiates a `FrameBuffer` using the `bytes` object in the Python file. This `FrameBuffer` is read-only, because its underlying buffer is a `bytes` object. The `FrameBuffer.blit` method is used to render the read-only `FrameBuffer` onto the display driver (which is subclassed from a conventional r/w buffer).

The modification posted above works, both in the proposed usage with a frozen Python source file and in normal graphics tests where the `FrameBuffer` graphics primitives are used to draw to the display driver.

I have tested writing to a read-only `FrameBuffer`: there is no effect - no exception is raised and `blit` renders the image normally. I'm not sure how (or if) this should be trapped.

Gadgetoid, 2024-06-16:
Yes after working on graphics and asset pipelines for 32blit and PicoSystem and then creating PicoGraphics and blazing the trail with embedded vector graphics I have somehow misunderstood a simple use case 😐🤣 please give me _some_ credit, I’ll let you know when I don’t understand.

I’d probably want to do a little more digging when writing to something I expect to be read only. Where are those writes going, because to the best of my knowledge there are no checks and balances once you get past FrameBuffer’s constructor 😬

Just disabling the checks probably means you’re writing… uh… somewhere!? I guess the asset address would be in flash which is explicitly not directly writable - I’d have thought it would result in a hardfault.

The point of raising *some* kind of error is to avoid this question, and additionally avoid users falling under the assumption they can write something that’s not writable and wasting time debugging it.

peterhinch, 2025-10-29:
If `glyph` is a memoryview of a bytes object you can convert it to a bytearray with
```py
ba = bytearray(glyph)
```