micropython/micropython #15282: mpremote cannot cat with recent MP on pyboard

rkompass, 2024-06-14:
### Port, board and/or hardware

pyboard

### MicroPython version

MicroPython v1.24.0-preview.36.gd7d77d91b on 2024-06-12; PYBv1.1 with STM32F405RG

### Reproduction

`mpremote cat main.py`

### Observed behaviour

```
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
UnicodeError: 
```

### Additional Information

This error is also observed with MP version 1.22.2 .

Perhaps it is related to this error message (RuntimeError below) at boot up ? :

`mpremote`

```
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] or Ctrl-x to exit this shell
RuntimeError: name too long
MicroPython v1.24.0-preview.36.gd7d77d91b on 2024-06-12; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> 
```

### Code of Conduct

Yes, I agree

massimosala, 2024-06-17:
Hi @rkompass  Raul

BTW does your source code use utf8 characters of non European countries? like Chinese or Japanese?

rkompass, 2024-06-18:
Hi Massimo (@massimosala),

it is exaclty as you assumed. If I copy an older version (essentially empty) of mani.py over to the board then afterwards the 
`mpremote cat main.py` command works.

Same with the other 3 files: boot.py, pybcdc.inf and readme.txt.

So all these 4 files have a strange coding and essentially this is the issue now,
as they appear so after flashing the official binary.

Do you have an idea of the origin of this issue?

massimosala, 2024-06-18:
Hi Raul @rkompass 

What do you mean with

_So all these 4 files have a strange coding and essentially this is the issue now,_

---

Raul:
_Do you have an idea of the origin of this issue?_

I was intrigued by this other issue
[Unable to print unicode characters when running repl with mpremote](https://github.com/micropython/micropython/issues/15228)

because 3 years ago I have to dig into console behaviour, both linux and windows, with utf8 text.

IMHO mpremote was written thinking about pure _ascii_ charset, so reading/writing a byte at a time disrupt multi bytes characters encodings (like utf8).

Pls read my comment to the issue 15228 and let me know.
Ciao

rkompass, 2024-06-20:
I close this. A reformat of the file system solves the issue.
I will look at the other linked issue later.