micropython/micropython #15286: asyncio module seems missing in self built firmware for ESP8266 board

jonesthefox, 2024-06-14:
### Port, board and/or hardware

ESP8266, ESP8266_GENERIC

### MicroPython version

MicroPython v1.24.0-preview.36.gd7d77d91b on 2024-06-14; ESP module with ESP8266

### Reproduction

Compare ports/ESP8266/manifest.py with ports/ESP32/manifest.py

### Expected behaviour

```
>>> import asyncio
>>>
```

### Observed behaviour

Got this:
```
>>> import asyncio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'asyncio'
```

### Additional Information

`include("$(MPY_DIR)/extmod/asyncio")` seems missing in ESP8266/manifest.py

I added the line to my own manifest.py which i include in build with `FROZEN_MANIFEST=` and then the asyncio module is included in my firmware.

### Code of Conduct

Yes, I agree

ricksorensen, 2024-06-15:
Intentiional to save resources.

[https://github.com/micropython/micropython/blob/master/ports/esp8266/boards/ESP8266_GENERIC/board.md](https://github.com/micropython/micropython/blob/master/ports/esp8266/boards/ESP8266_GENERIC/board.md)

peterhinch, 2024-06-15:
To clarify, `asyncio` is only missing on reduced flash variants of ESP8266. Versions with 2MB and above should include it - see [the manifest](https://github.com/micropython/micropython/blob/master/ports/esp8266/boards/ESP8266_GENERIC/manifest_2MiB.py).

jonesthefox, 2024-06-15:
@peterhinch Saw that, but im actually using the 4MB variant..

peterhinch, 2024-06-16:
The approach I'd use is first to check whether pre-built firmware for your size of board has `asyncio`. If not there is a definite issue.

Assuming the pre-built firmware still has it, then the problem may be in your manifest files.

jonesthefox, 2024-06-16:
@peterhinch works with pre-built firmware, so i'm closing this now in eternal shame! 😅

closed 2024-06-16 by jonesthefox

jonesthefox, 2024-06-16:
To whom it may concern:
I experimented with building ESP32_GENERIC firmware, where i had `include("$(PORT_DIR)/boards/manifest.py")` in my manifest, so i guessed it may work the same with esp8266. 

However, with the esp8266 i had to add `include("$(PORT_DIR)/boards/ESP8266_GENERIC/manifest_2MiB.py")`.

@peterhinch thanks for pointing me in the right direction!