micropython/micropython #15287: nimble compile errors

kdschlosser, 2024-06-15:
### Port, board and/or hardware

ESP32-S3

### MicroPython version

1.23.0

### Reproduction

I am simply compiling for the ESP32-S3 and the errors happen.

I am using ESP-IDF 5.2.1. In the release notes for MicroPython 1.23.0 it states that support for ESP-IDF 5.2 has been added. 

### Expected behaviour

For it to compile properly. 

### Observed behaviour

```
from /micropython/extmod/nimble/modbluetooth_nimble.c:28:
/micropython/extmod/nimble/modbluetooth_nimble.c: In function 'ble_secret_store_read':
/micropython/extmod/nimble/modbluetooth_nimble.c:1914:33: error: 'const struct ble_store_key_sec' has no member named 'ediv_rand_present'
1914 |                 assert(!key->sec.ediv_rand_present);
|                                 ^
/micropython/extmod/nimble/modbluetooth_nimble.c:1920:33: error: 'const struct ble_store_key_sec' has no member named 'ediv_rand_present'
1920 |                 assert(!key->sec.ediv_rand_present);
|                                 ^
/micropython/extmod/nimble/modbluetooth_nimble.c:1931:28: error: 'const struct ble_store_key_sec' has no member named 'ediv_rand_present'
1931 |             assert(key->sec.ediv_rand_present);
|                            ^
/micropython/extmod/nimble/modbluetooth_nimble.c: In function 'ble_secret_store_write':
/micropython/extmod/nimble/modbluetooth_nimble.c:1980:27: error: 'struct ble_store_key_sec' has no member named 'ediv_rand_present'
1980 |             assert(key_sec.ediv_rand_present);
|                           ^
```

### Additional Information

I know that nimble is included in the ESP-IDF and I know that nimble that is added as a submodule to MicroPython is not downloaded with the ESP32 build so I am assuming that the nimble that is included with the ESP-IDF is what ends up getting used. When I check the `ble_store_key_sec` structure in that nimble the code below is what is seen. 

```c
struct ble_store_key_sec {
    /**
     * Key by peer identity address;
     * peer_addr=BLE_ADDR_NONE means don't key off peer.
     */
    ble_addr_t peer_addr;

    /** Number of results to skip; 0 means retrieve the first match. */
    uint8_t idx;
};
```

You can clearly see there is no `ediv_rand_present` field. I am thinking that the nimble that is added as a submodule to MicroPython is what needs to be used but I think that it will cause issues with the one that is included in the ESP-IDF if it is added to the ESP32 build.

### Code of Conduct

Yes, I agree

kdschlosser, 2024-06-15:
I do believe I might have this issue sorted out. There is a version naming lack of clarity. In the esp32 port configuration header file there is an explicit check for espidf version 5.2.0 but when a person tries to do a checkout of the esp-idf for `v5.2.0` it says there is no such release. Upon further digging the release tag that the folks over at espressif have decided to use doesn't conform to the `{major}.{minor}.{micro}` version naming. the version that needs to be checked out is `v5.2`.

I am not 100% sure if this has fixed the issue by checking out the correct version because I am faced with a bunch of linking errors now and the build doesn't complete. I opened #15289 for that. So until those linking errors are solved and the build is able to complete I will not know if this issue is actually solved.

kdschlosser, 2024-06-16:
This is solved. 5.2.0 does have the field in the nimble supplied with the IDF.

closed 2024-06-16 by kdschlosser

brainstorm, 2024-07-06:
I'd encourage revisiting (reopening?) this issue since I am seeing the same errors with **esp-idf 5.4.0 (master)** while using `micropython-builder` with command `./scripts/esp32/generic-s3.sh`.

I suspect that `sdkconfig.combined.in` changes such as `CONFIG_BT_NIMBLE_TASK_STACK_SIZE` for `CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE` (and others) might be partially responsible for the errors?

```
(...)
-- Building ESP-IDF components for target esp32s3
Processing 4 dependencies:
[4/4] idf (5.4.0)
-- Project sdkconfig file /Users/rvalls/dev/personal/micropython-builder/micropython/ports/esp32/build-ESP32_GENERIC_S3/sdkconfig
Loading defaults file /Users/rvalls/dev/personal/micropython-builder/micropython/ports/esp32/build-ESP32_GENERIC_S3/sdkconfig.combined...
/Users/rvalls/dev/personal/micropython-builder/micropython/ports/esp32/build-ESP32_GENERIC_S3/sdkconfig.combined:100 CONFIG_ESP32_WIFI_IRAM_OPT was replaced with CONFIG_ESP_WIFI_IRAM_OPT
/Users/rvalls/dev/personal/micropython-builder/micropython/ports/esp32/build-ESP32_GENERIC_S3/sdkconfig.combined:101 CONFIG_ESP32_WIFI_RX_IRAM_OPT was replaced with CONFIG_ESP_WIFI_RX_IRAM_OPT
/Users/rvalls/dev/personal/micropython-builder/micropython/ports/esp32/build-ESP32_GENERIC_S3/sdkconfig.combined:145 CONFIG_BT_NIMBLE_TASK_STACK_SIZE was replaced with CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE
/var/folders/s8/0tz3dy0d1sbg1_cv0ylkx_m80000gn/T/kconfgen_tmpsem__9dw:164 line was updated to CONFIG_ESPTOOLPY_FLASHSIZE_4MB=n
/var/folders/s8/0tz3dy0d1sbg1_cv0ylkx_m80000gn/T/kconfgen_tmpsem__9dw:166 line was updated to CONFIG_ESPTOOLPY_FLASHSIZE_16MB=n
warning: unknown kconfig symbol 'XTAL_FREQ_AUTO' assigned to 'y' in /Users/rvalls/dev/personal/micropython-builder/micropython/ports/esp32/build-ESP32_GENERIC_S3/sdkconfig.combined
-- Compiler supported targets: xtensa-esp-elf
-- App "micropython" version: 4d16a9c
-- Adding linker script /Users/rvalls/dev/personal/micropython-builder/micropython/ports/esp32/build-ESP32_GENERIC_S3/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/rvalls/dev/personal/micropython-builder/micropython/ports/esp32/build-ESP32_GENERIC_S3/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
Including User C Module(s) from ../../../../ulab/code/micropython.cmake
Found User C Module(s): usermod_ulab
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_driver_ana_cmpr esp_driver_cam esp_driver_dac esp_driver_gpio esp_driver_gptimer esp_driver_i2c esp_driver_i2s esp_driver_isp esp_driver_jpeg esp_driver_ledc esp_driver_mcpwm esp_driver_parlio esp_driver_pcnt esp_driver_ppa esp_driver_rmt esp_driver_sdio esp_driver_sdm esp_driver_sdmmc esp_driver_sdspi esp_driver_spi esp_driver_touch_sens esp_driver_tsens esp_driver_uart esp_driver_usb_serial_jtag esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_vfs_console esp_wifi espcoredump espressif__esp_tinyusb espressif__mdns espressif__tinyusb esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main_esp32s3 mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/rvalls/esp/esp-idf/components/app_trace /Users/rvalls/esp/esp-idf/components/app_update /Users/rvalls/esp/esp-idf/components/bootloader /Users/rvalls/esp/esp-idf/components/bootloader_support /Users/rvalls/esp/esp-idf/components/bt /Users/rvalls/esp/esp-idf/components/cmock /Users/rvalls/esp/esp-idf/components/console /Users/rvalls/esp/esp-idf/components/cxx /Users/rvalls/esp/esp-idf/components/driver /Users/rvalls/esp/esp-idf/components/efuse /Users/rvalls/esp/esp-idf/components/esp-tls /Users/rvalls/esp/esp-idf/components/esp_adc /Users/rvalls/esp/esp-idf/components/esp_app_format /Users/rvalls/esp/esp-idf/components/esp_bootloader_format /Users/rvalls/esp/esp-idf/components/esp_coex /Users/rvalls/esp/esp-idf/components/esp_common /Users/rvalls/esp/esp-idf/components/esp_driver_ana_cmpr /Users/rvalls/esp/esp-idf/components/esp_driver_cam /Users/rvalls/esp/esp-idf/components/esp_driver_dac /Users/rvalls/esp/esp-idf/components/esp_driver_gpio /Users/rvalls/esp/esp-idf/components/esp_driver_gptimer /Users/rvalls/esp/esp-idf/components/esp_driver_i2c /Users/rvalls/esp/esp-idf/components/esp_driver_i2s /Users/rvalls/esp/esp-idf/components/esp_driver_isp /Users/rvalls/esp/esp-idf/components/esp_driver_jpeg /Users/rva [... cut ...]
[1363/1479] Building C object esp-idf/main_esp32s...uilder/micropython/shared/readline/readline.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/rvalls/dev/personal/micropython-builder/micropython/ports/esp32/build-ESP32_GENERIC_S3/log/idf_py_stderr_output_34975 and /Users/rvalls/dev/personal/micropython-builder/micropython/ports/esp32/build-ESP32_GENERIC_S3/log/idf_py_stdout_output_34975
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [all] Error 1
```

andrewleech, 2024-07-06:
The ESP32_GENERIC_S3 is built in CI with the officially supported IDF version (5.0.4) for every commit, https://github.com/micropython/micropython/blob/4d16a9cced42ad0a298f1be8b993357abe7398ab/tools/ci.sh#L155

Do you mean 5.0.4 (rather than 5.4.0)? Did you do a `make BOARD=ESP32_GENERIC_S3 submodules` ?

I've also been building for S3 a lot with docker lately and never seen this issue myself, though I've mostly been using the newer IDF 5.2.1 and 5.2.2

Perhaps that (third party) builder script has an error?

brainstorm, 2024-07-06:
> Do you mean 5.0.4 (rather than 5.4.0)?

Look at the build log I posted: 5.4.0 (master).

andrewleech, 2024-07-06:
Oh, sorry when I read your message I thought you meant micropython master, rather than IDF master. 
This makes more sense, you're talking about a compatibility issue with the upcoming / unreleased next IDF.

Yeah every new release of IDF tends to need a fair bit of work to support. I think people here typically wait until it's released though, to find out if breakage is from a changed API or just bugs in the IDF master.

brainstorm, 2024-07-06:
> Yeah every new release of IDF tends to need a fair bit of work to support. I think people here typically wait until it's released though, to find out if breakage is from a changed API or just bugs in the IDF master.

Indeed, my bad, I just checked and micropython-builder checks out IDF with 5.0.2 tag. Instead of letting it install a new IDF, I pointed to my current master/local installation.

Thanks heaps for your fast reply and also for the surprising low amount of breakage against an upstream master branch as-is, kudos!

And hey, at least you have some pre-release diff fixups for the next IDF, potentially fixing it now before it comes out (with a better fix than just commenting asserts out) :)

sang8052, 2024-08-05:
你好，这个问题在编译过程中在一次的出现了,所以申请打开这个issues 。我使用的 idf 版本 为 5.4.0 
基于当前micropython 的最新的分支编译 https://github.com/micropython/micropython/commit/ad3552ae2cb5be65a8d61dc1c3bb2014d7ef4cae
错误提示:
FAILED: esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/build_esp32/micropython/extmod/nimble/modbluetooth_nimble.c.obj 
/root/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc -DBERKELEY_DB_CONFIG_FILE=\"/build_esp32/micropython/extmod/berkeley-db/berkeley_db_config_port.h\" -DESP_PLATFORM -DFFCONF_H=\"/build_esp32/micropython/lib/oofatfs/ffconf.h\" -DIDF_VER=\"v5.4-dev-1873-g41dd1a351b\" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMICROPY_ESP_IDF_4=1 -DMICROPY_MODULE_FROZEN_MPY="(1)" -DMICROPY_PY_BTREE=1 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -DNDEBUG -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/build_esp32/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/build_esp32/micropython -I/build_esp32/micropython/lib/berkeley-db-1.xx/include -I/build_esp32/micropython/examples/usercmodule/cam -I/build_esp32/esp-idf/components/esp32-camera/driver/include -I/build_esp32/esp-idf/components/esp32-camera/driver/private_include -I/build_esp32/esp-idf/components/esp32-camera/conversions/include -I/build_esp32/esp-idf/components/esp32-camera/conversions/private_include -I/build_esp32/esp-idf/components/esp32-camera/sensors/private_include -I/build_esp32/micropython/ports/esp32 -I/build_esp32/micropython/ports/esp32/boards/ESP32_GENERIC_S3 -I/build_esp32/micropython/ports/esp32/build-ESP32_GENERIC_S3 -I/build_esp32/esp-idf/components/bt/host/nimble/nimble -I/build_esp32/esp-idf/components/newlib/platform_include -I/build_esp32/esp-idf/components/freertos/config/include -I/build_esp32/esp-idf/components/freertos/config/include/freertos -I/build_esp32/esp-idf/components/freertos/config/xtensa/include -I/build_esp32/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/build_esp32/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/build_esp32/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/build_esp32/esp-idf/components/freertos/esp_additions/include -I/build_esp32/esp-idf/components/esp_hw_support/include -I/build_esp32/esp-idf/components/esp_hw_support/include/soc -I/build_esp32/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/build_esp32/esp-idf/components/esp_hw_support/dma/include -I/build_esp32/esp-idf/components/esp_hw_support/ldo/include -I/build_esp32/esp-idf/components/esp_hw_support/debug_probe/include -I/build_esp32/esp-idf/components/esp_hw_support/port/esp32s3/. -I/build_esp32/esp-idf/components/esp_hw_support/port/esp32s3/include -I/build_esp32/esp-idf/components/heap/include -I/build_esp32/esp-idf/components/log/include -I/build_esp32/esp-idf/components/soc/include -I/build_esp32/esp-idf/components/soc/esp32s3 -I/build_esp32/esp-idf/components/soc/esp32s3/include -I/build_esp32/esp-idf/components/hal/platform_port/include -I/build_esp32/esp-idf/components/hal/esp32s3/include -I/build_esp32/esp-idf/components/hal/include -I/build_esp32/esp-idf/components/esp_rom/include -I/build_esp32/esp-idf/components/esp_rom/esp32s3/include -I/build_esp32/esp-idf/components/esp_rom/esp32s3/include/esp32s3 -I/build_esp32/esp-idf/components/esp_rom/esp32s3 -I/build_esp32/esp-idf/components/esp_common/include -I/build_esp32/esp-idf/components/esp_system/include -I/build_esp32/esp-idf/components/esp_system/port/soc -I/build_esp32/esp-idf/components/esp_system/port/include/private -I/build_esp32/esp-idf/components/xtensa/esp32s3/include -I/build_esp32/esp-idf/components/xtensa/include -I/build_esp32/esp-idf/components/xtensa/deprecated_include -I/build_esp32/esp-idf/components/lwip/include -I/build_esp32/esp-idf/components/lwip/include/apps -I/build_esp32/esp-idf/components/lwip/include/apps/sntp -I/build_esp32/esp-idf/components/lwip/lwip/src/include -I/build_esp32/esp-idf/components/lwip/port/include -I/build_esp32/esp-idf/components/lwip/port/freertos/include -I/build_esp32/esp-idf/components/lwip/port/esp32xx/include -I/build_esp32/esp-idf/components/lwip/port/esp32xx/include/arch -I/build_esp32/esp-idf/components/lwip/port/esp32xx/include/sys -I/build_esp32/esp-idf/components/app_update/include -I/build_esp32/esp-idf/components/bootloader_support/include -I/build_esp32/esp-idf/components/bootloader_support/bootloader_flash/include -I/build_esp32/esp-idf/components/esp_app_format/include -I/build_esp32/esp-idf/components/esp_bootloader_format/include -I/build_esp32/esp-idf/components/esp_partition/include -I/build_esp32/esp-idf/components/bt/include/esp32c3/include -I/build_esp32/esp-idf/components/bt/common/osi/include -I/build_esp32/esp-idf/components/bt/common/api/include/api -I/build_esp32/esp-idf/components/bt/common/btc/profile/esp/blufi/include -I/build_esp32/esp-idf/components/bt/common/btc/profile/esp/include -I/build_esp32/esp-idf/components/bt/common/hci_log/include -I/build_esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -I/build_esp32/esp-idf/components/bt/host/nimble/nimble/nimble/include -I/build_esp32/esp-idf/components/bt/host/ni [... cut ...]
root@VM-11-6-ubuntu:/build_esp32/micropython/ports/esp32# make BOARD=ESP32_GENERIC_S3 USER_C_MODULES=/build_esp32/micropython/examples/usercmodule/cam/micropython.cmake submodules
make[1]: Entering directory '/build_esp32/micropython/ports/esp32'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/berkeley-db-1.xx lib/micropython-lib
Synchronizing submodule url for '../../lib/berkeley-db-1.xx'
Synchronizing submodule url for '../../lib/micropython-lib'
make[1]: Leaving directory '/build_esp32/micropython/ports/esp32'

andrewleech, 2024-08-05:
@sang8052 it's hard to see with so much output, but is this IDF current master? Aka v5.4 (dev) build of IDF? If so, that's not currently supported.
v5 0.4 and v5.1 are officially supported and support for v5.2.2 release is currently under development.

riderman10000, 2024-09-16:
does any one have solution for this issue, I keep on getting this message doing build.

/home/rlwagun/Files/micropy_new_test/micropython/extmod/nimble/modbluetooth_nimble.c:1931:28: error: 'const struct ble_store_key_sec' has no member named 'ediv_rand_present'
 1931 |             assert(key->sec.ediv_rand_present);
      |                            ^
/home/rlwagun/Files/micropy_new_test/micropython/extmod/nimble/modbluetooth_nimble.c: In function 'ble_secret_store_write':
/home/rlwagun/Files/micropy_new_test/micropython/extmod/nimble/modbluetooth_nimble.c:1980:27: error: 'struct ble_store_key_sec' has no member named 'ediv_rand_present'
 1980 |             assert(key_sec.ediv_rand_present);
      |                           ^
[20/103] Building C object esp-idf/ma...n_esp32s3.dir/__/network_common.c.obj
FAILED: esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/network_common.c.obj 
...

kdschlosser, 2024-09-16:
you need to use a lower version of the ESP-IDF. 5.0.X to 5.2.x will work.

rambo, 2025-02-25:
Same issue with micropython v1.24-release and IDF 5.4 even though documentation claims 5.4 is supported, now trying with IDF 5.2

kdschlosser, 2025-02-25:
You can use 5.2.3 but if you compile for a C series ESP32 you need to add a macro to one of the port header files.

rambo, 2025-02-25:
> I believe it's 5.0.4 NOT 5.4. there is a very large difference there in the versioning.

https://github.com/micropython/micropython/blob/master/ports/esp32/README.md#setting-up-esp-idf-and-the-build-environment (linked from the troubleshooting guide, url of which was output by make) says:

"Currently MicroPython supports v5.2, v5.2.2, v5.3 and v5.4."

but Aha! https://github.com/micropython/micropython/blob/v1.24.1/ports/esp32/README.md says: "Currently MicroPython supports v5.0.4, v5.0.5, v5.1.2, v5.2.0, v5.2.2."

So I was looking in the wrong branch/tag, the troubleshooting guide should probably make it *very* clear that one needs to verify the branch/tag of the linked document before trusting it.

kdschlosser, 2025-02-25:
the master branch is NOT the released version. People make that mistake all the time. You have to checkout the tag for the release after you clone it. Then you will be seeing the correct information. Unfortunately MicroPython doesn't create an actual branch when they make releases so you cannot go and look at the code on GitHub. You need to clone it and checkout the tag in order to see what versions are being used.

They really should add a simple version check of the IDF and output an error if an unsupported version is being used with a message telling the user that they need to change the IDF version to one that is supported. It really would save a lot of time for both the user and also the developers of MicroPython.

kdschlosser, 2025-02-25:
5.2.3 is easily supported with only a single change needed. adding the following code block to `lib/micropython/ports/esp32/mpconfigport.h` will allow you to use IDF 5.2.3. I add the code right before `#ifndef MICROPY_CONFIG_ROM_LEVEL`

```
#ifndef USB_SERIAL_JTAG_PACKET_SZ_BYTES
#define USB_SERIAL_JTAG_PACKET_SZ_BYTES (64)
#endif
```