micropython/micropython #15289: ESP32-S3 Linking errors

kdschlosser, 2024-06-15:
### Port, board and/or hardware

ESP32-S3 SPIRAM_OCT

### MicroPython version

MicroPython: 1.23.0
OS: Ubuntu 22.04
ESP-IDF: 5.2

### Reproduction

I am just compiling, nothing special.

### Expected behaviour

Fir it to compile

### Observed behaviour

```shell
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gattc_exchange_mtu+0x0): undefined reference to `mp_bluetooth_gattc_exchange_mtu'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gattc_write+0x0): undefined reference to `mp_bluetooth_gattc_write'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gattc_read+0x0): undefined reference to `mp_bluetooth_gattc_read'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gattc_discover_descriptors+0x0): undefined reference to `mp_bluetooth_gattc_discover_descriptors'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gattc_discover_characteristics+0x4): undefined reference to `mp_bluetooth_gattc_discover_characteristics'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gattc_discover_services+0x0): undefined reference to `mp_bluetooth_gattc_discover_primary_services'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_set_buffer+0x0): undefined reference to `mp_bluetooth_gatts_set_buffer'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_notify_indicate+0x0): undefined reference to `mp_bluetooth_gatts_notify_indicate'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_write+0x0): undefined reference to `mp_bluetooth_gatts_write'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_read+0x0): undefined reference to `mp_bluetooth_gatts_read'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_register_services+0x10): undefined reference to `mp_bluetooth_gatts_register_service_begin'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_register_services+0x14): undefined reference to `mp_bluetooth_gatts_register_service'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_register_services+0x18): undefined reference to `mp_bluetooth_gatts_register_service_end'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gap_passkey+0x0): undefined reference to `mp_bluetooth_gap_passkey'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gap_pair+0x0): undefined reference to `mp_bluetooth_gap_pair'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gap_disconnect+0x0): undefined reference to `mp_bluetooth_gap_disconnect'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gap_scan+0x8): undefined reference to `mp_bluetooth_gap_scan_stop'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gap_scan+0xc): undefined reference to `mp_bluetooth_gap_scan_start'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gap_advertise+0x4): undefined reference to `mp_bluetooth_gap_advertise_stop'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gap_advertise+0x8): undefined reference to `mp_bluetooth_gap_advertise_start'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_config+0x20): undefined reference to `mp_bluetooth_gap_get_device_name'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_config+0x24): undefined reference to `mp_bluetooth_get_current_address'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_config+0x28): undefined reference to `mp_bluetooth_get_preferred_mtu'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_config+0x2c): undefined reference to `mp_bluetooth_gap_set_device_name'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_config+0x30): undefined reference to `mp_bluetooth_set_preferred_mtu'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_config+0x34): undefined reference to `mp_bluetooth_set_address_mode'
/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(modbluetooth.c.obj):(.literal.bluetooth_ble_config+0x38): undefined reference to `mp_bluetooth_set_bonding'
[... cut ...]
ninja failed with exit code 1, output of the command is in the /micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_333487 and /micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_333487
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:66: all] Error 1
make: Leaving directory '/micropython/ports/esp32'
```

### Additional Information

I did make sure to checkout the correct version of the ESP-IDF and the correct version is being detected by the build system.

### Code of Conduct

Yes, I agree

andrewleech, 2024-06-16:
I just did a clean `git checkout v1.23.0` followed by a compile of esp32 S3 with the official Espressif docker image v5.2.1 with no errors. 

I've got a script I now use to easily compile any esp32 board with any IDF version I simply have in `~/.local/bin/esp-compile`
```
if [ "$1" == "" ]; then echo "$0 IDF_VER BOARD <DEPLOY_PORT> <clean>"; exit; fi
IDF_VER=$1
BOARD=$2
DEPLOY_PORT=$3
CLEAN=$([ "${@: -1}" == "clean" ] && echo "1")
docker run -it --rm -e BOARD=$BOARD -e CLEAN=$CLEAN -e DEPLOY_PORT=$DEPLOY_PORT -e UID=$(id -u) -v $(pwd):/micropython -v /sys/bus:/sys/bus -v /dev:/dev --net=host --privileged -w /micropython espressif/idf:$IDF_VER bash -c '\
useradd -ms /bin/bash -g root -G sudo -u $UID esp;\
git config --global --add safe.directory /opt/esp/idf;\
su esp -m -c ". /opt/esp/entrypoint.sh; export HOME=/tmp; \
if [ ! -e mpy-cross/build/mpy-cross ]; then \
make -C mpy-cross ; fi && \
make -C ports/esp32 BOARD=$BOARD submodules && \
if [ \"$CLEAN\" == \"1\" ] ; then \
make -C ports/esp32 BOARD=$BOARD clean; \
else \
if [ \"$DEPLOY_PORT\" == \"\" ] ; then \
make -C ports/esp32 BOARD=$BOARD V=1; \
else \
make -C ports/esp32 BOARD=$BOARD deploy PORT=$DEPLOY_PORT; \
fi;fi;"'

# Examples
# esp-compile v5.0.4 ESP32_GENERIC /dev/ttyUSB0
# esp-compile v5.2.1 ESP32_GENERIC_S3
# esp-compile v5.2.1 ESP32_GENERIC_C6 /dev/ttyACM0
```

I didn't see any link errors or nimble errors. 

If your micropython repo is an older clone it might be worth running `git submodule sync` to make sure they're also up to date.

kdschlosser, 2024-06-16:
Firstly it is impossible for you to not have had ANY errors when using IDF 5.2.1...

right at the very bottom of network_common.c is the following code.

```c
#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 2, 0)
_Static_assert(WIFI_AUTH_MAX == 13, "Synchronize WIFI_AUTH_XXX constants with the ESP-IDF. Look at esp-idf/components/esp_wifi/include/esp_wifi_types.h");
#elif ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 0, 5) && ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5, 1, 0) || ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 1, 2)
_Static_assert(WIFI_AUTH_MAX == 11, "Synchronize WIFI_AUTH_XXX constants with the ESP-IDF. Look at esp-idf/components/esp_wifi/include/esp_wifi_types.h");
#else
_Static_assert(WIFI_AUTH_MAX == 10, "Synchronize WIFI_AUTH_XXX constants with the ESP-IDF. Look at esp-idf/components/esp_wifi/include/esp_wifi_types.h");
#endif

```

WIFI_AUTH_MAX in esp-idf 5.2.1 has a value of 13. The code above would have been checking to make sure it had a value of 11. guess what? the build will fail. You cannot compile 1.23.0 using IDF 5.2.1 without the build failing. 

I am using the released version of 1.23.0. This is supposed to be stable code and what is in the master branch is not released and is not to be considered as stable. the code in the master branch is also not the same code that is seen in the 1.23.0 release.

kdschlosser, 2024-06-16:
I know where the problem is.

in modbluetooth.c there is the inclusion of modbluetooth.h where the functions are declared but not where they are defined. They are defined in modbluetooth_nimble.c  and in modbluetooth_btstack.c

modbluetooth_btstack is wrapped in `MICROPY_BLUETOOTH_BTSTACK` and modbluetooth_nimble is wrapped in `MICROPY_BLUETOOTH_NIMBLE`  so when I had the nimble compile error I changed `MICROPY_BLUETOOTH_NIMBLE` to `0` to disable the nimble module and low and behold there is no wrapper around modbluetooth.c code to ensure that either `MICROPY_BLUETOOTH_NIMBLE` is set to a `1` or `MICROPY_BLUETOOTH_BTSTACK` is set to a `1`. if neither of those are set to a `1` then an error should be generated informing the user that there is a requirement to have either of those set to `1`. Even better would be testing to see if they are both set to zero and defaulting one of them to a `1` and spitting out a compiler warning about it.

Ultimately there is a bug.

kdschlosser, 2024-06-16:
now I am not 100% sure as to why the functions are declared in a header file that the source file doesn't define. The macros should control the inclusion of the header file that declares the functions and that header file should have the matching source file that defines them.

The current way it is done adds unneeded complexity to the code.

kdschlosser, 2024-06-16:
Move the declarations from those functions out of modbluetooth.h and into the header files that have the matching source file where the functions are defined and the following code should be added to the modbluetooth.c source file at the top.

```c
#if MICROPY_PY_BLUETOOTH
#if MICROPY_BLUETOOTH_NIMBLE 
#include "nimble/modbluetooth_nimble.h"
#elif MICROPY_BLUETOOTH_BTSTACK 
#include "btstack/modbluetooth_btstack.h"
#else
#error "MICROPY_BLUETOOTH_NIMBLE or MICROPY_BLUETOOTH_BTSTACK needs to be set to a 1"
#endif
//... rest of code
#endif /* MICROPY_PY_BLUETOOTH */
```

with the current code there is zero connection that is able to be seen between modbluetooth.c and where the functions are defined.

andrewleech, 2024-06-16:
The structure hung used is for `modbluetooth.h` to act like an "interface", that is it's the single source of truth for low level functions that need to be defined for the high level modbluetooth module to compile. 
These functions can be provided by a number of different stacks, currently this includes nimble, btstack (as you've seen) and zephyr also. 

These low level functions are not declared in the headers for each stack because that would mean duplicating all the exact same declarations between all of them and then doing something to make sure they all stay in sync. Keeping just the one common declaration that all stack implementations rely on helps keep them aligned. I didn't create this structure, but I do think it makes a lot of sense. I've worked with it a lot in past years though, I can see how she extra comments or doc entries the explain this would be helpful.

Most if not all ports using the stacks do have make checks to ensure that multiple ones aren't enabled, but yeah there isn't checks to ensure that one is. I guess this linker error _is_ the thing saying the low level implementation is missing :-) not the easiest error to understand though for sure. Having make checks for missing impl would certainly make it easier for developers new to this area of mpy.

To take a step back I still don't know why you were seeing a problem at all though (the previous issue before disabling nimble), unless your IDF was broken somehow? Or were you compiling a different S3 board that was configured differently?

jonnor, 2025-03-03:
Not reproducible.

closed 2025-03-03 by jonnor