micropython/micropython #15297: rp2 - mDNS not working

massimosala, 2024-06-17:
### Port, board and/or hardware

rp2040

### MicroPython version

mp 1.22.2

### Reproduction

There isn't a code sample, the feature is built-in in the firmware.

### Expected behaviour

The micro working on WiFi (STA_IF or AP_IF) should respond to mDNS queries.

### Observed behaviour

Not working.

### Additional Information

I code in Python a mDNS and LLMNR responder.

It works like a charm on the esp8266: the micro, with its hostname xxxxxx.local is recognized by any browser (browsers use mDNS) and by ping (on microsoft pc, ping use also LLMNR).

Running my module on the rp2:

- LLMNR works
- MDNS doesn't work: when my code binds a UDP socket to the local port 5353, mp raises an exception
_already in use_

So the firmware has already taken the mDNS port ... is trying to do something... but mDNS isn't working.

### Code of Conduct

Yes, I agree

massimosala, 2024-06-18:
Tested also with MP 1.23.0 : not resolved

felixdoerre, 2024-07-06:
On rp2 MDNS is enabled to be provided by lwip: https://github.com/micropython/micropython/blob/master/ports/rp2/lwip_inc/lwipopts.h#L35. Maybe you want to compile your own firmware with that flag disabled?

mseminatore, 2024-11-19:
I am looking for a fix for this as well. I would prefer not to build my own custom firmware and use slimdns.

mseminatore, 2024-11-19:
> On rp2 MDNS is enabled to be provided by lwip: https://github.com/micropython/micropython/blob/master/ports/rp2/lwip_inc/lwipopts.h#L35. Maybe you want to compile your own firmware with that flag disabled?

But I think the point being made is that LWIP is **not** providing a working mDNS implementation on rp2 and yet is holding onto the port. Could that perhaps be a result of a build order of operations with the LWIP library and the rps lwipopts.h? Or some missing glue code?

It is fair if mDNS support is not yet implemented for the rp2 port. But it seems very much like a bug for rp2 lwipopts.h to define the following if mDNS support is not properly included/implemented in the fw:

```
#define LWIP_DNS_SUPPORT_MDNS_QUERIES   1
#define LWIP_MDNS_RESPONDER             1
```

I also notice that the following lines are not included in the rp2 mpconfigport.h file:

```
#ifndef MICROPY_HW_ENABLE_MDNS_QUERIES
#define MICROPY_HW_ENABLE_MDNS_QUERIES      (1)
#endif

#ifndef MICROPY_HW_ENABLE_MDNS_RESPONDER
#define MICROPY_HW_ENABLE_MDNS_RESPONDER    (1)
#endif
```

The ESP port has code in network_wlan.c that specifically calls mDNS initialization and provides some handling code. The rp2 port does not seem to have similar code.

mseminatore, 2024-12-04:
@felixdoerre Looking over the codebase some more I think I might see the issue. In ports/rp2/main.c:125 there is this code:

``` C
    #if MICROPY_PY_LWIP
    // lwIP doesn't allow to reinitialise itself by subsequent calls to this function
    // because the system timeout list (next_timeout) is only ever reset by BSS clearing.
    // So for now we only init the lwIP stack once on power-up.
    lwip_init();
    #if LWIP_MDNS_RESPONDER
    mdns_resp_init();
    #endif
    #endif
```

Per the LWIP docs, the `mdns_resp_init()` opens the mDNS socket/port, but mDNS is not activated on a network interface until a call to `mdns_resp_add_netif()` which is not called anywhere that I can find. If accurate, RP2 basically has only a partial mDNS implementation and **LWIP_MDNS_RESPONDER** should probably be 0 by default until/unless a complete implementation is provided.

If someone can point me to where the netif is initialized in the RP2 port I am happy to build up a PR.

mseminatore, 2025-01-03:
@dpgeorge any comment on the default behavior? Seems like mDNS should either be fully functional or disabled by default. Happy to contribute a PR if appropriate.

projectgus, 2025-01-29:
> If someone can point me to where the netif is initialized in the RP2 port I am happy to build up a PR.

That would be great! 

The layering is a bit complex. The netif for the PICO-W board actually gets initialised inside the external cyw43-driver (submodule in lib/cyw43-driver), look for callback `cyw43_cb_tcpip_init` (called as the low-level Wi-Fi interface goes "up").

So I think the way to do this that is most consistent with the cyw43 driver layering, would be to add a `CYW43_MDNS_RESPONDER` config macro (or similar) in the driver and add the necessary calls in in cyw43_lwip.c then submit that as a PR to the driver repo at https://github.com/georgerobotics/cyw43-driver - once it's added then it can be enabled conditionally in MicroPython.

It'd be good to have a more generic way of implementing this stuff, as it seems most of the bare metal lwIP interfaces will have this limitation (only `extmod/network_wiz5k.c` seems to implement it at all so far). However I don't see an easy way to achieve that without some major refactors...

projectgus, 2025-01-29:
(Apologies, I had this tab open since before you opened the other PR and looks like you found a way to do it from the MicroPython side. Disregard my suggestion!)

dpgeorge, 2025-04-03:
Should be fixed by f96417dbf28617c533e4f2e65c65d1ed11f089fa (although looking up mDNS on the Pico W in AP mode might not work yet).