micropython/micropython #15308: mpremote mount chdir() in mount fails after second call

MathyV, 2024-06-19:
### Port, board and/or hardware

esp32, custom board

### MicroPython version

MicroPython v1.23.0-81.g12cedaa81.dirty on 2024-06-19; Fri3d Badge 2024 with ESP32S3

### Reproduction

```
Local directory ./ is mounted at /remote
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] or Ctrl-x to exit this shell
>
MicroPython v1.23.0-81.g12cedaa81.dirty on 2024-06-19; Fri3d Badge 2024 with ESP32S3
Type "help()" for more information.
>>> os.chdir('/remote/fri3d/apps/splash')
>>> os.listdir()
['log.py', 'splash.py', 'app.json', '__init__.py']
>>> os.chdir('/remote/fri3d/apps')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 189, in chdir
  File "<stdin>", line 233, in stat
OSError: [Errno 2] ENOENT
```

### Expected behaviour

All the chdir() calls should work

### Observed behaviour

Only the first call, no matter where into the mounted files, works. All next calls to any other directory in the mount fails.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

felixdoerre, 2024-06-19:
If you're interested in a simple workaround. Just "chdir" out of the mountpoint in between: `os.chdir(''): os.chdir('/remote/fri3d/apps')`.

MathyV, 2024-06-19:
I tried that before actually but it doesn't help:

```
MicroPython v1.23.0-81.g12cedaa81.dirty on 2024-06-20; Fri3d Badge 2024 with ESP32S3
Type "help()" for more information.
>>> os.chdir('/remote/fri3d')
>>> os.chdir('/')
>>> os.chdir('/remote/fri3d/apps')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 189, in chdir
  File "<stdin>", line 233, in stat
OSError: [Errno 2] ENOENT
>>>
```

dpgeorge, 2024-06-20:
Thanks for the report.  I can confirm the issue (on a PYBD-SF2).

It looks like a simple workaround is to do `os.chdir('/remote')` to reset things.

dpgeorge, 2024-06-20:
See fix in #15311.

dpgeorge, 2024-06-21:
Fixed by cebc9b0ae2b12c61eac39a3c599edb3b1b65dd54

closed 2024-06-21 by dpgeorge