micropython/micropython #15325: STM32F407+MII+DP83848 not working properly

RichardWei, 2024-06-21:
### Port, board and/or hardware

stm32f407

### MicroPython version

i use the master banch

### Issue Report

I have made the following modifications based on the VCC_GND_F407ZG template:

Hardware Platform: STM32F407ZGT6 + DP83848. My connection platform is as follows. I use a 25MHz crystal oscillator instead of a 50MHz one, considering it is more common. Although the RMII interface uses fewer pins, the MCU's I/O is usually sufficient for most projects.

```python
import network
station = network.LAN(phy_addr=0x01, phy_type=network.LAN.PHY_DP83848)
station.active("True")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 110] ETIMEDOUT
```

Jumper Settings
JP5: Shorted 1-2, no U3
JP6: Shorted 2-3
JP8: Open
I measured the MII interface waveforms. It seems there is no proper TX information. The RX signals on RX0, RX1, RX2, and RX3 show varying waveforms, but TX only has a waveform on CLK. I am not sure if this is because the PHY is not initialized or if there are errors in the MII interface.

I am confident that my hardware platform is functioning correctly. Using STM32Cube, I exported a simple project with MII + lwIP, and I can ping it within the local network, that mean the MII interface and PHY communication are working properly.

Code Modifications
In mpconfigboard.h, I added the following macros:
```c

#define MICROPY_HW_RTH_PHY_MII (1)
// #define MICROPY_HW_USB_VBUS_DETECT_PIN (pin_A9)
// #define MICROPY_HW_USB_OTG_ID_PIN      (pin_A10)

// Ethernet via RMII
#define MICROPY_HW_ETH_MDC          (pin_C1)
#define MICROPY_HW_ETH_MDIO         (pin_A2)

#if MICROPY_HW_ETH_PHY_RMII

#define MICROPY_HW_ETH_RMII_REF_CLK (pin_A1)
#define MICROPY_HW_ETH_RMII_CRS_DV  (pin_A7)
#define MICROPY_HW_ETH_RMII_RXD0    (pin_C4)
#define MICROPY_HW_ETH_RMII_RXD1    (pin_C5)
#define MICROPY_HW_ETH_RMII_TX_EN   (pin_B11)
#define MICROPY_HW_ETH_RMII_TXD0    (pin_B12)
#define MICROPY_HW_ETH_RMII_TXD1    (pin_B13)

#elif MICROPY_HW_ETH_PHY_MII

#define MICROPY_HW_ETH_MII_TX_CLK    (pin_C3)
#define MICROPY_HW_ETH_MII_TX_EN     (pin_B11)
#define MICROPY_HW_ETH_MII_TXD0      (pin_B12)
#define MICROPY_HW_ETH_MII_TXD1      (pin_B13)
#define MICROPY_HW_ETH_MII_TXD2      (pin_C2)
#define MICROPY_HW_ETH_MII_TXD3      (pin_E2)
#define MICROPY_HW_ETH_MII_RX_CLK    (pin_A1)
#define MICROPY_HW_ETH_MII_RX_DV     (pin_A7)
#define MICROPY_HW_ETH_MII_RXD0      (pin_C4)
#define MICROPY_HW_ETH_MII_RXD1      (pin_C5)
#define MICROPY_HW_ETH_MII_RXD2      (pin_B0)
#define MICROPY_HW_ETH_MII_RXD3      (pin_B1)
#define MICROPY_HW_ETH_MII_RX_ER     (pin_B10)
#define MICROPY_HW_ETH_MII_CRS       (pin_A0)
#define MICROPY_HW_ETH_MII_COL       (pin_A3)

#endif
```
In eth.c, I made the following definitions:
```c
#if MICROPY_HW_ETH_PHY_RMII 

    mp_hal_pin_config_alt_static_speed(MICROPY_HW_ETH_RMII_REF_CLK, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, MP_HAL_PIN_SPEED_MEDIUM, STATIC_AF_ETH_RMII_REF_CLK);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_RMII_CRS_DV, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_RMII_CRS_DV);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_RMII_RXD0, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_RMII_RXD0);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_RMII_RXD1, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_RMII_RXD1);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_RMII_TX_EN, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_RMII_TX_EN);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_RMII_TXD0, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_RMII_TXD0);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_RMII_TXD1, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_RMII_TXD1);

#elif MICROPY_HW_ETH_PHY_MII

    mp_hal_pin_config_alt_static_speed(MICROPY_HW_ETH_MII_TX_CLK, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, MP_HAL_PIN_SPEED_MEDIUM, STATIC_AF_ETH_MII_TX_CLK);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_TX_EN, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_TX_EN);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_TXD0, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_TXD0);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_TXD1, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_TXD1);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_TXD2, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_TXD2);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_TXD3, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_TXD3);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_RX_CLK, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_RX_CLK);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_RX_DV, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_RX_DV);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_RXD0, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_RXD0);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_RXD1, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_RXD1);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_RXD2, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_RXD2);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_RXD3, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_RXD3);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_RX_ER, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_RX_ER);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_CRS, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_CRS);
    mp_hal_pin_config_alt_static(MICROPY_HW_ETH_MII_COL, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_MII_COL);

#endif
```

When selecting the rmii or mii mode, I added the following:
```c
#if MICROPY_HW_ETH_PHY_RMII
    SYSCFG->PMC |= SYSCFG_PMC_MII_RMII_SEL;
#elif MICROPY_HW_ETH_PHY_MII
    SYSCFG->PMC &= ~(SYSCFG_PMC_MII_RMII_SEL);
    SYSCFG->PMC |= (uint32_t)(ETH_MEDIA_INTERFACE_MII);
#endif

```

[image]

### Code of Conduct

Yes, I agree

robert-hh, 2024-06-21:
You can try to modify the phy_addr, check whether there is proper communication at MDC and MDIO, and check the 25MHz ref_clk being sent to the STM32.

RichardWei, 2024-06-21:
I measured MDC and MDIO, their waveforms are normal, and when I use lan.active(“true”)

TX has no waveform except CLK, which is very strange. I bought a DP83848 module and will receive it next Monday. I will use it and my development board to test it. I use an external 25MHZ clock on my PCB. When connected to the network, the LED will flash, indicating that there is communication between RJ45 and PHY, but the communication between MAC and PHY is not normal. I think it may be a problem with the MII interface, but I am not sure.

RichardWei, 2024-06-22:
@robert-hh,updated,I received the module and was unlucky and seems to have a similar issue.
I changed to RMII mode and recompiled the firmware,[LAN8720](https://www.waveshare.com/lan8720-eth-board.htm)，[9844731](https://github.com/orgs/micropython/discussions/13406#discussioncomment-9844731)，

robert-hh, 2024-06-22:
In the above settings of the first post I see `#define MICROPY_HW_RTH_PHY_MII (1)`. But it should be `#define MICROPY_HW_RTH_PHY_RMII (1)` to use the RMII interface. The pin assignments for the RMII interface look fine.
Two things have to be checked:
- When the LAN interface is started, the driver sends a soft reset command to the PHY. At that moment, the LEDs at the RJ45 plug should switch off for a moment. If they do not, then the MDIO communication fails. You can tra a different phy_addr setting. You should as well check the value of the MDC clock. it must not exceed 2.5 MHz.
- REF_CLK must be 50 MHz for RMII mode, and it must be input to the STM32F407. The LAN8720 board provides this, and in the above schematics the DP83848 board as well.

RichardWei, 2024-06-24:
Thank‘'s for your reply. I tested my MDC frequency and it was 4MHZ, which is abnormal. To ensure that it was not accidentally modified by me, I decided to find a most similar platform to build the firmware from scratch to test the RMII interface on stm32f407zg. My steps are as follows,

```shell
git clone https://github.com/micropython/micropython.git
cd micropython
git tag
git checkout tags/v1.23.0 -b my-branch-1.23.0
git submodule update --init --recursive
```

Modify the crystal oscillator. My crystal oscillator is 8mhz.
In the VCC_GND_F407ZG folder, modify the following files

mpconfigboard.h
```C
#define MICROPY_HW_CLK_PLLM (8)            // divide external clock by this to get 1MHz
```
stm32f4xx_hal_conf.h
```C
#define HSE_VALUE (8000000)
```
Then start building
```shell
make -C mpy-cross
cd ports/stm32
BOARD=VCC_GND_F407ZG 
rm -rf build-VCC_GND_F407ZG
make BOARD=$BOARD
cp build-VCC_GND_F407ZG/firmware.hex boards/VCC_GND_F407ZG/
```
I only changed the crystal frequency, and did not make any other changes, but the MDC clock is still 4MHZ. I think this is most likely the reason.

I checked the following mdc clock frequency code in eth.c and found no abnormality

```C
  #if defined(STM32H5)
    cr_div = ETH->MACMDIOAR & ~ETH_MACMDIOAR_CR;
    if (hclk < 35000000) {
        cr_div |= ETH_MACMDIOAR_CR_DIV16;
    } else if (hclk < 60000000) {
        cr_div |= ETH_MACMDIOAR_CR_DIV26;
    } else if (hclk < 100000000) {
        cr_div |= ETH_MACMDIOAR_CR_DIV42;
    } else if (hclk < 150000000) {
        cr_div |= ETH_MACMDIOAR_CR_DIV62;
    } else if (hclk < 250000000) {
        cr_div |= ETH_MACMDIOAR_CR_DIV102;
    } else {
        cr_div |= ETH_MACMDIOAR_CR_DIV124;
    }
    ETH->MACMDIOAR = cr_div;
    #elif defined(STM32H7)
    cr_div = ETH->MACMDIOAR & ~ETH_MACMDIOAR_CR;
    if (hclk < 35000000) {
        cr_div |= ETH_MACMDIOAR_CR_DIV16;
    } else if (hclk < 60000000) {
        cr_div |= ETH_MACMDIOAR_CR_DIV26;
    } else if (hclk < 100000000) {
        cr_div |= ETH_MACMDIOAR_CR_DIV42;
    } else if (hclk < 150000000) {
        cr_div |= ETH_MACMDIOAR_CR_DIV62;
    } else {
        cr_div |= ETH_MACMDIOAR_CR_DIV102;
    }
    ETH->MACMDIOAR = cr_div;
    #else
    if (hclk < 35000000) {
        cr_div = ETH_MACMIIAR_CR_Div16;
    } else if (hclk < 60000000) {
        cr_div = ETH_MACMIIAR_CR_Div26;
    } else if (hclk < 100000000) {
        cr_div = ETH_MACMIIAR_CR_Div42;
    } else if (hclk < 150000000) {
        cr_div = ETH_MACMIIAR_CR_Div62;
    } else {
        cr_div = ETH_MACMIIAR_CR_Div102;
    }
    ETH->MACMIIAR = cr_div;
    #endif

```

When I use active("True"), the LED does not turn off and then on, indicating that the lan8720 did not receive the reset command.

RichardWei, 2024-06-24:
> In the above settings of the first post I see `#define MICROPY_HW_RTH_PHY_MII (1)`. But it should be `#define MICROPY_HW_RTH_PHY_RMII (1)` to use the RMII interface. The pin assignments for the RMII [...]
@robert-hh Good news, I tried the old version and it worked, version number V1.18, I repeated the above work, just changed the frequency of the crystal, everything is normal. Now STM32F407ZGTE+RMII+LAN8720 works on V1.18. This is a good start, even if DP83848 is not supported on 1.18.

The IP has been successfully obtained.

···shell
>>> import network
>>> station = network.LAN()
>>> station.active("True")
>>> station.ifconfig()
('192.168.2.112', '255.255.255.0', '192.168.2.1', '192.168.2.1')
···

The frequency of mdc is also normal on v1.18, which is 1.647MHZ, see the picture below
[image]

robert-hh, 2024-06-24:
That looks indeed good. You could compare the clock settings between v1.18 and v1.23. I do not see changes in the VCC_GND_F407ZG files for the clock setting. What is the value of the crystal at your board. VCC_GND_F407 uses 25MHz.

I ordered a F407 breakout to replicate the work. But it will take a few days to arrive.

RichardWei, 2024-06-24:
> That looks indeed good. You could compare the clock settings between v1.18 and v1.23. I do not see changes in the VCC_GND_F407ZG files for the clock setting. What is the value of the crystal at your [...]

My crystal is 8MHZ, I changed the crystal from 25MHZ to 8MHZ.I will continue to try some newer version（v1.19，v1.20), so I can narrow down the comparison.

robert-hh, 2024-06-29:
I reecived the F407 board and managed to get a working firmware. The only disappointing part is the low size of the Python free heap, which is about 22k after boot. What is the the free heap at your board after boot reported by `import gc;gc.collect();gc.mem_free()`?
I did not start yet attaching a ETHERNET board.

robert-hh, 2024-06-29:
I wired up a LAN8720 to that board and it works. A few observations:
- After a power cycle, the first communication has a MDC of 4 MHz, then it switches down to 1.647 MHz (168MHz / 102). That seems to be introduced in 2021, before v1.18
- When the 50MHz RefClk is missing at the board, nic.active(1) times out after about 5 seconds.
- I always had to start a test with a power cycle.

So the lesson is:

a) check the wiring
b) Verify that RefCLK is provided by the LAN8720 or DP83848 to the STM32F407, Pin A1.

RichardWei, 2024-06-30:
> I reecived the F407 board and managed to get a working firmware. The only disappointing part is the low size of the Python free heap, which is about 22k after boot. What is the the free heap at your [...]

i test the my board
```shell
>>> import gc
>>> gc.collect()
>>> gc.mem_free()
34912
```

RichardWei, 2024-06-30:
> I wired up a LAN8720 to that board and it works. A few observations:
> 
> * After a power cycle, the first communication has a MDC of 4 MHz, then it switches down to 1.647 MHz (168MHz / 102). That seems to be introduced in 2021, before v1.18
> * When the 50MHz RefClk is missing at the board, nic.active(1) times out after about 5 seconds.
> * I always had to start a test with a power cycle.
> 
> So the lesson is:
> 
> a) check the wiring b) Verify that RefCLK is provided by the LAN8720 or DP83848 to the STM32F407 at Pin A1.
> 
> Edit: A DP83848 breakout works as well.
thanks and I would like to know
1. Whichever version you are using?
2. What does power cycle mean? How do I repeat this process?

robert-hh, 2024-06-30:
I've built a fresh version of the firmware since the board files had to be adapted. So it's 1.24.0 preview xx. Since I do not use the same board than you, providing a firmware Image seems not to be useful.
Power cycle is board power off - power on. That ensures that all hardware is in a clean reset state. Pushing reset should do the same. Once it worked, the power cycle/reset was not required any more.

I tested using a jlink debugger going through the various steps of the set-up with various configurations. If after issuing `nic.active(True)` I got a timeout error after about a second, then the reset of the Ethernet controller failed. Reason: `ref_clk` was missing. If later I got a timeout after about 5 seconds, the LAN cable was not attached properly.

What did not matter was the initial 4MHz MDC clock. That changed with the next MDIO commands to 1.647 MHz. Surely something to fix, but not critical.

When you made your tests with v1.18, did you use the same LAN adapter board wiring and use the same configuration? That would be strange. I checked the commits for stm32/eth.c between v1.18 and the actual version one of the relevant changes was to specify the `phy_type` and `phy_addr`. So you could try to change the phy_addr in the constructor. For LAN8720 suitable values are 0 and 1. The DP83848 has 5 address pins, with 1 as default address. So it depends on the wiring of the board which address is selected. You DP83848 seems to have a jumper at Pin 39 for the MII mode. That Pin must be pulled high. I cannot read the jumper number at your post, but it must be closed.

robert-hh, 2024-06-30:
Going to your first post with the DP83848 and comparing it with the data sheet, for RMII mode JP8 must be closed. and JP6 in positions 1-2, causing a 50MHz clock at the REFCLK pin. The solder bridge SB1 should then be open.
If the Address pins are not driven high or low, the internal pull resistors will set phy_addr to 1.

RichardWei, 2024-06-30:
> I've built a fresh version of the firmware since the board files had to be adapted. So it's 1.24.0 preview xx. Since I do not use the same board than you, providing a firmware Image seems not to be [...]

I will pay special attention to the REF_CLK signal. When the Ethernet connection is lost, the different timeouts will be an important indicator.

I used the same configuration to test the LAN8720. If the hard reset or soft reset is repeated, the network cannot ensure that it can be connected every time, so I ordered a new development board with integrated LAN8720, which does not require additional connections to ensure stability.

So I will use the new development board to verify this problem again.

Could you share the schematics of your DP83848 module and LAN8720 module?

robert-hh, 2024-06-30:
I did not look especially at the behavior during several hard and soft resets. Only that it connected every time instantly.
Here are the schematics of the two Waveshare boards I used for testing.

[LAN8720_DP83848_schem.zip](https://github.com/user-attachments/files/16045085/LAN8720_DP83848_schem.zip)

RichardWei, 2024-06-30:
> I did not look especially at the behavior during several hard and soft resets. Only that it connected every time instantly. Here are the schematics of the two Waveshare boards I used for testing. [...]

Thank‘s for your test. I need to compare my hardware carefully.If anyone else has similar problems, they can find some ideas here.Thanks again.

RichardWei, 2024-07-01:
Now all my platforms are working properly, including LAN8720 with RMII interface, DP83847 with RMII interface, and DP83848 with MII interface on my own PCB. I made some major improvements.

1. Directly control the reset of the PHY chip, which is more reliable than the RC reset, and I can freely control the reset time.

2. I use a breakout board, so my ref_clk and RMII waveforms are very bad. But the connection wires used now are very short.

I think this issue can be closed, I compiled a new firmware that supports the MII interface, and here is my code

```python
import time
import network
import socket
import time
from pyb import LED
p0 = machine.Pin("C13", machine.Pin.OUT)
led1 = LED(1)
led2 = LED(2)

def led_flash(n, led):
    for m in range(2*n):
        time.sleep_ms(50)
        led.toggle()
    led.off()

led_flash(3, led1)
station = network.LAN(phy_addr=0x01, phy_type=network.LAN.PHY_DP83848)
p0.value(0)
time.sleep_ms(10)
p0.value(1)

station.active("True")
led_flash(3, led2)
station.ifconfig()
print("station.status():", station.status())
print("station.isconnected():", station.isconnected())
print("station.ifconfig():", station.ifconfig())
```
Whether it is a software reset or the first power-on, it can connect to the network normally.However, if I change it to the following,The first time you power on, it will not work properly and you need to use software reset to connect.

```python
import time
import network
import socket
import time
from pyb import LED
p0 = machine.Pin("C13", machine.Pin.OUT)
led1 = LED(1)
led2 = LED(2)

def led_flash(n, led):
    for m in range(2*n):
        time.sleep_ms(50)
        led.toggle()
    led.off()

p0.value(0)
time.sleep_ms(10)
p0.value(1)

led_flash(3, led1)
station = network.LAN(phy_addr=0x01, phy_type=network.LAN.PHY_DP83848)

station.active("True")
led_flash(3, led2)
station.ifconfig()
print("station.status():", station.status())
print("station.isconnected():", station.isconnected())
print("station.ifconfig():", station.ifconfig())
```
Anyway, it works fine now. just a little confused.

robert-hh, 2024-07-01:
Sounds good. If you opened the issue, you can close it yourself. When testing, I made a small change to the F407 linker scripts gaining a python head size increase of 14kByte. Changed file mpconfigboard.mk:
```
MCU_SERIES = f4
CMSIS_MCU = STM32F407xx
AF_FILE = boards/stm32f405_af.csv
LD_FILES = boards/stm32f405.ld boards/VCC_GND_F407ZG/stm32f407_dma_bss.ld boards/common_ifs.ld
TEXT0_ADDR = 0x08000000
TEXT1_ADDR = 0x08020000

MICROPY_PY_LWIP = 1
MICROPY_PY_SSL = 1
MICROPY_SSL_MBEDTLS = 1

FROZEN_MANIFEST = $(BOARD_DIR)/manifest.py
```
New file `stm32f407_dma_bss.ld`:
```
/*
    GNU linker script for DevEBox STM32F407
*/

/* Define output sections */
SECTIONS
{
    .bss :
    {
        . = ALIGN(4);
        *eth.o*(.bss.eth_dma)
    } >RAM
}
```
It just moved the DMA buffer to the start of the BSS segment. The buffer must be allocated at a 16k boundary, and it it is not, up to 16k RAM may be wasted just for a *fill* space. In the actual case it was about 14k.

RichardWei, 2024-07-01:
That's a good idea, after turning on the network, if there is only 22K of  head size , it is less when running a complex script.14K is a significant increase.

closed 2024-07-01 by RichardWei