micropython/micropython #15340: ESP32 Outputting Garbage in REPL

MATTYGILO, 2024-06-24:
### Port, board and/or hardware

esp32s3

### MicroPython version

Hi I was using micropython on my esp32s3 and then it started outputting garbage. I really don't want this to happen to the device whilst it is running in production. What do you think the cause of this error is. Also is there any best practices in stopping this if it happens or preventative measures. All help is much appreciated.
[image]

> MicroPython 66af921a3 on 2024-04-13; ESP32S3 module (spiram octal) with ESP32S3

### Reproduction

I am not fully sure this happened out of the blue a few times and I am not sure how to recreate it

### Observed behaviour

[image]

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

projectgus, 2024-06-26:
Hi @MATTYGILO,

Certainly some unusual behaviour! I have a few questions:

* Does the ESP32-S3 crash or reset when it does this? It looks like after the garbage is done printing it's outputting the hard reset boot messages, is that because you manually reset it or did it happen by itself?  If it's resetting by itself then it will print the reset reason as part of the boot messages, which will give a clue.
* Are you using the builtin ESP32-S3 serial or an external USB to serial converter chip?
* I see from your link that you're using your own fork of MicroPython which has quite a few changes, and is forked from master commit 41c91422 - over a year old. Any chance of reproducing with a recent unmodified upstream MicroPython?
* Have you seen this on more than one physical ESP32-S3? Do the board(s) with the problem have good power supply, and good hardware layout principles?

My best (educated) guess looking at this is that it may be a hardware issue with power stability, or an otherwise faulty board. Brownouts or insufficient power supply capacitance can trigger behaviour that looks a lot like this.

projectgus, 2026-09-15:
Closing as there's not enough information to determine if this is a MicroPython problem (sounds more like a hardware problem).

closed 2026-09-15 by projectgus