micropython/micropython #15356: ValueError: incompatible .mpy file

KLFgrm, 2024-06-27:
Hi community,
in my projects I use precompiled mpy modules which I upload to an ESP32 WROOM (my OS is Win10).
For the precompile I use python version python-3.12.4-amd64
This works very nice with firmware version esp32-20220117-v1.18.
Now I downloaded the newest firmware version ESP32_GENERIC-20240602-v1.23.0 from https://micropython.org/download/ESP32_GENERIC/. 
But when I start a mpy file on the ESP32 I get an error: “ValueError: incompatible .mpy file“
If I go back to firmware version 1.18 all is running nice again.
Did I forgot something to update?
thanks a lot for a little info

robert-hh, 2024-06-27:
You have to re-create the .mpy files using the recent version of mpy-cross.

KLFgrm, 2024-06-27:
Yes ... thanks a lot for your fast answer. 
I used vers 5 for the cross compiler but vers 6 is needed.
Now it works fine

closed 2024-06-28 by peterhinch