micropython/micropython #15396: array: Add typecode property

jonnor, 2024-07-02:
### Description

An instance of array.array always has a `typecode`, which specifies the data format of the array.

When using an array, each item can only store a specific range of values based on the typecode. For example `B` is 0-255. When writing functions that operate on arrays, it is useful to check that the provided arrays have appropriate typecode values. For example, a function might only support arrays that are of the same typecode - or require at least a minimum range for values.

This is standard part of the CPython API that allows read-only access to the typecode info: https://docs.python.org/3/library/array.html#array.array.typecode

I propose to implement that, to enable Python code that checks the array typecode.

### Code Size

Would be behind the standard feature switch for array support.

### Implementation

I hope the MicroPython maintainers or community will implement this feature, I intend to implement this feature and would submit a Pull Request if desirable

### Code of Conduct

Yes, I agree

Gadgetoid, 2024-07-05:
The extra weight for *one* probably rarely used property with a working (if ugly) workaround is probably untenable, but it could probably be done in the same way memoryview's *one* (optional) property is implemented:

https://github.com/micropython/micropython/blob/4d16a9cced42ad0a298f1be8b993357abe7398ab/py/objarray.c#L248-L264

https://github.com/micropython/micropython/blob/4d16a9cced42ad0a298f1be8b993357abe7398ab/py/objarray.c#L636-L641

https://github.com/micropython/micropython/blob/4d16a9cced42ad0a298f1be8b993357abe7398ab/py/objarray.c#L649-L662

IE: an `ifdef` guard to make the feature optional where code size is paramount.

Note that this one memoryview property requires `MICROPY_CONFIG_ROM_LEVEL_EVERYTHING`, which is only enabled on the unix port so this wont get much mileage... probably why nobody has bothered so far!

jonnor, 2024-07-19:
Thank you for the consideration @Gadgetoid and the reference to how similar code is implemented.

I am actually OK with the workaround as it is. My only use case so far was to do some sanity checks on function inputs, and in unit tests. I made an `assert_array_typecode` function that encapsulates the odd implementation, so the rest of the code stays clean.

So I am closing this issue. Others can reopen if they really need this functionality.

jonnor, 2025-01-01:
This became a rather large performance issue in https://github.com/jonnor/micropython-npyfile/
Because the above workaround to inspect the typecode will cause the entire array to be serialized. For non-trivial array sizes this takes both a lot of time (especially float formatting...) and RAM.

So I intend to submit an MR to fix this. Would be happy if someone else beats me to the punch though ;)