micropython/micropython #15399: from_bytes, sign=True not recognized, behaves differently in micropython than in python

Brun059, 2024-07-03:
### Port, board and/or hardware

esp32-WROOM

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Generic ESP32 module with ESP32

### Reproduction

in micropython
```
int.from_bytes(b'\xFF\xFF' , 'big' , True)   --->  65535
```
in Python 3.11.4
```
int.from_bytes(b'\xFF\xFF' , byteorder='big' , signed=True)   ---> -1
```

### Expected behaviour

Expected to return  -1  
according to https://docs.python.org/3/library/stdtypes.html#int.from_bytes 

### Observed behaviour

returns 65535 which is equivalent to signed=False

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

Gadgetoid, 2024-07-05:
Looks like fixing this is a TODO:

https://github.com/micropython/micropython/blob/4d16a9cced42ad0a298f1be8b993357abe7398ab/py/objint.c#L392

Judging by it hitting `int_from_bytes`, `mp_obj_int_from_bytes_impl` and `mpz_set_from_bytes` and trying to work with arbitrary-sized ints, across multiple long-int implementations... it's a spicy and very low-level issue.

The complexity might explain why that TODO has been in the codebase for 10 years...

Suffice to say, I took a shot at fixing it and got completely rekd.

mattytrentini, 2024-07-05:
Related...

This is _almost_ documented as a CPython difference: [to_bytes method doesn’t implement signed parameter.](https://docs.micropython.org/en/latest/genrst/builtin_types.html#to-bytes-method-doesn-t-implement-signed-parameter). It would be better to fix the issue, but if that's too difficult then perhaps updating that to also indicate that `from_bytes` suffers from the (presumably) same issue.