micropython/micropython #15408: esp32: Enabling debug printers bootloops ESP32C3 boards

agatti, 2024-07-04:
### Port, board and/or hardware

esp32 port, generic ESP32C3 board

### MicroPython version

Without the changes to `ports/esp32/mpconfigport.h` mentioned later the banner is:

`MicroPython v1.24.0-preview.96.g92484d882 on 2024-07-04; ESP32C3 module with ESP32C3`

### Reproduction

Starting from a fresh git checkout from the master branch, enable both `MICROPY_DEBUG_PRINTERS` and `MICROPY_DEBUG_VERBOSE` in `ports/esp32/mpconfigport.h`, then build the ESP32 port for an `ESP32_GENERIC_C3` board with ESP-IDF 5.0.4.

### Expected behaviour

The board should show a regular REPL prompt on boot.

### Observed behaviour

The device enters into a boot loop with `RTC_SW_CPU_RST` as the reset cause:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xe8c
load:0x403cc710,len:0x6ec
load:0x403ce710,len:0x2b24
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x403807a8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xe8c
load:0x403cc710,len:0x6ec
load:0x403ce710,len:0x2b24
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x403807a8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xe8c
load:0x403cc710,len:0x6ec
load:0x403ce710,len:0x2b24
entry 0x403cc710
# Keeps printing boot messages
```

A transcript of a `gdb` session when a JTAG debugger is attached and the board is reset can be seen at https://gist.github.com/agatti/faa58c548824bdd4931f4f93fd8154e0

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

vshymanskyy, 2024-08-10:
I'm seeing a similar bootloop with `ESP32_GENERIC_C3`  when I build it with IDF `5.2.0`, and it does not happen on `5.0.4`.
Note, that I'm not enabling `MICROPY_DEBUG_PRINTERS` and `MICROPY_DEBUG_VERBOSE`.

agatti, 2024-08-10:
@vshymanskyy No idea if it's the same situation as mine, but if you can attach a debugger via the built-in JTAG port, can you please check if `mp_init` is called before the device crashes and restarts?

In my case the serial port `printf` equivalent tries to release the GIL mutex because the string is too long and it would hold things up, however the `printf` call is triggered by a part of the initialisation process that occurs _before_ the GIL mutex is set up (the garbage collector in my specific instance).

dpgeorge, 2024-08-16:
Fixed by 0b75e180a3b9fe72681fb942873102cdf2f4cae1

closed 2024-08-16 by dpgeorge