micropython/micropython #15420: BLE with multiple periperals (OSError: [Errno 107] ENOTCONN)

BaseCrusher, 2024-07-07:
### Port, board and/or hardware

Raspberry Pi Pico W

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040

### Reproduction

1. Connect to a BLE peripheral using BLE.gap_connect. This works flawlessly
2. Connect to a second BLE peripheral using BLE.gap_connect, while still connected to the first peripheral. (Error)

### Expected behaviour

Connections to both peripherals is possible without any errors

### Observed behaviour

OSError: [Errno 107] ENOTCONN

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

felixdoerre, 2024-07-07:
Maybe you want to compile your own firmware variant while increasing this config option:

https://github.com/micropython/micropython/blob/358e501e75c028133cff1045d2062ae31206a22c/extmod/btstack/btstack_config_common.h#L18

dpgeorge, 2024-07-08:
If the above `MAX_NR_HCI_CONNECTIONS` setting works then it might be a good idea to increase this by default in the standard config.  Because connecting to multiple peripherals is a relatively common thing to do.

(Would be interesting to see if NimBLE already supports multiple peripherals.)

But because increasing `MAX_NR_HCI_CONNECTIONS` takes up RAM (2264 bytes per connection) it might be good to then enable dynamic memory management with BTstack, so that this memory is allocated on demand from the MicroPython heap rather than statically allocated.

BaseCrusher, 2024-07-08:
Took me a bit of time to get it working, but now it does indeed work. For now, this unblocks me. Thank you very much. 

Never the less it would be a nice addition to have this allocated on demand.