micropython/micropython #15456: docs: explain optional global-interpreter-lock

mw66, 2024-07-12:
### Documentation URL

https://micropython.org/

### Description

I saw this on the homepage:

multithreading via the "_thread" module, with an optional global-interpreter-lock (still work in progress, only available on selected ports)

My question:

So by default micropython does not have GIL? then does it use native multithreading instead of green thread? E.g on a Linux multicore machine?

And how to do mutex between different native threads?

### Code of Conduct

Yes, I agree

peterhinch, 2024-07-13:
MicroPython does have a GIL. [This unofficial doc](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/THREADING.md) aims to give some insight into the GIL and threading.

jonnor, 2024-09-01:
Merge requests to include more details about threading/GIL etc into the official documentation are welcomed. They are in the docs/ folder in the git repository.