micropython/micropython #15465: super().__init__(*args, **kwargs) doesn't pass kwargs to parent C class.

kwagyeman, 2024-07-14:
### Port, board and/or hardware

openmv

### MicroPython version

v1.23.0

### Reproduction

See below.

### Expected behaviour

There should be no error.

### Observed behaviour

We have a class like this:

class Model(uml.Model):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
Its parent is a custom C class that implements make_new and accepts positional and keyword arguments.

When you pass just positional arguments to Model() in code using this subclass things work as expected. When you pass kwargs you get this error:

TypeError: function doesn't take keyword arguments

It appears MicroPython refuses to pass the kwargs to the parent class. Note that the error is not coming from code being executed in our make_new in C. It appears something else has rejected the kwargs being passed.

### Code of Conduct

Yes, I agree

ricksorensen, 2024-07-14:
This discussion on the micropython site might be pertinent:
https://github.com/orgs/micropython/discussions/10236

And the documentation on CPython differences:
https://docs.micropython.org/en/latest/genrst/modules.html#positional-only-parameters

kwagyeman, 2024-07-14:
Thanks, I think this clears it up.

closed 2024-07-14 by kwagyeman

stinos, 2024-07-15:
FWIW only did a quick test but this patch makes passing kwargs to a native base work for me; 

```
diff --git a/py/objtype.c b/py/objtype.c
index b6d600e94..e468ff116 100644
--- a/py/objtype.c
+++ b/py/objtype.c
@@ -84,14 +84,14 @@ static int instance_count_native_bases(const mp_obj_type_t *type, const mp_obj_t

 // This wrapper function is allows a subclass of a native type to call the
 // __init__() method (corresponding to type->make_new) of the native type.
-static mp_obj_t native_base_init_wrapper(size_t n_args, const mp_obj_t *args) {
+static mp_obj_t native_base_init_wrapper(size_t n_args, const mp_obj_t *args, mp_map_t *kw_args) {
     mp_obj_instance_t *self = MP_OBJ_TO_PTR(args[0]);
     const mp_obj_type_t *native_base = NULL;
     instance_count_native_bases(self->base.type, &native_base);
-    self->subobj[0] = MP_OBJ_TYPE_GET_SLOT(native_base, make_new)(native_base, n_args - 1, 0, args + 1);
+    self->subobj[0] = MP_OBJ_TYPE_GET_SLOT(native_base, make_new)(native_base, n_args - 1, kw_args->used, args + 1);
     return mp_const_none;
 }
-static MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN(native_base_init_wrapper_obj, 1, MP_OBJ_FUN_ARGS_MAX, native_base_init_wrapper);
+static MP_DEFINE_CONST_FUN_OBJ_KW(native_base_init_wrapper_obj, 1, native_base_init_wrapper);
```

apart from not being sure if there are caveats, the main issue is that because of MP_DEFINE_CONST_FUN_OBJ_KW it's needed to accept an `mp_map_t *` argument, so kwargs will get parsed already, but because make_new has a different signature they'll be parsed again. Been a while I looked into this so I don't know if there's a way to make instance_count_native_bases take `(size_t n_args, size_t n_kw, const mp_obj_t *args)` arguments.