micropython/micropython #15466: os.statvfs() misreports first 9 values on the list as 0 on Arduino Nano RP2040 Connect

Beguwula, 2024-07-15:
### Port, board and/or hardware

RP2 port, Arduino Nano RP2040 Connect

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Arduino Nano RP2040 Connect with RP2040

### Reproduction

import os
s = os.statvfs('/')
(f"Free storage: {s[1] * s[4] / 1024} KB")

### Expected behaviour

Expected to print "Free storage: 14800.0 KB" (or similar between 14000.0 and 16000.0)

### Observed behaviour

The code prints "Free storage: 0.0 KB"

### Additional Information

print(s) prints (0, 0, 0, 0, 0, 0, 0, 0, 0,128)

### Code of Conduct

Yes, I agree

Beguwula, 2024-07-15:
The code prints "Free storage: 1400.0 KB" on MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040

Beguwula, 2024-07-15:
> The code prints "Free storage: 1400.0 KB" on MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040

Works fine on a Raspberry Pi Pico with 2 MB of flash.
Doesn't work on an Arduino Nano RP2040 Connect with 16 MB of flash.
Same firmware version (v1.23.0 on 2024-06-02) on both.

Beguwula, 2024-07-15:
> The code prints "Free storage: 1400.0 KB" on MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040

print(s) prints (4096, 4096, 352, 350, 350, 0, 0, 0, 0, 255) on MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040

robert-hh, 2024-07-15:
There seem to be  a problem with the file system at the Aurduino Nano Connect. Please try to recreate the file system manually. statvfs itself works fine. To re-format the disk for LFS use:
```
import vfs
import rp2
bdev = rp2.Flash()
vfs.VfsLfs2.mkfs(bdev, progsize=256)
fs = vfs.VfsLfs2(bdev, progsize=256)
vfs.mount(fs, "/")
```
To reformat the disk for FAT use:
```
import vfs
import rp2
bdev = rp2.Flash()
vfs.VfsFat.mkfs(bdev)
fs = vfs.VfsFat(bdev)
vfs.mount(fs, "/")
```
Note that you must not change the files on the drive by code at the RP2040 as long as the drive is mounted. And you always have to properly eject the drive from the PC after having written to the board's file system.

Beguwula, 2024-07-15:
> There seem to be a problem with the file system at the Aurduino Nano Connect. Please try to recreate the file system manually. statvfs itself works fine. To re-format the disk for LFS use:
>
> ``` [...]

Which is a better option?

Beguwula, 2024-07-15:
@robert-hh I need to use 16 megabytes, that code only gives 2 megabytes according to the script.

Beguwula, 2024-07-15:
> @robert-hh I need to use 16 megabytes, that code only gives 2 megabytes according to the script.

I formatted as FAT

robert-hh, 2024-07-15:
The more robust option is LFS. You board will not mount as a drive if you use LFS.
Are you sure that you use the proper firmware for the board?

Beguwula, 2024-07-15:
> The more robust option is LFS. You board will not mount as a drive if you use LFS. Are you sure that you use the proper firmware for the board?

I accidentally forgot to swap to the correct board. My fault.

Beguwula, 2024-07-15:
[image]
Minicom supports hot-plug, so I thought it was the Arduino.

Beguwula, 2024-07-15:
@robert-hh Now it's correct.

Beguwula, 2024-07-15:
I think the problem is in hardware.

robert-hh, 2024-07-15:
What is reported by statvfs now, after re-formatting the file system? The Arduino Nano Connect configuration allocates 14MB for the file system.

Beguwula, 2024-07-15:
> What is reported by statvfs now, after re-formatting the file system? The Arduino Nano Connect configuration allocates 14MB for the file system.

Here's the output: Free storage: 14308.0 KB
@robert-hh

Beguwula, 2024-07-15:
> > What is reported by statvfs now, after re-formatting the file system? The Arduino Nano Connect configuration allocates 14MB for the file system.
> 
> Here's the output: Free storage: 14308.0 KB @robert-hh

Linux (and FreeBSD) reports it as a 15 MB FAT drive.

mattytrentini, 2024-07-15:
Appears resolved, so I'll close this. Please re-open if there is anything further to discuss.

robert-hh, 2024-07-15:
> Linux (and FreeBSD) reports it as a 15 MB FAT drive.

The file system size is 14*1024*1024 = 14_680_064, which rounds to 15 * 1000. The free space is always lower than that..