micropython/micropython #15471: rp2: Board does not enable USB after hard reset if an error is raised in boot.py.

robert-hh, 2024-07-16:
### Port, board and/or hardware

rp2, but maybe other's as well

### MicroPython version

MicroPython v1.24.0-preview.106.g55e75c4ad.dirty on 2024-07-16; Raspberry Pi Pico with RP2040

### Reproduction

Just create a boot.py file, which raises an exception and push reset or do a power cycle. On Soft reset this problem does not occur. I just the following test script with the option to bypass the test:
```
from machine import Pin
p=Pin(2, Pin.IN, Pin.PULL_UP)
if p():
    1/0
```

### Expected behaviour

Just boot with an exception message and bypass main.py.

### Observed behaviour

Board does not register USB and seems bricked. Uploading an older firmware including v1.23.0 helps.
Just a few interim commits: 
Fails with commit 8809ae716 shared/tinyusb: Buffer startup CDC data to send to host on connection.
Runs with the previous commit d144f0699 lib/tinyusb: Update to the most recent master.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

robert-hh, 2024-07-17:
Going with a debugger through the code, it hangs when trying to write an error message to the FIFO, but the TX FIFO is not overwritable. File tusb_fifo.c, line 484.  In that line, the flag `f->overwritable` is `false` at that time. And thus it locks up. Overwritable will be set to `true` later in the call to `mp_tusb_init()`, which is called **after** boot.py is executed.

robert-hh, 2024-07-18:
b.t.w.:  It does not have to be an exception in boot.py. Any output (e.g. print) by boot.py locks up the board.

dpgeorge, 2024-07-19:
Fixed by a734ee9057a760b0316eca110d71db1524142fec

closed 2024-07-19 by dpgeorge