micropython/micropython #15513: Base64 encoding into buffer, to reduce allocations

jonnor, 2024-07-20:
### Description

base64 encoding is supported by [binasci.b2a_base64](https://docs.micropython.org/en/latest/library/binascii.html). This implements the standard API in CPython. It always allocates a new buffer, and returns it. The allocations could be reduced/eliminated if a `b2a_base64_into` type API would (also) be available. This function would take an additional parameter: the buffer (bytearray) that is to be filled with encoded data. The caller must make sure it is sufficiently large (4/3 the size of the input + up to 4 bytes of padding).
This is similar to how there is struct.unpack_into, et.c.

 

### Code Size

Existing code could be moved into the `b2a_base64_into` implementation. And then have b2a_base64 be a compatibility wrapper, which creates the needed buffer internally. This should make the code increase minimal.

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

jonnor, 2024-07-20:
I ran into this when I needed to put a bunch of rather large binary data (tens of kB) through a JSON HTTP API. To keep memory usage down, I needed to do the base64 encoding in chunks / streaming. Which was doable with the existing APIs - see below for an implementation. The base64 encoder was then used as a generator for `request.requests` - to send out the chunks incrementally.  But it needs to run the garbage collector during encoding to avoid running out of memory. With an API as proposed above, it could just have a single buffer for the chunk.

```
import binascii
import gc

def compute_base64_length(n):
    o = ((4 * n // 3) + 3) & ~3
    return o

def base64_encoder(buf, chunksize=1000):
    """Generator for base64 encoding

    Keeps memory pressure down by calling the GC on each iteration

    Ideally MicroPython would offer a base64 encoder that could work with caller-supplied buffers.
    This would make it possible to allocate once, not per chunk.
    """

    # make chunk divisible by 3, as that is the unit base64 operate on
    chunksize = chunksize - (chunksize % 3)

    view = memoryview(buf)
    encoded_length = 0
    expect_length = compute_base64_length(len(buf)) 

    for start in range(0, len(buf)+chunksize, chunksize):

        # Encode one chunk
        end = start+chunksize
        if end > len(buf):
            end = len(buf)

        chunk = view[start:end]
        encoded = binascii.b2a_base64(chunk, newline=False)
        encoded_length += len(encoded)
        if encoded:
            yield encoded

        # Run garbage collection
        gc.collect()

    # check post conditions
    assert encoded_length == expect_length

def base64_encode(buf, chunksize=100):
    combined = bytearray()
    generator = base64_encoder(buf, chunksize=chunksize)
    for chunk in generator:
        combined += chunk
    return combined

def test_base64_encoder():

    chunksize = 10 # force splits into chunks
    for length in range(0, 50, 1):
        original = bytes(range(0, length))
        enc = base64_encode(original)
        decoded = binascii.a2b_base64(enc)
        assert decoded == original
```

dpgeorge, 2024-07-22:
I'm always in favour of making things inplace.  That just makes sense for embedded programming.

But we must consider CPython compatibility.  AFAIK CPython doesn't have any way to do inplace base64 encoding (or decoding, which might be good for symmetry of the API).  So we can't add the functionality as `binascii.b2a_base64_into`.  It needs to be put elsewhere.

We don't really have a good place for these kind of ad-hoc extensions to CPython and we did previously just add methods to CPython modules (eg `asyncio.sleep_ms`, `os.dupterm`), but we don't want to do that anymore (see recent move from `ssl` to `tls` module so we can add our own TLS functionality).

Or if you're brave and have the time, you could put an argument forward to the CPython devs to add this function in CPython, then we can add it here. 

---------

> But it needs to run the garbage collector during encoding to avoid running out of memory.

Do you really need to explicitly run the GC?  The runtime should be able to do this automatically when memory is exhausted.  Alternatively you can set `gc.threshold()` to run the GC more often.

jonnor, 2024-07-27:
> But we must consider CPython compatibility. AFAIK CPython doesn't have any way to do inplace base64 encoding (or decoding, which might be good for symmetry of the API). So we can't add the functionality as binascii.b2a_base64_into. It needs to be put elsewhere.
Agreed that decoding should probably be implemented at the same time, to have the symmetry.
Is it really not considered acceptable to have additional functions on a CPython type library? It seems very unlikely to break any existing or future programs by nature of just being there. Of course, if someone uses it (explicitly), the resulting program is now MicroPython specific. But as long as the functions are clearly documented as being implemented only in MicroPython / not CPython, I do not see practical problems?
Putting it in other modules introduces discoverability problems for API consumers, and complexity in the implementation for sharing the code. At least one would want a well-defined way of just doing that... There are many places where the CPython API is not the most embedded friendly, and it would be desirable to provide tailored functionality (in addition to the compatible ones). In-place operations being one category.

> Or if you're brave and have the time, you could put an argument forward to the CPython devs to add this function in CPython, then we can add it here.
That is not a level of bravery and resources that I do not have, unfortunately. At least for something like base64...

> Do you really need to explicitly run the GC? The runtime should be able to do this automatically when memory is exhausted. Alternatively you can set gc.threshold() to run the GC more often.

I was struggling with connection failures (exception thrown somewhere inside `requests.request`), if I did not do this GC/streaming encoding. But I must admit I do not understand how the requests library or its implementation of chunked requests work. So it is possible the issue was something else.