micropython/micropython #15521: RMT read pulses?

MarkHmnv, 2024-07-22:
### Description

The esp32.RMT has a method for sending pulses, but no method for reading them. I would like to know if there are plans to implement this functionality in the future?

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

elvis-epx, 2024-11-19:
I have the same needs and started working on support, initially for my own use (https://github.com/elvis-epx/micropython/blob/rmt_rx/ports/esp32/esp32_rmt2.c).

Found at least one complication: current MP uses a legacy RMT driver/API (ESP-IDF 4.x). The only good RX example I found was for the current RMT driver (https://github.com/junkfix/esp32-rmt-rf-rx). A build cannot use both RMT drivers at the same time, so I disabled existing esp32.RMT (and bitstream since it also uses RMT).

Ideally, the TX side of RMT should be updated to the new driver as well, but this will change the API since their concepts are a bit different (ticks are specified in Hz instead of a clock_div, etc).

elvis-epx, 2024-11-24:
Have submitted an RFC PR to update the current RMT module to the new API [1] which then allows us to collaborate the RMT RX that we have a PoC for [2]

[1] https://github.com/micropython/micropython/pull/16293
[2] https://github.com/elvis-epx/micropython/blob/rmt_rx/ports/esp32/esp32_rmt2.c

Zibri, 2025-01-15:
Any news? https://github.com/micropython/micropython/issues/16589