micropython/micropython #15528: Update libffi to 3.3 or later to unblock CI for RV64.

agatti, 2024-07-23:
### Description

After the fix for #12838 has landed, I've looked into adding CI support for the Unix port on RISC-V 64 bits.

Whilst normal builds work (as in, no cross compilation involved), cross-compilation using the normal features set for the Unix port to an RV64 target fails because libffi is too old - RISC-V support was added in version 3.3 of libffi (almost five years ago) and MicroPython uses .  I see that MicroPython's libffi submodule points to a fork that backports a few commits, maybe those changes are not needed anymore on a more recent version of libffi?

I've marked this as an improvement/enhancement rather than a bug as things still work as they are at the moment and whilst RISC-V 64 bits is not a priority for MicroPython, an update to libffi would impact on all other Unix ports at the very least.  On the other hand, having some sort of early warning for code breakages on RISC-V may prove helpful, especially as MicroPython for RISC-V 64 is now packaged in the OpenWRT distribution (see the original bug report), and is probably brought in somewhere else.

### Code Size

I have no idea on whether more recent versions of libffi would increase the code size.  Also, libffi would be used on systems much larger than a microcontroller, where modest size increases don't hurt as much as they would elsewhere (although it'd be better if size still stays small anyway).

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

dlech, 2024-07-23:
Orthogonal to updating the submodule, if the CI is Ubuntu, it should be possible to install the `libffi-dev:riscv64` package and use that for cross-compiling as well (probably need some extra steps to configure apt for riscv64 foreign arch).

agatti, 2024-07-23:
That's interesting - the Unix/i386 build installs `libffi-dev:i386` when setting its build environment up so it should be doable.  I'll look into that, thanks!

agatti, 2024-07-24:
A PR that puts the Unix RV64 port under CI, without having to update the libffi submodule, was submitted.  Thank you, @dlech for your suggestion!

I guess this can be closed then.

dpgeorge, 2024-07-24:
I think it would be a good idea to update our `lib/libffi` submodule to point to a more recent libffi version.

>  I see that MicroPython's libffi submodule points to a fork that backports a few commits,

Where do you see these commits?  The submodule we point to is `https://github.com/atgreen/libffi`, and following that link now goes to https://github.com/libffi/libff`.  And as far as I can tell, the latter is the official release and its master branch directly descends from the atgreen repo.

Note that we only use `lib/libffi` if `MICROPY_STANDALONE=1` is set.  That's the case only for the qemu unix builds (MIPS and ARM).

dpgeorge, 2024-07-24:
> > I see that MicroPython's libffi submodule points to a fork that backports a few commits,
> 
> Where do you see these commits?

Oh, I see now, https://github.com/atgreen/libffi is indeed 2 currently commits ahead of the official fork.

I was comparing against the commit pinned in our libffi submodule.  Our submodule does not have any patches on it compared to https://github.com/libffi/libffi

agatti, 2024-07-25:
With libffi 3.4.6 in the CI environment the Unix/Arm port cannot build due to `ffi_closure_alloc` and `ffi_closure_free` being reimplemented:

```
/usr/lib/gcc-cross/arm-linux-gnueabi/11/../../../../arm-linux-gnueabi/bin/ld: build-coverage/lib/libffi/out/lib/libffi.a(closures.o): in function `ffi_closure_alloc':
closures.c:(.text+0x139c): multiple definition of `ffi_closure_alloc'; build-coverage/alloc.o:/micropython/ports/unix/alloc.c:95: first defined here
/usr/lib/gcc-cross/arm-linux-gnueabi/11/../../../../arm-linux-gnueabi/bin/ld: build-coverage/lib/libffi/out/lib/libffi.a(closures.o): in function `ffi_closure_free':
closures.c:(.text+0x2670): multiple definition of `ffi_closure_free'; build-coverage/alloc.o:/micropython/ports/unix/alloc.c:101: first defined here
collect2: error: ld returned 1 exit status
```

Interestingly enough, the MIPS port using GCC 10 doesn't complain about that and works just fine.

@dpgeorge Would it be enough to remove MicroPython's `ffi_closure_alloc` and `ffi_closure_free` implementations, and change `mod_ffi_callback` to add the result of libffi's `ffi_closure_alloc` to the mmap region linked list (defined in `ports/unix/alloc.c`) there?  FFI closures created from the python side would still have their mmap allocations tracked, but not closures created from the C side, and I don't know if that would be an issue.

Otherwise, unless there's some flag somewhere I've missed, the only other option I can think of would be to weaken those two symbols inside `libffi.a` using objcopy.  That way MicroPython's implementations would take over and no real code change would be needed.  I'd rather keep that as a last resort if possible :)

dpgeorge, 2024-09-22:
Let's try to update libffi to 3.4.6.

> Would it be enough to remove MicroPython's `ffi_closure_alloc` and `ffi_closure_free` implementations, and change `mod_ffi_callback` to add the result of libffi's `ffi_closure_alloc` to the mmap region linked list (defined in `ports/unix/alloc.c`) there?

Yes, we should remove MicroPython's `ffi_closure_alloc` and `ffi_closure_free`.  They were added a long time ago in c9598604c6872d87b6c818c92efc4bce357d988c with the aim to reduce the size of the unix MicroPython executable.  I don't think that's a good reason anymore to have these custom functions, considering the trouble they are giving here.

The main issue is tracking GC pointers, in particular the result of of `ffi.callback()`.  The result of that call (which is a Python object) may be passed directly into a an ffi function, in which case the only reference to this Python callback object is that held by the C function.  For example in `micropython-lib`'s `signal.py` implementation, the callback is created and passed directly into the C `signal()` function, which may hold on to that pointer for the lifetime of the program.  There are no other pointers to this callback.

Currently the Python callback is tracked indirectly through `alloc.c` by putting the ffi closure's memory on the `mmap_region_head` linked-list, and then tracing all words within that memory (which is somewhat wasteful, computationally).

Instead of that I think we can use `m_tracked_calloc()` to allocate the `mp_obj_fficallback_t` object.  That will make it live forever but I can't see a way around that (they currently live forever because `ffi
closure_free` doesn't do anything).

agatti, 2024-09-22:
This should be now handled by #15879.  Thanks for the hints on memory management for the FFI callbacks!

closed 2024-09-23 by dpgeorge

dpgeorge, 2024-09-23:
libffi updated in 2b5feb9121a2fd6679c1ec0c6f4f2cb00b6e3616

PR #15879 (2024-09-23): unix/modffi: Let RV64 pass FFI tests again.
### Summary

PR #15800 combined with #15767 would trigger an incompatibility in the libffi version used by the CI environment for testing the RISC-V 64 Unix port.  Using a source-built libffi library would also bring up a RISC-V 64 specific issue when handling unsigned 32 bit values in certain cases.

The first problem is solved by updating the vendored libffi version to v3.6.4 (the latest stable at this point), and pointing the libffi submodule to the official libffi repository.  This, along with some FFI callback memory management changes, closes # [... cut]