micropython/micropython #15548: EXTI Power Voltage Detector doesn't work on STM32WB55

thibv, 2024-07-25:
### Port, board and/or hardware

STM32WB55

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; NUCLEO-WB55 with STM32WB55RGV6

### Reproduction

from pyb import Pin, ExtInt

def callback(line):
    line = line
    
EXTI_PVD_NUM = 16
exti = ExtInt(EXTI_PVD_NUM, ExtInt.IRQ_RISING_FALLING, Pin.PULL_NONE, callback)

exti.swint()

### Expected behaviour

Expected to call callback function, then get back to interpreter.

### Observed behaviour

- The host does not respond anymore, a reset is needed
- Power consumption goes from 5.2 mA to 9.7 mA after exti.swint()

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

thibv, 2024-07-29:
This code does not work either on Nucleo-STM32WL55JC1 board.

thibv, 2024-08-05:
@pi-anl and @dpgeorge , I saw that you've worked with these boards. Maybe you have some ideas ?
Thanks !

andrewleech, 2024-08-05:
I haven't used that feature myself, but a gdb debugger session would probably show pretty quickly what's going on...

thibv, 2024-08-13:
I've managed to build microPython from source and flash to my Nucleo board, but I don't know how to process with debugging.
- Do I need to use GDB or OpenOCD ?
- Do I need to enable something when building, to enable GDB ?

I've started openocd with this command :  
`openocd -f /usr/share/openocd/scripts/board/st_nucleo_wb55.cfg -c "program build-NUCLEO_WB55/firmware.hex verify" -c reset`

But when I send commands through REPL, nothing is shown in openocd.

andrewleech, 2024-08-14:
Hi, yes getting debugging going is a large topic in itself.
I happen to have a nucleo_wb55 set up for debugging in vscode so took a look when I found a little time.

Turns out WB55 exti works for most other lines / channels, however for PVD you get the lockup you've found.
The debugger showed me as soon as the swint is triggered the cpu gets stuck in a "default interrupt handler" which means that the correct interrupt handler hasn't been defined.

Looking at the startup asm code for the WB55 I found the handler for pvd is supposed to be `PVD_PVM_IRQHandler`  
https://github.com/micropython/stm32lib/blob/928df866e4d287ebc3c60726151513ebee609128/CMSIS/STM32WBxx/Source/Templates/gcc/startup_stm32wb55xx_cm4.s#L259

In the micropython stm interrupt handlers file however, it only had `PVD_IRQHandler` enabled by default, there's a separate `PVD_PVM_IRQHandler` which is currently only enabled for STM32L4. Once I enabled that handler for WB as well the lock-up is solved!
https://github.com/micropython/micropython/blob/6c870dc5ecc048309f7fcb21324b1aed165afdb0/ports/stm32/stm32_it.c#L518

Fix submitted in https://github.com/micropython/micropython/pull/15658

thibv, 2024-08-14:
Awesome @andrewleech , a big thank you ! It works like a charm.

Adding `|| defined(STM32WL)` allows the fix to correct the bug for the NUCLEO_WL55 too (I've tested it).

andrewleech, 2024-08-14:
Thanks for the feedback, I've updated my PR to include WL as well!

dpgeorge, 2024-08-20:
Fixed by 185116ea4139e84e7f7afb27d52b14dc03f0c2a8

closed 2024-08-20 by dpgeorge

andrewleech, 2024-12-03:
Hi @thibv I'm actually now interested in using the PVD on the WB55 - I don't suppose you'd have an example you could share of how you enable the PVD and set the trigger levels etc? 
Is it working for you that you then get the interrupt function called if the supply voltage drops below the threshold?

thibv, 2024-12-05:
Hello @andrewleech ,

Here is a file I wrote a few months ago to test PVD for a STM32WB board. If I remember correctly, I succeed in waking up from deep sleep with PVD, but I cannot get the root cause of the wakeup (as I enable both RTC and PVD for wake up sources).

Hope it can help you !
Thibv

```
from pyb import Pin, ExtInt
from micropython import const
import stm
from machine import mem32
import time

EXTI_PVD_OUTPUT = (16)

# Start the fw with start_loop()
# The FW should print "VDD is above/below THR" each time you change VDD supply

# Part 6.2.2 page 149 from documentation RM0434 Rev 13
# See page 68 : AHB4 PWR 0x5800 0400 - 0x5800 07FF
# The PVD can be used to monitor the VDD power supply by comparing it to a threshold
# selected by the PLS[2:0] bits in the PWR control register 2 (PWR_CR2).
# The PVD is enabled by setting the PVDE bit.
# A PVDO flag is available, in the PWR status register 2 (PWR_SR2), to indicate if VDD or the
# voltage level on PVD_IN is higher or lower than the PVD threshold. This event is internally
# connected to the EXTI Line16 and can generate an interrupt if enabled through the EXTI
# registers.
# The PVD output interrupt can be generated when VDD or the voltage level on PVD_IN drops
# below the PVD threshold, and/or when V DD or the voltage level on PVD_IN rises above the
# PVD threshold, depending on EXTI Line16 rising/falling edge configuration. If the EXTI
# Line16 is configured to rising/falling edge sensitivity (rising/falling edge of PVDO), the
# interrupt is generated when VDD or the voltage level on PVD_IN drops below/rises above
# the PVD threshold. As an example, the service routine can perform emergency shutdown
# tasks.

ADDR_PVDO = const(stm.PWR + stm.PWR_SR2)
ADDR_PLS = const(stm.PWR + stm.PWR_CR2) # PVD Threshold definition

PVD0 = 0b000 # ~ 2.0V
PVD1 = 0b001 # ~ 2.2 V
PVD2 = 0b010 # ~ 2.4 V
PVD3 = 0b011 # ~ 2.5 V
PVD4 = 0b100 # ~ 2.6 V
PVD5 = 0b101 # ~ 2.8 V
PVD6 = 0b110 # ~ 2.9 V

interrupt = False

#def callback(line):
#    print("line =", line)
#    #is_vdd_higher_than_thr()
#    global interrupt
#    interrupt = True

def is_vdd_higher_than_thr():
    reg = mem32[ADDR_PVDO] & (1 << 11)
    if (reg == 0):
        #print("VDD is above selected PVD thresold")
        return True
    else:
        #print("VDD is below selected PVD thresold")
        return False

def write_pvd_value(val):
    reg = (mem32[ADDR_PLS] & 0xffffffff)
    reg = reg & 0xfffffff1 # Remove PLS[2:0]
    reg = reg | (val << 1)
    mem32[ADDR_PLS] = reg

def enable_pvd(enable):
    reg = (mem32[ADDR_PLS] & 0xffffffff)
    if (enable):
        print("Enable PVD")
        reg = reg | 0x01
    else:
        print("Disable PVD")
        reg = reg & 0xfffffffe
    mem32[ADDR_PLS] = reg

def PVD_config():
    #/* 1. Enable Power Clock*/
    #_PWR_CLK_ENABLE();

    #/*2. Configure the NVIC for PVD */
    #HAL_NVIC_SetPriority(PVD_IRQn, 0, 0);
    #HAL_NVIC_EnableIRQ(PVD_IRQn);
    #extint = ExtInt(EXTI_PVD_OUTPUT, ExtInt.EVT_RISING_FALLING, Pin.PULL_NONE, callback)

    #/* Configure the PVD Level to */
    #sConfigPVD.PVDLevel = PWR_PVDLEVEL_3;
    #sConfigPVD.Mode = PWR_MODE_IT_RISING_FALLING;
    #HAL_PWR_PVDConfig(&sConfigPVD);
    print("Configure PVD to 2.9V")
    write_pvd_value(PVD6)

    #/* Enable the PVD Output */
    #HAL_PWR_EnablePVD();
    enable_pvd(1)

#enable_pvd(1)
#print("Configure PVD to 2.9V")
#write_pvd_value(PVD6)
#extint = ExtInt(EXTI_PVD_OUTPUT, ExtInt.IRQ_RISING_FALLING, Pin.PULL_NONE, callback)

#sw1 = Pin('C4', Pin.IN, Pin.PULL_UP)
sw2 = Pin('D0', Pin.IN, Pin.PULL_UP)
sw3 = Pin('D1', Pin.IN, Pin.PULL_UP)

led1_b = Pin('B5', Pin.OUT)
led2_g = Pin('B0', Pin.OUT)
led3_r = Pin('B1', Pin.OUT)

#ext_sw1 = ExtInt(Pin('C4'), ExtInt.IRQ_RISING_FALLING, Pin.PULL_UP, callback)

above_below = False

PVD_config()

def start_loop():
    global above_below
    global interrupt
    while (1):           
        if (is_vdd_higher_than_thr() != above_below):
            if (above_below):
                print("VDD is below THR")
                above_below = False
                led1_b.value(0)
            else:
                print("VDD is above THR")
                above_below = True
                led1_b.value(1)
        
        time.sleep_ms(10)
```

andrewleech, 2024-12-05:
That looks fantastic thanks!