micropython/micropython #15563: ESP32 wifi issues

fjmttty, 2024-07-28:
### Port, board and/or hardware

ESP32

### MicroPython version

esp32-20220618-v1.19.1.bin

### Reproduction

Issue No.1:On startup
Issue No.2:On startup / while running

### Expected behaviour

Connected to the wifi smooth and keep that.

### Observed behaviour

Issue No.1: "Wifi internal error" occurs. After power off and on, sometimes it works but mostly not. Once it gets recovered, it keep working without any error which looks strange for me. Just adding print(anything) may have solved...

Issue No.2:I'm using 30+ ESP32 with micropython. Most of them connect wifi smooth and keep it. But some 1 or 2 of them suddenly start struggling to connect wifi when start up(cannot connect even after the reboot, unplug power and plug) or lose connection when it's working. But after a while, it works completely normal. The boards which have this issue is random.

(1)This is the code to connect wifi.
"""Wifi"""
def connect_wifi(ssid, passkey, timeout=5): 
  global speed_filtered, bias, speed_raw 
  wifi= network.WLAN(network.STA_IF)
  wifi.active(False)
  time.sleep(1)
  wifi.active(True)
  wifi.connect(ssid, passkey)
  start = time.time()
  start_count = time.time()
  while not wifi.isconnected():
      print('*')
      time.sleep(1)
      if time.time()-start > timeout:
        wifi.disconnect()
        time.sleep(1)
        wifi.connect(ssid, passkey)
        start = time.time()
    
  if wifi.isconnected():
      print('Connected')
      print(time.time() - start_count)
      return wifi

(2) this is the code to check wifi status. if it's not connected, it try to reconnect.
if wifi.isconnected():
      if wifi_strength > 80:
        wifi_strength_category = 4
      elif wifi_strength > 60:
        wifi_strength_category = 3
      elif wifi_strength > 40:
        wifi_strength_category = 2
      else:
        wifi_strength_category = 1
    else:
      wifi = connect_wifi(SSID_NAME, SSID_PASS)

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

PsuFan, 2024-08-05:
have you inspected the physical antenna circuit? I've seen problems with physical damage.

projectgus, 2024-08-06:
> esp32-20220618-v1.19.1.bin

Please try using a current MicroPython binary from https://micropython.org/download/  (either the current release which is v1.23 or the nightly preview builds from the master branch) and report back if the problem still exists.