micropython/micropython #15564: Implement machine.USBDevice for ESP32-S3

tom709, 2024-07-28:
### Description

The ESP32-S3 chip has excellent USB support built-in. Please implement machine.USBDevice for ESP32-S3.

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

projectgus, 2024-07-30:
I'd also like to see this supported!

I realised that we might be able to do it another way, but locking the GIL for each of the USB callbacks. But that's also a lot of work, and I like the approach in your PR better.

TomFahey, 2024-10-14:
I noticed that your commits were merged last week, and so took the opportunity to compile a couple of micropython images for the Arduino Nano ESP32 and a Waveshare Pico S2.

I've only tested the use of basic USB CDC instantiation (using the 
 higher level micropython-lib [usb-device](https://github.com/micropython/micropython-lib/tree/master/micropython/usb) drivers, which make use of machine.USBDevice, however so far everything has worked flawlessly.

Fantastic job and thank you very much.

andrewleech, 2024-10-15:
That's really interesting to hear @TomFahey did you just enable the runtime build config?
```
#define MICROPY_HW_ENABLE_USB_RUNTIME_DEVICE    (1) // Support machine.USBDevice
```

TomFahey, 2024-10-15:
Pretty much - I also had to add `${MICROPY_DIR}/shared/tinyusb/mp_usbd_runtime.c` to

```c++
    list(APPEND MICROPY_SOURCE_TINYUSB
          ${TINYUSB_SRC}/tusb.c
          ${TINYUSB_SRC}/common/tusb_fifo.c
          ${TINYUSB_SRC}/device/usbd.c
          ${TINYUSB_SRC}/device/usbd_control.c
          ${TINYUSB_SRC}/class/cdc/cdc_device.c
          ${TINYUSB_SRC}/portable/synopsys/dwc2/dcd_dwc2.c
          ${MICROPY_DIR}/shared/tinyusb/mp_usbd.c
          ${MICROPY_DIR}/shared/tinyusb/mp_usbd_cdc.c
          ${MICROPY_DIR}/shared/tinyusb/mp_usbd_descriptor.c
          ${MICROPY_DIR}/shared/tinyusb/mp_usbd_runtime.c
      )
``` 

in the esp32_common.cmake file to get it to compile and also added 

```c++
    #if MICROPY_HW_ENABLE_USB_RUNTIME_DEVICE
    mp_usbd_deinit();
    #endif
``` 

to the `soft_reset_exit` block in main.c, however I don't know if the latter change is necessary (I was basically just trying to make it match the main.c file in the rp2 port).

sebromero, 2024-10-16:
Hi @TomFahey 👋  Would you mind sharing your build? I'd love to test HID support on the Arduino Nano ESP32.

TomFahey, 2024-10-16:
Sure, here's the .uf2 image, as well as the separate bootloader.bin, partition-table.bin and micropython.bin binaries. Just FYI, for the flash_args.txt file, for some reason it sets the application binary offset to 0xf70000, which doesn't leave enough space to write the application to the flash memory - I'm not entirely sure why? I just changed this to 0x10000, however YMMV.

[Arduino Nano ESP32.zip](https://github.com/user-attachments/files/17403386/Arduino.Nano.ESP32.zip)

sebromero, 2024-10-17:
I flashed the binaries using `esptool.py write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader.bin 0x10000 micropython.bin 0x8000 partition-table.bin 0xe000 ota_data_initial.bin`
and then ran the HID keyboard [example](https://github.com/micropython/micropython-lib/blob/master/micropython/usb/examples/device/keyboard_example.py) from the `usb-device-keyboard` library.
Works beautifully ✨ Thanks everyone who contributed to enable USB device support on the S3 ❤️

sebromero, 2024-10-18:
If I interpret @dpgeorge's comment [here](https://github.com/micropython/micropython/pull/15108#issuecomment-2390362561) correctly, the next step would be to create a PR that enables the runtime support by default. @TomFahey 's approach works fine for me, so maybe we can proceed with this? Or is there anything else needed from your point of view @andrewleech?

andrewleech, 2024-10-19:
Yep you're right, a PR with those changes is the next step! 
I'd suggest adding the define next to the existing USB ones in mpconfigport.h surrounded by a #ifndef like many others in that file so boards can override it if needed.

sebromero, 2024-10-19:
> I'd suggest adding the define next to the existing USB ones in mpconfigport.h surrounded by a #ifndef like many others in that file so boards can override it if needed.

Sounds good 👌 @TomFahey wanna do the honours given that you pretty much already have the changes locally? 🙏

Vincent1-python, 2025-07-29:
But now this driver is too old for ESP32P4 driver, and the code needs to be updated.