micropython/micropython #15568: NUCLEO_F746ZG LAN stops working after few days.

schef, 2024-07-29:
### Port, board and/or hardware

NUCLEO_F746ZG

### MicroPython version

Micropython v1.17

### Reproduction

Just using it for a while.

### Expected behaviour

To continue communicating.

### Observed behaviour

This happens randomly. It can happen after 1 day or after 20 days. The code continues to work.

### Additional Information

Micropython stops communicating with the LAN chip. I have no way of resetting the chip. I tried to reactivate the LAN driver. I also tried extending the C part of the LAN driver and look at the registers but did not find a solution there. Resetting the micropython solves the issue. I guess the LAN chip also gets a reset. Is there anything i can test or extend to find out what the issue is and how to fix it. Thanks.

### Code of Conduct

Yes, I agree

jonnor, 2024-08-24:
Thank you for the bug report. What does "stop working" mean exactly?

The latest version of MicroPython is 1.23. Does the issue also appear there?

I would recommend configuring the watchdog timer, and periodically refresh at a core location in your main loop. See
https://docs.micropython.org/en/latest/library/machine.WDT.html
This means your device will automatically reset if the issue happens, to automatically recover. This is a best practice for all embedded devices that run for extended periods of time.
Also make sure to catch all exceptions, and reset the board in that case. Otherwise the main.py will just exit

schef, 2024-08-24:
Hi, thanks for anwsering. I was trying to get to the bottom of this issue
but was unable to find. Deactivating and activating the network from the
code did not help. In the end i have written a logic if the network was
working but stoped to restart the device. But I would rather fix the issue
at the source. If you have any ideas on how to procede this issue i would
like to try it.

jonnor, 2024-08-24:
Debugging rare intermittent issues in combined software/hardware systems is notoriously difficult. You will have to catch the problem in the act. Do logging of all relevant state from the network subsystems, with timestamps. I would also use a hardware debugger (like Segger RTT) and trigger it when the code detects the issue. And also check signal integrity on all lines. A protocol debugger and oscilloscope will come in handy.
And try to then develop strategies to provoke the issue. This can be anything from increasing network activity, increasing other program activity, detuning clocks, power glitching, temperature swings, EMI, etc etc.

jonnor, 2024-08-24:
A classic problem in multi-chip systems is that the peripheral chip was reset or got into some wierd state, but the application processor does not detect this, and continues communicating as if there is no issue.

projectgus, 2024-08-27:
Just as one point of clarification, most of the ethernet (and higher level network) logic on this board is inside the main microcontroller - the STM32F746 has an integrated Ethernet MAC and the external LAN chip is the much simpler ethernet PHY. So I would guess probably the problem is internal to the micro, and it's either the ethernet driver, the lwIP TCP/IP stack, or the MicroPython program has gotten into a bad state.

Probably the first thing to try is to update MicroPython versions to the latest stable releases (1.23) and see if the problem still occurs.

If it does reproduce on 1.23, then for us to debug this further then we'd probably need some simple sample code that eventually gets stuck and instructions for how to set up whatever else that code is communicating with via ethernet. @schef does that sound possible?