micropython/micropython #15571: frambuf initalizes with ValueError

hoihu, 2024-07-30:
### Port, board and/or hardware

all variants that use the framebuf module

### MicroPython version

MicroPython v1.24.0-preview.149.g6007f3e20 on 2024-07-26; Raspberry Pi Pico W with RP2040

### Reproduction

I'm using a simple frambuffer with one bit per pixel. This code segment used to work on earlier micropython versions but fails now: 

```
import framebuf
w=21; h=8
data = bytearray(w*h // 8)
fb = framebuf.FrameBuffer(data, w, h, framebuf.MONO_HMSB)
```

### Expected behaviour

Expected to accept the input buffer without a `ValueError`

### Observed behaviour

this causes a 

```
 File "<stdin>", line 1, in <module>
ValueError: 
```

ValueError is raised with no further details

### Additional Information

I tracked this due to this input validation here:

https://github.com/micropython/micropython/blob/master/extmod/modframebuf.c#L318

and I think that's due to 
https://github.com/micropython/micropython/blob/master/extmod/modframebuf.c#L294

which modifies the `stride` value to be 24 (`(21 + 7) & ~7 == 24`) and therefore the provided input buffer is too small.  Modifying the example above with a input buffer of 24 then works as expected.

I'm not sure why this check was introduced. Maybe due to other functionalities in the framebuffer. But in any case it's now difficult to provide a matching input buffer size without the knowledge of what is checked inside the module.

This could be addressed via a doc update. Or raise a ValueError that indicates what went wrong (e.g. "expected buffer length of xxx, got yyy")

### Code of Conduct

Yes, I agree

hoihu, 2024-07-30:
seems to be introduced by https://github.com/micropython/micropython/commit/d040478d8aad7a05aac98651ec6ac5c2e9265a22 which solved https://github.com/micropython/micropython/issues/12562

Gadgetoid, 2024-07-31:
```
>>> 21/8
2.625
>>> ((21+7) & ~7)/8
3.0
```

The code mentioned seems to make `stride` a multiple of 8, presumably because `mono_horiz_setpixel` couldn't handle pixels wrapping on the Y axis to an adjacent byte.

21 is not a multiple of 8, so if you had a buffer 21x2 you would get:

```
111111111111111111111222
222222222222222222xxxxxx
```

Where 1 and 2 are rows 1 and 2 (Y axis) respectively. (might be that X/Y are swapped by its academic.

It's easier to round up to the nearest 8 and require your buffer be byte aligned, eg if stride is bumped to `24` in this case the bits in your buffer would be laid out like this:

```

111111111111111111111xxx
222222222222222222222xxx
```

And thus you would need to supply a 48 (or 24 in your case) length bytearray.

It's much easier to deal with the bits in the latter case than the former, so AIUI it's a wart you've just got to cope with.

Gadgetoid, 2024-07-31:
Hmm this is odd because

```
if (height_required * stride * bpp / 8 > bufinfo.len) {
```

implies (with the rounded up stride value):

```
1 * 24 * 1 / 8 == 3.0
```

So your buffer should be 3 bytes, not 24 since `FRAMEBUF_MHMSB` is bit packed :thinking:

hoihu, 2024-07-31:
@Gadgetoid Thanks a lot for the feedback!

>  So that would make your code something like (if we added a hypothetical sizeof method):

Sounds good to me. It would add some code though. 

> A possible alternate approach would be to allow a None in lieu of a bytearray or buffer object, and have the FrameBuffer constructor simply allocate a buffer of the correct size. You can always use fb.get_buffer() to access it.

Yes why not? It would hide the size of the frame buffer though.... I kind of like that the real buffer size and thus how many bytes are allocated are visible in the python code.

Btw, it's not just the mono-bitmap that has this behaviour: It's also the case for most other framebuffer modes where either `width` or `height` are patched. (The  exception would probably be `FRAMEBUF_GS8` and `FRAMEBUF_RGB565`).

I have no opinion for a solution. I feel it has to be improved in a way that's memory efficient and also not breaking any existing code. So perhaps one way to go is to add in the docs the constraints on buffer sizes per framebuffer mode. And add more verbosity / details to the Exception.

I'm happy to make a proposal if any of the maintainers could join in and share their opinion which way to go.

Gadgetoid, 2024-08-01:
You're right, the current situation is not great and the generic ValueError is not helpful either.

> It would hide the size of the frame buffer though.... 

You could just use `len(fb.get_buffer())` or in the case of both methods being added `framebuf.sizeof(w, h, type)`. Neither of these feel great, though. It's kinda difficult to hide all the quirks of graphics implementations, so one answer is just not to try.

> I'm happy to make a proposal

I reckon a pull-request adding `sizeof` (for want of a better name) would potentially be accepted, though maintainers are reticent to add anything that increases code size without a really good reason.

Of course if we're adding `sizeof` and expecting people to know about and use it (ie: read the documentation) then maybe it's just better to update the documentation like you suggest.

In any case, not raising a `NULL` `ValueError` when there is potentially very useful information to be surfaced to the user would be a really unambiguously good start. Though, again, precious, precious bytes!

hoihu, 2024-08-02:
> This should definitely be implemented in framebuf.

hmm isn't this switch used in the core module (`py` directory) exclusively? 

And a compile-in option forces the user to rebuild micropython.... not sure if that's the right way to go.

hoihu, 2024-08-02:
@Gadgetoid  Just pushed https://github.com/micropython/micropython/pull/15587 . This updates the framebuf documentation for the situation as-is. Happy to receive feedback.

Gadgetoid, 2024-08-03:
> isn't this switch used in the core module (py directory) exclusively?

It might be, but I'm not sure that implies any kind of thinking or rule that it must.

> And a compile-in option forces the user to rebuild micropython.

The intent here would be that the full error text is enabled by default, and if you're a user with some explicit low-flash use case you can rebuild with this (and the other) errors trimmed.

Of course I say this with the full knowledge that I've grappled with low flash for a while now, and am happy with the status quo 😬 😆 

> Happy to receive feedback.

Looks good to me!